1technophile/OpenMQTTGateway #355: Cannot send raw RF code using Pilight

arnie580, 2019-02-08:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
Sending raw codes (or any code for that matter) using pilight does not seem to work.

**To Reproduce**
Only Pilight module enabled.
Publish to: home/OpenMQTTGateway/commands/MQTTtoPilight
Payload: {"message":"{"code":"391 273 717 294 713 296 712 300 706 316 689 653 361 645 363 649 356 657 367 640 356 653 356 322 687 11767"}","protocol":"RAW"}

Result:
In serial logger "Hey I got a callback" and nothing else output. Log shows Pilight module is setup using the connected pins.

`WiFi ok with manual config credentials
OpenMQTTGateway mac:
* REMOVED *
OpenMQTTGateway ip:
192.168.0.189
1883
Connecting to MQTT by IP adress
192.168.0.249
RF_EMITTER_PIN
12
RF_RECEIVER_PIN
13
ZgatewayPilight setup done
MQTT_MAX_PACKET_SIZE
1024
JSON_MSG_BUFFER
512
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects
Hey I got a callback
`

**Expected behavior**
RF code sends, serial logger shows activity.

**Environment (please complete the following information):**
* OpenMQTTGateway version used : 0.9 master

arnie580, 2019-02-09:
I have tried sending the sample code on the Pilight RF wiki page, this also does not appear to work for me.

1technophile, 2019-02-09:
> I have tried sending the sample code on the Pilight RF wiki page, this also does not appear to work for me.

It seems that the wiki is rendering badly the quotation marks, could you try with this syntax:
`mosquitto_pub -t "home/OpenMQTTGateway/commands/MQTTtoPilight" -m '{"message":"{"systemcode":12,"unitcode":22,"off":1}","protocol":"elro_400_switch"}'`

arnie580, 2019-02-09:
I had spotted that and changed them to standard double quotes before posting, however no joy,

Double checked using ESP8266 Nodemcu v2 - same result as above.

1technophile, 2019-02-09:
I have just tested receiving, and got it working:
`home/OpenMQTTGateway/PilighttoMQTT {"message":"{\"id\":4,\"systemcode\":4181,\"unit\":4,\"state\":\"on\"}","protocol":"daycom","length":"4","repeats":102,"status":2}`

```
Creating RF PiLight buffer
Rcv. Pilight
Pub json into:
home/OpenMQTTGateway_ESP8266_TEST/PilighttoMQTT
{"message":"{\"systemcode\":21,\"unitcode\":1,\"state\":\"off\"}","protocol":"elro_800_switch","length":"","repeats":102,"status":2}
Creating RF PiLight buffer
```

Regarding publishing could you try with this syntax:
` mosquitto_pub -t "home/OpenMQTTGateway/commands/MQTTtoPilight" -m '{"message":"{\"systemcode\":12,\"unitcode\":22,\"on\":1}","protocol":"elro_400_switch"}'`

arnie580, 2019-02-09:
Tried sending, got a bit further but now have this in the logs:

```
Hey I got a callback
MQTTtoPilight json data analysis
MQTTtoPilight msg & protocol ok
MQTTtoPilight raw KO
ESPiLight Error:
invalid pulse train message
```

Receiving is now working - The only thing I have changed since last time is the transmitter pin from rx to D2.

1technophile, 2019-02-09:
Here is more info about sending with pilight

1technophile, 2019-02-09:
I can confirm sending is working:
```
home/OpenMQTTGateway/commands/MQTTtoPilight {"message":"{\"systemcode\":21,\"unitcode\":4,\"off\":1}","protocol":"elro_800_switch","length":"","repeats":103,"status":2}
home/433toMQTT 1119316
home/433toMQTT 1119316
home/433toMQTT 1119316
home/433toMQTT 1119316
```
Above you can see the command sent with Pilight from an ESP8266 and the reception made by ZgatewayRF on an Arduino Uno
Please note that the emitter is connected to D2 (RF_EMITTER_PIN 4)

arnie580, 2019-02-09:
Thanks for looking at this, I'll check tomorrow unfortunately I'm at work the rest of today.

1technophile, 2019-02-09:
In all the case thanks for pointing this, the wiki has been updated

1technophile, 2019-02-15:
Hello, Do you have some feedback to give about this request?

arnie580, 2019-02-23:
Hi, really sorry for the delay in updating, life got in the way.

I tried in and sending the example code from one ESP to another worked, however I still haven't been able to get it to send using a raw code. I have managed to get my particular remote working with the basic 433 implementation though. Happy for the issue to be closed.

closed 2019-08-14 by stale[bot]

melyux, 2020-08-09:
@1technophile Is there an example on how to send raw code to MQTTtoPilight?

1technophile, 2020-08-09:
> @1technophile Is there an example on how to send raw code to MQTTtoPilight?
No I don't have unfortunately

melyux, 2020-08-10:
I actually figured it out through trial and error. You have the feature in OMG ([see here](https://github.com/puuu/ESPiLight/issues/39)), so adding an example to the documentation would be nice, and also a way to change how many times the raw code is repeated (the default seems to be 10).

Here's how you do it. You send an MQTT message with a JSON payload to the `home/OpenMQTTGateway/commands/MQTTtoPilight` topic (substituting `home` and `OpenMQTTGateway` with the optionally changed settings in `User_config.h`, with a single key "raw", e.g.:
`{"raw":"c:03020202010102020102010101010101010202020201020102020202020101010201010202;p:563,935,1935,3930@"}`

This example sends a Nexus protocol message containing a temperature of 24.5°C, battery status 1 (OK), humidity 19% on channel 1 with a sensor ID of 230. But you can do it for any protocol. Here's how.

This uses the "pulse train string" protocol of MQTT433gateway/ESPilight/Pilight (one of those, I don't know which one of them it's proprietary to, but OMG uses it). The numbers after `c:` are 0-based indexes that reference the numbers after `c:`, which are time measurements. These time measurements are the microseconds involved; however, I had to add/subtract with them to get them to work properly, since the timing is somewhat poor. I don't know if the problem is the software, my 433MHz transmitter, the antenna I used with it, or what.

The numbers after `c:` are in pairs (the first one in the example is `03`), the first in each pair (in this case `0`) being an index to the array in `p:` for how long the **pulse** should be (in this case 563, which is a theoretic 500 microseconds). The second in the pair (in this case `3`) is an index to the array in `p:` for how long the subsequent **gap** should be (in this case 935, which is a theoretic 4000 microseconds). This is the delimiter that starts each Nexus protocol transmission repetition. A 500 microsecond pulse (a high state) followed by a 4000 microsecond gap (a low state).

The second pair of digits after `c:` is `02`. The 0 is a pulse (high) of p[0] = ~500µs and the 2 is a gap (low) of p[2] = ~2000 µs. This represents a logical `1` according to the Nexus protocol, which will change based on the protocol you want to transmit, of course. So the pair `02` here means you're sending a logical `1`. The only other kind of pair we use here is `01`, which represents a pulse of 500µs followed by a gap of 1000µs, which is a logical `0`.

I used a SDR dongle with `rtl_433` software to verify that the message I sent was correctly parsed (even though it only uses 10 repetitions instead of the original 12) and the Nexus-protocol weather station I aimed it at correctly received it. You can come up with your own raw message by finding what length of pulse (highs) and gap (lows) combinations correspond to logical bits in the protocol you want to send in, and then put the various lengths of time in the `p:` array while referencing them in the `c:` array.

You can use `rtl_433` with an SDR dongle to check if/how you need to tweak the lengths to overcome the aforementioned inaccuracies. Running `rtl_433 -vvvv` will show the pulses and gaps of your transmitted message. But it shows them as 1/4 of the actual microseconds involved for some reason. So in the below snippet from `rtl_433`'s verbose output, 126/127 is a theoretical 125, which multiplied by 4 gives us the actual 500µs gap. The 499/500 is a theoretical 500 * 4 = 2000µs. The 247 is a theoretical 250 * 4 = 1000µs. So in the Nexus protocol, this snippet represents a logical `1001`.

```
...
439] Pulse:  126, Gap:  500, Period:  626
[440] Pulse:  126, Gap:  247, Period:  373
[441] Pulse:  127, Gap:  247, Period:  374
[442] Pulse:  127, Gap:  499, Period:  626
...
```

For some reason, OMG didn't parse the transmitted message itself despite `rtl_433` and the weather station itself parsing it correctly. I don't know if this a feature of OMG to ignore parsing messages you transmit yourself, or because it uses only 10 repetitions (which OMG has no way to change, like I said) instead of 12, or if there's a timing issue somewhere, or whether the protocol parsing code I used is more strict than even the weather station itself.

1technophile, 2020-08-10:
Thanks for the investigation! It is interesting.
Could you describe me your goal of sending raw weather station signals?

melyux, 2020-08-11:
I can use this to display any temperature I want on the weather station screen! For example, I can make it show the temperature in a different country on Channel 2. You can use these as cheap screens for temperature data in that way. For Channel 1, I have a well-calibrated DS18B20 thermometer already established that I will use to deliver exact temperature data to the weather station instead of the uncalibratable temperatures delivered by the wireless sensors included with the weather station.

1technophile, 2020-08-11:
Nice idea !