1technophile/OpenMQTTGateway #356: RFM95 (LoRa) support

knopserl, 2019-02-10:
Hello, great project! I'm trying to integrate a couple of Xiaomi Mijia Temp/Hum BLE sensors with LCD display  (link dbelow). I think people have alreayd tested it with this project.
Finally I wanted to connect it with OpenHAB (via MQQT).
SInce I have also other devces (LoRA) using RFM95, it would be great  to have that in addition to the RFM69.
The reason is, that I have a couple of ESP32 boards, which have the RFM95 (SX1276) on-baord (not the RFM69). 
A couple of that product are availbel (very cheap).
Another quick question: If I want to use the W5500 instead of the W5100, does that work? Oth do we need another W500 library? Or should I use the ESP32 WiFI together wit Bluetooth instead of Ethernet? Does that work in parallel?

Here is the sensor I wanted to connect:
https://www.banggood.com/Xiaomi-Mijia-Bluetooth-Thermometer-Hygrometer-with-LCD-Screen-Magnetic-Suction-Wall-Stickers-p-1232396.html?cur_warehouse=CN

1technophile, 2019-02-10:
> Hello, great project! I'm trying to integrate a couple of Xiaomi Mijia Temp/Hum BLE sensors with LCD display (link dbelow). I think people have alreayd tested it with this project.

It is tested and working since V0.8

> SInce I have also other devces (LoRA) using RFM95, it would be great to have that in addition to the RFM69. The reason is, that I have a couple of ESP32 boards, which have the RFM95 (SX1276) on-baord (not the RFM69).

I agree, it could be an interesting feature

> Another quick question: If I want to use the W5500 instead of the W5100, does that work? Oth do we need another W500 library? Or should I use the ESP32 WiFI together wit Bluetooth instead of Ethernet? Does that work in parallel?

If you have the choice between ethernet and wifi I would go for ethernet, you will get faster response time.
Regarding the w5500 I never tested it, it could be interesting to have your feedback if you have one.

For questions and support I advise you to go to the [forum/community](https://community.openmqttgateway.com/), github issues are didicated to bug/feature request (and on per issue)

knopserl, 2019-02-10:
thanks for the quick response. I'll test it later with the W5500 (which I have at home).

knopserl, 2019-02-18:
I have seen, that I ran out of W5500, so I ordered W5500 and W5100 (the W5500 are the newer actually better LAN chips). However, the main request was rather the RFM95 (LoRa) support, as a couple of the  ESP32 modules have it on-board (RFM95 or SX1276 which is the same), thus it would be a more interesting Radio than the RFM69 and has even longer range.

1technophile, 2019-02-21:
A first implementation tested on 2 TTGO LORA32 V2.0

closed 2019-02-21 by 1technophile

kiralikbeyin, 2019-02-22:
@1technophile 
Can i use 433MHz RF libaries with TTGO or Heltec Lora ESP32? Because RF Receive/send Modules are really crappy..

https://github.com/1technophile/OpenMQTTGateway/wiki/ESP32-RF-Send-and-Receive

Heltec
https://www.ebay.com/itm/SX1278-LoRa-ESP32-0-96-inch-Blue-ESP8266-OLED-Display-Bluetooth-WIFI-Lora-Kit-32-/282594201585

TTGO
https://www.banggood.com/Wemos-TTGO-LORA-SX1278-ESP32-0_96OLED-16-Mt-Bytes-128-Mt-bit-433Mhz-For-Arduino-p-1205930.html?cur_warehouse=CN

1technophile, 2019-02-22:
If you mean to use the ttgo integrated transceiver for basics protocols supported by ZgatewayRF, the answer is not with the current code and I don't if we can hack the transceiver to do others protocols than Lora. By the way why do you think RF modules are crappy ? 
May I advise you to continue this discussiin on the community forum, i don't want to hijack this issue.

knopserl, 2019-02-23:
@1technophile  wow, I'm impressed, you did it! I will test it tomorrow. I have ESP32 Heltec and TTGo modules with SX1276 (equal to RFM95) on board which are the 868MHz versions of LoRa. I also have own modules with ESP32 and RFM95 modules (which are really good).
Now with LoRa we have the longest and reliable range, a really good enhancement!
With BLE I had the problem, that from my office to the kitchen (less then 8meters but 2 walls, I loose the connection to the Xiaomi BLE sensors. 
I'll give you feedack soon and thanks a lot. I hope that also others will like it!

https://de.aliexpress.com/item/2-st-cke-868-mhz-915-mhz-SX1276-ESP32-LoRa-0-96-zoll-Blau-OLED-Display/32882205132.html?spm=a2g0x.search0104.3.2.124838c42uobD7&ws_ab_test=searchweb0_0%2Csearchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10902_10059_10884_10887_321_322_10103%2Csearchweb201603_51%2CppcSwitch_0&algo_pvid=5f63627f-d246-441e-a373-0843398b29cb&algo_expid=5f63627f-d246-441e-a373-0843398b29cb-0

https://de.aliexpress.com/item/RFM95-RFM95W-868-915-RFM95-868MHz-RFM95-915MHz-LORA-SX1276-wireless-transceiver-module-Best-quality/32832739318.html?spm=a2g0x.search0104.3.29.4a2660e1M3R6Mt&ws_ab_test=searchweb0_0,searchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10902_10059_10884_10887_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=1266facc-103d-4c7f-9b7c-627f372785a8-4&algo_pvid=1266facc-103d-4c7f-9b7c-627f372785a8

https://de.aliexpress.com/item/TTGO-SX1276-SX1278-LoRa-ESP32-868-915-mhz-433-mhz-Bluetooth-WI-FI-Internet-Antenne-Development/32844017936.html?spm=a2g0x.search0104.3.39.124838c4Pq6TGQ&ws_ab_test=searchweb0_0%2Csearchweb201602_10_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10902_10059_10884_10887_321_322_10103%2Csearchweb201603_51%2CppcSwitch_0&algo_pvid=5f63627f-d246-441e-a373-0843398b29cb&algo_expid=5f63627f-d246-441e-a373-0843398b29cb-5

1technophile, 2019-02-24:
Indeed it is an interesting add to OMG !  I wait for your feedback, at this time this a first implementation. There is place for improvments (like adding sender and receiver adresses) but I need to go in deep with lora protocol.