1technophile/OpenMQTTGateway #358: Does autodiscovery in hass with mqtt work ?

rolfberkenbosch, 2019-02-12:
Hi, first of all thank you for your great work!

I have setup an ESP32 with MQTT with autodiscovery to hass. But it looks like it doesn't work as expected. I see only tree new items in hass, but not my bluetooth devices. The following items i see in hass:

Entity | State | Attributes
-- | -- | --

binary_sensor.openmqttgateway | off | friendly_name: OpenMQTTGateway device_class: connectivity
binary_sensor.openmqttgatewayblt | on | friendly_name: OpenMQTTGateway device_class: connectivity
sensor.gatewaybt | unknown | friendly_name: gatewayBT

I have the log from my mqtt server down below:

home/OpenMQTTGateway/LWT online
home/OpenMQTTGateway/version 0.9
home/home_presence/OpenMQTTGateway {"id":"4f:26:2f:cb:0c:3b","manufacturerdata":"L","rssi":-88,"distance":19.7325}
home/home_presence/OpenMQTTGateway {"id":"50:e3:6c:58:2a:1c","manufacturerdata":"L","rssi":-89,"distance":21.51847}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-82,"distance":11.4949}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-82,"distance":11.4949,"servicedata":"71209800de1110668d7cc40d071003fd0000","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"lux":"253"}
home/home_presence/OpenMQTTGateway {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-89,"distance":21.51847}
home/OpenMQTTGateway/BTtoMQTT/69D9A1BE3F43 {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-89,"distance":21.51847,"servicedata":"0000000000000000000000000000000000000000","servicedatauuid":"0000fe9f-0000-1000-8000-00805f9b34fb"}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-89,"distance":21.51847}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-89,"distance":21.51847,"servicedata":"71209800e0cb0e668d7cc40d0910023300","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"fer":"51"}
homeassistant/binary_sensor/30AEA44E8B08/config {"stat_t":"home/OpenMQTTGateway/LWT","name":"OpenMQTTGateway","uniq_id":"30AEA44E8B08","dev_cla":"connectivity","pl_on":"online","pl_off":"offline","opt":false,"ret":true,"pl_avail":"online","pl_not_avail":"offline","device":{"name":"OpenMQTTGateway","manufacturer":"OMG_community","sw_version":"0.9","identifiers":["30AEA44E8B08"]}}
homeassistant/sensor/30AEA44E8B08gatewayBT/config {"stat_t":"home/OpenMQTTGateway/BTtoMQTT/","name":"gatewayBT","uniq_id":"30AEA44E8B08gatewayBT","val_tpl":"{{ value_json.message }}","opt":false,"ret":false,"device":{"name":"OpenMQTTGateway","manufacturer":"OMG_community","sw_version":"0.9","identifiers":["30AEA44E8B08"]}}
home/OpenMQTTGateway/SYStoMQTT {"uptime":120,"freeMem":24652,"rssi":-43,"SSID":"Marrol","modules":"BTHADiscovery"}
home/home_presence/OpenMQTTGateway {"id":"4f:26:2f:cb:0c:3b","manufacturerdata":"L","rssi":-92,"distance":27.75274}
home/home_presence/OpenMQTTGateway {"id":"50:e3:6c:58:2a:1c","manufacturerdata":"L","rssi":-89,"distance":21.51847}
home/home_presence/OpenMQTTGateway {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-87,"distance":18.07764}
home/OpenMQTTGateway/BTtoMQTT/69D9A1BE3F43 {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-87,"distance":18.07764,"servicedata":"0000000000000000000000000000000000000000","servicedatauuid":"0000fe9f-0000-1000-8000-00805f9b34fb"}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-93,"distance":30.15532}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-93,"distance":30.15532,"servicedata":"71209800e51110668d7cc40d0410024800","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"tem":"7.2"}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-89,"distance":21.51847}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-89,"distance":21.51847,"servicedata":"71209800e6cb0e668d7cc40d0710036e0000","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"lux":"110"}
home/home_presence/OpenMQTTGateway {"id":"4f:26:2f:cb:0c:3b","manufacturerdata":"L","rssi":-90,"distance":23.44428}
home/home_presence/OpenMQTTGateway {"id":"50:e3:6c:58:2a:1c","manufacturerdata":"L","rssi":-90,"distance":23.44428}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-87,"distance":18.07764}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-87,"distance":18.07764,"servicedata":"71209800eccb0e668d7cc40d0910023200","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"fer":"50"}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-81,"distance":10.46739}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-81,"distance":10.46739,"servicedata":"71209800eb1110668d7cc40d08100111","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"moi":"17"}
homeassistant/binary_sensor/30AEA44E8B08/config {"stat_t":"home/OpenMQTTGateway/LWT","name":"OpenMQTTGateway","uniq_id":"30AEA44E8B08","dev_cla":"connectivity","pl_on":"online","pl_off":"offline","opt":false,"ret":true,"pl_avail":"online","pl_not_avail":"offline","device":{"name":"OpenMQTTGateway","manufacturer":"OMG_community","sw_version":"0.9","identifiers":["30AEA44E8B08"]}}
homeassistant/sensor/30AEA44E8B08gatewayBT/config {"stat_t":"home/OpenMQTTGateway/BTtoMQTT/","name":"gatewayBT","uniq_id":"30AEA44E8B08gatewayBT","val_tpl":"{{ value_json.message }}","opt":false,"ret":false,"device":{"name":"OpenMQTTGateway","manufacturer":"OMG_community","sw_version":"0.9","identifiers":["30AEA44E8B08"]}}
home/OpenMQTTGateway/SYStoMQTT {"uptime":240,"freeMem":24724,"rssi":-43,"SSID":"Marrol","modules":"BTHADiscovery"}
home/home_presence/OpenMQTTGateway {"id":"59:4c:4f:05:45:91","manufacturerdata":"L","rssi":-93,"distance":30.15532}
home/home_presence/OpenMQTTGateway {"id":"4f:26:2f:cb:0c:3b","manufacturerdata":"L","rssi":-92,"distance":27.75274}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-86,"distance":16.54559}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-86,"distance":16.54559,"servicedata":"71209800f21110668d7cc40d071003c60000","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"lux":"198"}
home/home_presence/OpenMQTTGateway {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-90,"distance":23.44428}
home/OpenMQTTGateway/BTtoMQTT/69D9A1BE3F43 {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-90,"distance":23.44428,"servicedata":"0000000000000000000000000000000000000000","servicedatauuid":"0000fe9f-0000-1000-8000-00805f9b34fb"}
home/home_presence/OpenMQTTGateway {"id":"50:e3:6c:58:2a:1c","manufacturerdata":"L","rssi":-90,"distance":23.44428}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-92,"distance":27.75274}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-92,"distance":27.75274,"servicedata":"71209800f3cb0e668d7cc40d08100106","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D660ECB {"moi":"6"}
home/home_presence/OpenMQTTGateway {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-86,"distance":16.54559}
home/OpenMQTTGateway/BTtoMQTT/69D9A1BE3F43 {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-86,"distance":16.54559,"servicedata":"0000000000000000000000000000000000000000","servicedatauuid":"0000fe9f-0000-1000-8000-00805f9b34fb"}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-85,"distance":15.12852}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-85,"distance":15.12852,"servicedata":"71209800f91110668d7cc40d0410024800","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"tem":"7.2"}
home/home_presence/OpenMQTTGateway {"id":"c4:7c:8d:66:0e:cb","name":"Flower care","rssi":-91,"distance":25.51913}
[... cut ...]
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"id":"c4:7c:8d:66:10:11","name":"Flower care","rssi":-86,"distance":16.54559,"servicedata":"71209800ff1110668d7cc40d08100111","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OpenMQTTGateway/BTtoMQTT/C47C8D661011 {"moi":"17"}
home/home_presence/OpenMQTTGateway {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-90,"distance":23.44428}
home/OpenMQTTGateway/BTtoMQTT/69D9A1BE3F43 {"id":"69:d9:a1:be:3f:43","manufacturerdata":"�","rssi":-90,"distance":23.44428,"servicedata":"0000000000000000000000000000000000000000","servicedatauuid":"0000fe9f-0000-1000-8000-00805f9b34fb"}

With best regards,

Rolf

1technophile, 2019-02-15:
Hello,

Indeed autodiscovery for mi flora or mijia is not still implemented, I suppose this is what you are looking for? Or you would like to have auto discovery for presence detection?

1technophile, 2019-03-18:
Please note that autodiscovery for mi flora and mi jia has been added to the development branch

1technophile, 2019-04-13:
If you are still having troubles you can reopen the issue

closed 2019-04-13 by 1technophile