1technophile/OpenMQTTGateway #359: Reset of MQTT values (multi gateway setup)

tunip, 2019-02-13:
**Describe the bug**
I use 3 ESP32 gateways with BT only enabled for my 3 Xiaomi Flower & 7 MiJia readings.
MQTT values are cleared from time to time for the MiJia's (Flower sensors are not affected).

Custom OMG settings for each gateway: 
As I need a unique state_topic in Home Assistant, I removed the Gateway_Name from the topic in config_BT.h.
The Gateway_Name is still unique on each ESP32 in User_config.h.

**To Reproduce**
Steps to reproduce the behavior:
1. Publish to 'OMG/BTtoMQTT/<xiaomi-mac>'
2. Subscribe on 'OMG/BTtoMQTT/<xiaomi-mac>'
3. See screenshot

**Expected behavior**
In my example, the temperature should not be "cleared" from one of the three gateways.

**Screenshots**
[image]
[image]

**Environment:**
* OpenMQTTGateway version used: V0.9

1technophile, 2019-02-13:
Hello,

Did you identified the mqtt message that is setting the value to undefined ?

tunip, 2019-02-13:
Gateway publish:
`OMG/BTtoMQTT/4C65A8.....8`
`{"hum":"63.5"}`
...temperature is gone

next scan...

`OMG/BTtoMQTT/4C65A8.....8`
`{"hum":"63.5","tem":"23.2"}`
...temperature is back

Why home assistant undefines temperature when only humidity is published from the gateway is a good question. If both values are published everything is fine.

Maybe I need to publish every value in a seperate topic.

1technophile, 2019-02-15:
We need to search on Hass doc to see if there is a way to don't reset the values when the key is not present, or maybe ask the commmunity in hass forum

tunip, 2019-02-15:
You are right. I will ask this in the HA community. Many thx

closed 2019-02-15 by tunip