1technophile/OpenMQTTGateway #361: Hassio - Home assistant 0.87.1 MQTT autodiscovery

kegra, 2019-02-16:
I am getting multiple errors using HA autodiscovery. and no devices appear under integrations.
i am using only the BT component.

Most of the errors are unable to parse JSON errors
Sat Feb 16 2019 19:26:42 GMT+0200 (EET)
Unable to parse JSON 5CCF7FFD89DAgatewayBT: '{{ value_json.message }}'

Sat Feb 16 2019 19:26:42 GMT+0200 (EET)
Unable to parse JSON 5CCF7FFD89DAgatewayBT: '5CCF7FFD89DAgatewayBT'

Sat Feb 16 2019 19:26:42 GMT+0200 (EET)
Unable to parse JSON 5CCF7FFD89DAgatewayBT: 'home/OpenMQTTGateway/BTtoMQTT/'

Sat Feb 16 2019 19:24:41 GMT+0200 (EET)
Unable to parse JSON 5CCF7FFD89DA: 'home/OpenMQTTGateway/LWT'

but also i get the following error.

Sat Feb 16 2019 19:26:42 GMT+0200 (EET)
Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'state_topic': 'home/OpenMQTTGateway/BTtoMQTT/', 'name': 'gatewayBT', 'unique_id': '5CCF7FFD89DAgatewayBT', 'value_template': '{{ value_json.message }}', 'optimistic': False, 'retain': False, 'device': {'name': 'OpenMQTTGateway', 'manufacturer': 'OMG_community', 'sw_version': '0.9', 'identifiers': ['5CCF7FFD89DA']}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/homeassistant/components/mqtt/sensor.py", line 70, in async_discover_sensor
    config = PLATFORM_SCHEMA(discovery_payload)
  File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['optimistic']

I haven't used OpenMQTTGateway v9 with HA auto discovery in prior versions of HA bellow 0.87.1

1technophile, 2019-02-16:
I didn't had problems with the previous hass version. I would need to upgrade and see if I reproduce your issue.

1technophile, 2019-02-18:
Seems that there was some changes with the last HASS version on payload handling:
https://community.home-assistant.io/t/mqtt-automation-broke/99826/5

Kedryn, 2019-02-18:
I have the same problem. It was working before, now i get the same errors.

kegra, 2019-02-27:
> I didn't had problems with the previous hass version. I would need to upgrade and see if I reproduce your issue.

did you had the chance to try it out ?

1technophile, 2019-02-27:
you will have news this week

1technophile, 2019-03-01:
test done on 0.88.2
I see two devices on an OMG frame:

- OMG binary sensor
- gatewayBT

The state of binary sensor is connected, the state of gatewayBT is unknown.

I don't see errors in system logs

indy55, 2019-03-09:
Running the latest dev version which introduced discovery for mi flora and am receiving similar errors to the original post. Discovery is enabled in HA. Any thoughts?

2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'home/OpenMQTTGateway/LWT'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'OpenMQTTGateway'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'B4E62DD9E0A9'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'connectivity'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'online'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'offline'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'online'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9: 'offline'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9gatewayBT: 'home/OpenMQTTGateway/BTtoMQTT/'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9gatewayBT: 'gatewayBT'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9gatewayBT: 'B4E62DD9E0A9gatewayBT'
2019-03-08 17:35:42 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON B4E62DD9E0A9gatewayBT: '{{ value_json.message }}'

1technophile, 2019-03-10:
@indy55 you are using 0.88.2 ?

indy55, 2019-03-10:
Using 0.89.1

thewillwilson, 2019-03-18:
Also having the issue on 0.9.1beta

1technophile, 2019-03-19:
should be ok now on v0.9.1beta

closed 2019-03-19 by 1technophile