1technophile/OpenMQTTGateway #363: it seems not to receive radio signals

Maddmax76, 2019-02-25:
sorry if I'm missing something but it's my first time on GitHub, I hope not to break any rules.
I'm trying to interface the sensors of my alarm to openHAB. To do this I'm trying to use arduino with an RF sensor. I did a first test with an arduino nano and through the serial monitor I can read the radio signals.
The sensors send codes that I interpret with the RCSwitch.h library. In decimal, the signals are type XXXXYY, where XXXX indicates the sensor and YY indicates the command. The remote control with 4 keys varies YY according to the key, the PIR and the door opening sensors should change YY to indicate the opening of the contact or the presence detection, and the low battery status.

I tried with a LoLin NodeMcu V3, I followed the instructions:
Install Arduino IDE
From Arduino IDE board manager install ESP8266 version 2.4.2 (2.5.0 beta is not working yet)
Download the code with the files and put them on a folder named OpenMQTTGateway
Copy OpenMQTTGateway / lib to "sketchbook folder" / libraries
Open the OpenMQTTGateway.ino file

I modified the User_config.h file, I customized the connection to the wifi at home, I loaded everything on the nodemcu.

From the serial it says that it connects to the wifi and the MQTT broket. With a MQTT browser I see that the structures are created and some sample data arrives.

When the radio sensors are active nothing happens, neither on the MQTT broker nor in the serial console.

I thought the problem was bad wiring. I double checked, I thought that the ids of the pins do not coincide with those written on the card, I tried to connect the output of the RF receiver to the other digital pins of the nodemcu. When I plug in the IR-enabled pin, I see the unknown coding message in the serial console, so the RF module sends data to the nodemcu.

I do not understand what I'm wrong, could you help me?

Thank you so much, have a nice day

1technophile, 2019-02-25:
Hello, Could you post your user_config.h without credentials and the serial monitor output?

Maddmax76, 2019-02-26:
Hi Florian, thank you for your help. Yesterday evening, after reading your answer, I turned on the PC and wired the components back to the prototype board in order to get the information you requested. Everything started working perfectly. Probably previously, being the very old prototyping board, pin 3 of the nodemcu did not make contact with the track and did not hear the output of the RF receiver. Yesterday, the gateway started receiving signals and converting them to MQTT messages in the broker:
[image]
[log.txt.txt](https://github.com/1technophile/OpenMQTTGateway/files/2904367/log.txt.txt)

Pub json into:
home/OpenMQTTGateway/433toMQTT
{"value":13749552,"protocol":1,"length":24,"delay":472}

 Now I have to configure openHAB to read those messages and interpret them as door opening sensors or as remote controls. can you help me?

Thanks again, bye!

Maddmax76, 2019-02-27:
Hi, taking a look at this article:

https://1technophile.blogspot.com/2016/09/433tomqttto433-bidirectional-esp8266.html

I read that the radio device interfaced with the gateway sends a message to the broker with the remote control code only: 

home/433toMQTT 1312084 

Do I have to change some parameters or modify the code to manage only the received value and not all the other data?

is this why I can not create the radio control item in my openhab?

thanks bye

1technophile, 2019-02-27:
Hi, this is an old article, if you want up to date documentation I advise to go to the [wiki ](https://github.com/1technophile/OpenMQTTGateway/wiki)
So as to integrate your device into OpenHAB you have 2 ways:
the easiest by uncommenting simplePublishing into user_config.h
and following the examples [there](https://github.com/1technophile/OpenMQTTGateway/wiki/OpenHAB-integration)

And a second approach by letting OMG config as it is, and using json publishing unfortunately I have not yet put my hands on json for OpenHAB but if you search a little bit you should find some examples. If it is the case I will be happy to have your feedback so as to integrate these examples into the wiki

Maddmax76, 2019-02-27:
Hello Florian, thanks for the reply.

I would like to help you, but I must first start somatizing all these new
concepts.

Keep in mind that a week ago I did not know openHab, I did not know your
job, I did not know the MQTT protocol, I had a vague idea of domotics, of
json, of nodejs....

Everything started from the fact that the alarm at home is quite old.
This has RF pir sensors and anti-intrusion contacts. I had a nodemcu and an
arduino B+, I looked on the internet if there was a way to get an
indication of which sensor was triggered, because when a sensor with the
old control unit triggers the alarm but I do not know why.

Through google I started an escalation of research that led me to openhab
and your excellent project.

I'm one step away from succeeding ....

I followed the instructions in the OpenHAB integration document before
writing to the forum.
I tried sending messages to the broker mqtt from the command line, from a
client in a terminal I saw the messages arrive, but openhab did not notice
anything. Also acting on the button in the panel I saw no message in the
mqtt client.

Assuming that the messages that the gateway sends to the mqtt broker are:

{ "Value": 5726144, "protocol": 1, "length": 24, "delay": 516}
{ "Value": 5724096, "protocol": 1, "length": 24, "delay": 514}

and assuming that 5726144 is the key that activates the alarm and 5724096
is the key that disables it, I would like to create a switch type item that
shows me the alarm status, turning green when I press the 5726144 key and
gray when I press 5724096.

The broker is configured, I tried to change the broker serif on openhab and
from the command line I could no longer send or receive messages, so I was
using the internal openhab broker.

After setting up the broker what should I do in an empty openhab?

Thanks again for your kindness

Maddmax76, 2019-02-27:
.... the only thing I have not done is uncommenting simplePublishing into
user_config.h.

What does it do?

is it a kind of Homie convention for integration with the Home Automation
systems?

Thanks bye

Maddmax76, 2019-02-27:
found:

// # define simplePublishing true // comment if you do not want to use simple publishing (one topic for one parameter)
// example
// home / OpenMQTTGateway_ESP32_DEVKIT / BTtoMQTT / 4XXXXXXXXXX4 / rssi -63.0
// home / OpenMQTTGateway_ESP32_DEVKIT / BTtoMQTT / 4XXXXXXXXXX4 / servicedata fe0000000000000000000000000000000000000000

In practice, if activated, it makes a topic for each parameter.

How should this method simplify the creation of the device in openhab? I'm sorry if I'm stressing you, but I'm trying to dispel the fog that obscures my vision of all the infrastructure!

Maddmax76, 2019-02-28:
Good morning ...... I think someone has made me a slump! :-(
Yesterday evening I turned on the PC, connected the prototyping board with my nodemcu and the RF receiver above, which had worked the night before. I try to make it work but it does not receive any RF signal. I modified the parameter you spoke to me yesterday, but still connects to the wifi, connects to the broker mqtt but does not receive the RF signals. I tested the receiver on arduino nano and via serial I see the RF messages arrive, but on the nodemcu no. I have dismantled everything and reassembled at least 10 times, but there is no way to make it work. Maybe when the libraries update has blocked something?

I am attaching the configuration file to see if I'm wrong.

Thank you and good day

[seriallog.txt](https://github.com/1technophile/OpenMQTTGateway/files/2913831/seriallog.txt)

[User_config.txt](https://github.com/1technophile/OpenMQTTGateway/files/2913833/User_config.txt)

1technophile, 2019-03-01:
If it works with the nano with the same library (rcswitch) there is no reason the library update is the cause of this.

Did you try to change your cables?

As you have several questions may we go the community forum instead of github, as we are not pointing a defect here (I think)?
It will be easier to share screenshots, advices...

Maddmax76, 2019-03-02:
hi, you change the cables, but it's the same fault.
Today, however, I see a bit of fuss.
What is the community forum? What of OpenHab?

Thanks bye!

1technophile, 2019-03-02:
The forum is here:
https://community.openmqttgateway.com/

Regarding OpenHAB you have some examples of integration there:
https://github.com/1technophile/OpenMQTTGateway/wiki/OpenHAB-manual-integration

Maddmax76, 2019-03-04:
OK perfect. I move on the forum, thank you!

>

closed 2019-04-13 by 1technophile