1technophile/OpenMQTTGateway #364: Compile error v0.9 master

jeanniquini, 2019-02-25:
#define jsonPublishing true
#define jsonReceiving true
#define ZgatewayRF     "RF"   
#define ZgatewayBT     "BT"

[log.txt](https://github.com/1technophile/OpenMQTTGateway/files/2902193/log.txt)

1technophile, 2019-02-25:
Did you followed the troubleshooting section:
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

As the compilation is ok with travis:
https://travis-ci.org/1technophile/OpenMQTTGateway

This is likely related to you environment, and from what I see from the libraries that you are using in particular check the version of arduinojson

jeanniquini, 2019-02-25:
I had version 6.8.0-beta of arduinoJson. That was the problem.
Thanks.

closed 2019-02-25 by jeanniquini