1technophile/OpenMQTTGateway #367: Different IR codes - wrong store

kiralikbeyin, 2019-03-01:
I always get the first ir code i pushed {"value":4294967295,"protocol":3,"bits":0,"raw":"9072,2178,642"}

but real NEW code is not the same with short version above 
{"value":14635183,"protocol":3,"bits":32,"raw":"9052,4426,638,520,640,472,664,468,612,530,638,464,654,484,604,528,624,480,652,1620,622,1622,602,528,626,1616,604,1642,644,1590,650,1604,624,1634,650,488,602,1640,658,476,604,1632,652,480,610,528,624,480,652,488,602,1632,654,486,604,1632,656,486,604,1628,654,1620,622,1620,604,1632,654"}

...etc
{"value":14651503,"protocol":3,"bits":32,"raw":"9096,4420,642,484,606,506,648,480,652,458,634,530,620,486,646,486,604,502,650,1618,608,1620,668,458,626,1624,662,1620,626,1620,600,1640,646,1622,620,1622,602,534,618,482,654,1620,620,486,648,484,606,522,630,482,652,462,626,1636,650,1624,616,482,652,1620,620,1620,606,1636,650,1620,622"}

problem looks like giving a value to variable  https://github.com/1technophile/OpenMQTTGateway/issues/366

1technophile, 2019-03-01:
sorry but I don't understand the issue, what do you mean by "get" and "pushed"

What are you expecting to have?

kiralikbeyin, 2019-03-02:
Sorry i am very confused. 

I am pressing my ir remote and i see these in my mqtt broker:

{"value":14655583,"protocol":3,"bits":32,"raw":"9070,4440,624,484,648,484,600,538,618,484,650,486,602,512,640,466,668,484,602,1616,672,1620,622,462,672,1618,624,1618,602,1640,650,1602,640,1598,618,1642,652,482,598,1624,668,484,598,534,624,484,650,488,596,534,622,486,648,1600,642,484,648,1602,640,1618,600,1642,650,1600,640,1582,636"}
after this i suddenly see another message:
**{"value":4294967295,"protocol":3,"bits":0,"raw":"9088,2146,672"}**

another example:
{"value":14633143,"protocol":3,"bits":32,"raw":"9014,4478,570,586,566,542,624,504,628,508,598,512,616,514,598,510,624,502,598,1662,596,1662,618,516,616,1604,654,1608,648,1644,580,1638,594,1674,614,494,594,1648,636,482,640,488,648,1616,626,508,632,478,640,482,624,1626,642,470,594,1650,636,1630,620,512,636,1600,642,1628,592,1648,632"}
after this i suddenly see another message:
**{"value":4294967295,"protocol":3,"bits":0,"raw":"9048,2216,616"}**

I dont know why **{"value":4294967295,"protocol":3,"bits":0,"raw":"9048,2216,616"}**
came becuase this message has no meaning - 4294967295 value is not a code of my ir remote.

kiralikbeyin, 2019-03-02:
Please check this https://github.com/arendst/Sonoff-Tasmota/issues/5379

I can get IR codes without any hardware modification (OpenMQTTGateway) 

Maybe value 4294967295 is related with USB power...

1technophile, 2019-03-02:
and if you do the tip given (cut the link with usb power) do you still see this strange code?

kiralikbeyin, 2019-03-02:
Just tested now, i still get value 4294967295

````
Creating IR buffer
Rcv. IR
LED MNG
8956,2264,574
isAduplicate?
Adv data IRtoMQTT
Pub json into:
home/Evakilli/IRtoMQTT
{"value":4294967295,"protocol":3,"bits":0,"raw":"8956,2264,574"}
Hey I got a callback 
Storing signal
Creating Json buffer
Min ind: 
9
store code :
4294967295/210969
Col: val/timestamp
0:14659663/23574
1:14618863/35898
2:4294967295/49171
3:14622943/51596
4:14655583/54599
5:4294967295/186736
6:14643343/188955
7:14622943/205688
8:4294967295/206257
9:4294967295/210969
10:0/0
11:0/0
Data from JSON stored
/5987654321 -> 
Creating IR buffer
Rcv. IR
LED MNG
9004,2264,572
isAduplicate?
--don't pub. duplicate--
Creating IR buffer
Rcv. IR
LED MNG
8920,4640,446,588,526,622,508,606,502,606,536,602,530,602,504,602,542,590,530,1686,602,1682,532,588,530,1814,442,1708,562,1676,562,1716,522,1808,460,598,514,1708,558,1678,556,628,500,610,508,600,524,630,508,626,482,1722,554,594,500,622,508,1742,508,1724,542,1690,536,1752,502,1740,528
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
118,15014,8970,2272,572
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
8880,2304,530
isAduplicate?
--don't pub. duplicate--
Creating IR buffer
Rcv. IR
LED MNG
8990,2250,570
isAduplicate?
--don't pub. duplicate--
Creating IR buffer
Rcv. IR
LED MNG
9022,4436,628,586,574,526,610,528,562,574,572,516,616,536,550,586,570,538,596,1654,590,1640,590,574,578,1640,568,1700,588,1660,590,1644,572,1674,622,534,560,574,572,1646,570,1672,618,538,558,576,574,516,616,534,554,1668,612,1658,584,534,602,540,558,1658,618,1654,580,1658,562,1680,612
isAduplicate?
Adv data IRtoMQTT
Pub json into:
home/Evakilli/IRtoMQTT
{"value":14627023,"protocol":3,"bits":32,"raw":"9022,4436,628,586,574,526,610,528,562,574,572,516,616,536,550,586,570,538,596,1654,590,1640,590,574,578,1640,568,1700,588,1660,590,1644,572,1674,622,534,560,574,572,1646,570,1672,618,538,558,576,574,516,616,534,554,1668,612,1658,584,534,602,540,558,1658,618,1654,580,1658,562,1680,612"}
Hey I got a callback 
Storing signal
Creating Json buffer
Min ind: 
10
store code :
14627023/214464
Col: val/timestamp
0:14659663/23574
1:14618863/35898
2:4294967295/49171
3:14622943/51596
4:14655583/54599
5:4294967295/186736
6:14643343/188955
7:14622943/205688
8:4294967295/206257
9:4294967295/210969
10:14627023/214464
11:0/0
Data from JSON stored
/5987654321 -> 
Creating IR buffer
Rcv. IR
LED MNG
9048,2232,604,96200,9044,2244,598
--no pub. unknown protocol--
````

1technophile, 2019-03-02:
do you get this code with all ir remotes ?

kiralikbeyin, 2019-03-02:
Yes i can get with all ir remotes. 

My current code is 
//4294967295 will not be gone to mqtt or firebase  db
```
else if (!isAduplicate(MQTTvalue) && MQTTvalue != 0) { // conditions to avoid duplications of RF -->MQTT
      trc(F("Adv data IRtoMQTT"));
      pub(subjectIRtoMQTT, IRdata);

      String jsonStr;
      IRdata.printTo(jsonStr);

      if ( GENELIR != jsonStr && jsonStr != "4294967295" ) //4294967295 BU ÇÖP SAYI
      {
       cloudrequest(jsonStr, "5987654321", false); //firebase
        yield();
        GENELIR = jsonStr;
      }
      else
      {    delay(500);    } 
      /*
        trc(F("Store to avoid duplicate"));
        storeValue(MQTTvalue);
        if (repeatIRwMQTT) {
        trc(F("Pub. IR for repeat"));
        pub(subjectMQTTtoIR, MQTTvalue);
        }
      */
    } 
```

note that ir receiver is more sensitive than before.. i see these alot now :

```
Creating IR buffer
Rcv. IR
LED MNG
134,5310,272,720,178,686,180
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
184,152,116,1144,134
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
118,6510,264,2872,208,3930,194
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
232,2392,166,1036,230,5374,126,6126,224
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
104,3156,278,634,162
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
176,1746,152,226,116,4460,240
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
164,7244,170,900,250
--no pub. unknown protocol--
Creating IR buffer
Rcv. IR
LED MNG
166,5294,192,1686,400,740,214,1242,184
--no pub. unknown protocol--
```

1technophile, 2019-03-02:
did you tried to revert pins 4 and 5 and change the config_IR accordingly?

kiralikbeyin, 2019-03-02:
Yes i tried this swap but the same...  Maybe this is a special condition for SRFB and didnt tried with nodemcu..

I think i will put a if condition to the top like this 

````
  decode_results results;

  if (irrecv.decode(&results) && LearnOn == true ) {
    trc(F("Creating IR buffer"));
...
````

I need to store these IR codes in raw format becuase protocols are not reliable but IR raws always work. I will also store RF codes but they will not be in raw format. 

I will ask your help about storing codes, in eeprom-spiffs or firebase db... 

Thank you very much.

closed 2019-03-02 by kiralikbeyin