1technophile/OpenMQTTGateway #369: Add Pilight repeater function

codegrau, 2019-03-09:
Add repeater function to OMG similar to #define repeatRFwMQTT & #define repeatIRwMQTT for IR and calssic RF. 

[see discussion in forum](https://community.openmqttgateway.com/t/pilight-mqtt-repeater/373/3)

1technophile, 2019-03-27:
@codegrau could you do some test to see if it is the expected beahaviour?

closed 2019-03-27 by 1technophile

codegrau, 2019-03-27:
Hi @1technophile appreciate your restless work!

Just wondering: I don't see the following serial output via MQTT (neither receiving PitoMQTT nor sending MQTTtoPi topics). Would really love to see that via MQTT.

````
23:13:57.777 -> Creating RF PiLight buffer
23:13:57.777 -> Rcv. Pilight
23:13:57.777 -> Pub json into:
23:13:57.777 -> home/OMG_BETA/PilighttoMQTT
23:13:57.814 -> {"message":"{\"systemcode\":25,\"programcode\":1,\"state\":\"on\"}","protocol":"impuls","length":"","repeats":2,"status":2}
23:13:57.814 -> Pub Pilight for rpt
23:13:57.814 -> Pub json into:
23:13:57.814 -> home/OMG_BETA/commands/MQTTtoPilight
23:13:57.814 -> {"message":"{\"systemcode\":25,\"programcode\":1,\"state\":\"on\"}","protocol":"impuls","length":"","repeats":2,"status":2}
````

1technophile, 2019-03-28:
Hi,

You would like to see the logs into a mqtt topic?

codegrau, 2019-03-28:
OMG didn't publish neither receive nor command to mqtt... both ONLY in serial console and NOT via mqtt. Would be cool to know if this node did receive the message.

´´´´
home/OMG_BETA/PilighttoMQTT
home/OMG_BETA/commands/MQTTtoPilight
´´´´

1technophile, 2019-03-31:
This could be an issue with your pubsubclient library, did you get the one from lib folder of the omg repo ?

codegrau, 2019-04-01:
Thank you @1technophile - I did indeed use the wrong pubsubclient library. Solved!

codegrau, 2019-04-01:
@1technophile maybe it isn't as easy as I thought - guess you have to modify the input for proper output...

like changing "state":"on" to "on":"1" 
````
RECV: {"message":"{\"systemcode\":12,\"unitcode\":22,\"state\":\"on\"}","protocol":"elro_400_switch","length":"","repeats":2,"status":2}

SEND: {"message":"{\"systemcode\":12,\"unitcode\":22,\"on\":1}","protocol":"elro_400_switch","length":"","repeats":2,"status":2}
````

````
15:14:26.653 -> home/OMG_BETA/PilighttoMQTT
15:14:26.691 -> {"message":"{\"id\":0,\"unit\":6,\"state\":\"up\"}","protocol":"arctech_screen_old","length":"0","repeats":2,"status":2}
15:14:26.691 -> Pub Pilight for rpt
15:14:26.691 -> Pub json into:
15:14:26.691 -> home/OMG_BETA/commands/MQTTtoPilight
15:14:26.691 -> {"message":"{\"id\":0,\"unit\":6,\"state\":\"up\"}","protocol":"arctech_screen_old","length":"0","repeats":2,"status":2}
15:14:26.760 -> Hey I got a callback 
15:14:26.760 -> MQTTtoPilight json data analysis
15:14:26.760 -> MQTTtoPilight msg & protocol ok
15:14:26.760 -> pilight(3): arctech_screen_old: insufficient number of arguments
15:14:26.760 -> message is invalid
````

1technophile, 2019-04-01:
Yes I could make some code to do that but maybe the proper way would be to do it at library level.
I think we should ask @puuu if he is interested in making the sending and receiving homogeneous.
@puuu what would yours thoughts on that?

puuu, 2019-04-08:
If you go a layer lower to pulsetrains instead of json messages, the implementation of this feature should be easy and already possible from the library side. You can register a callback function for pulse trains with `ESPiLight::setPulseTrainCallBack()` (see also the [Receive_Raw example](https://github.com/puuu/ESPiLight/blob/master/examples/Receive_Raw/Receive_Raw.ino)). The pulsetrain can be retransmitted with `ESP::sendPulseTrain()`. I have never tried to do something like this, but it may be necessary to introduce a small delay before retransmitting.

1technophile, 2019-04-17:
Thanks for the proposal, I will take a look!