1technophile/OpenMQTTGateway #373: On demand BL Scans.

maxi1134, 2019-03-17:
**Is your feature request related to a problem? Please describe.**
I think that scanning for BL devices at all time is like using a shotgun, and that having a way of requesting scans on demand could alleviate the network charge. Per instance using a MQTT feed as a trigger. This coupled with motion detection would be a great way of reducing the mqtt jitter.

**Describe the solution you'd like**
OpenMQTTGateway could wait for a command before scanning, the scan for X time(s).

**Describe alternatives you've considered**
I couldn't find anything like this in the doc. Or perhaps my poor english made me misunderstood if there is this function.

kabturek, 2019-04-25:
This is a good idea - for me (just for presence detection) i would run the scans when my door is opened as thats the only time when the number of BT devices could change.

1technophile, 2019-04-28:
done in dev branch, instructions are in the commit msg

closed 2019-04-28 by 1technophile