1technophile/OpenMQTTGateway #375: MQTTtoRFM69 not working when receiver included

aussieW, 2019-03-31:
Hi

I am trying to send an MQTTtoRFM69 message to a receiver that is not the defaultRFM69ReceiverId. 
e.g. 
`mosquitto_pub -t "home/OpenMQTTGateway/commands/MQTTtoRFM69/RCV_98" -m "Test"`

I am finding that it is not working and I suspect that the message is not sent because the following condition is not met
`if (topic == subjectMQTTtoRFM69) {`

As far as I can tell this results because 
`subjectMQTTtoRFM69 = home/OpenMQTTGateway/commands/MQTTtoRFM69`
and
`topic = home/OpenMQTTGateway/commands/MQTTtoRFM69/RCV_99`

It appears to be a problem for both simpleReceiving and jsonReceiving. Although in the case of jsonReceiving the test is expressed as
`if (strcmp(topicOri,subjectMQTTtoRFM69) == 0){`

I am not sure why the test is different in the two functions. Would it be better/more consistent if they were the same?

Cheers

1technophile, 2019-03-31:
Hi,

Indeed we should not have this test:
` if (topic == subjectMQTTtoRFM69) {`
in case of simple receiving. I will correct it.

Nevertheless you should be able to pass a "receiverid" in json format. 

`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoRFM69 -m '{"data":1315156,"receiverid":34}'`

aussieW, 2019-03-31:
Okay, thanks. I wasn't aware that I could pass the receiverid in the JSON payload. Tested it and it seems to work.

1technophile, 2019-03-31:
This was the opportunity to update the wiki, thanks for pointing this

closed 2019-04-13 by 1technophile