1technophile/OpenMQTTGateway #377: question re .bin file for ESP32

zala143, 2019-04-03:
Hi,
is there a .bin file version of OpenMQTTGateway that I can simply flash onto an ESP32?

I'm not too comfortable mucking with Arduino sketches so hoping there's something precompiled i can use. I'm looking for BLE support only.  I found the links to the [releases] page but couldnt tell which file to use?? (https://github.com/1technophile/OpenMQTTGateway/releases)

I saw a file named ESP8266_OpenMQTTGateway_V09_IRRFBT.ino.bin.zip but wasnt sure if that is only for ESP8266 boards or would that work for ESP32?? .... that was a zipped file containing "OpenMQTTGateway_V09_IRRFBT.ino.bin"

Many thanks in advance for all your great work.

akasma74, 2019-04-03:
I can see no suitable binaries on the release page, but it's easy to download sources, change config in Arduino and click Upload.
All the instructions are on wiki.

zala143, 2019-04-03:
ok thanks, will try that - so used to dropping tasmota .bin files on esps :) been spoiled.

zala143, 2019-04-04:
just opening up to see anyone has a compiled .bin file for ESP32? Thanks

akasma74, 2019-04-04:
Were you unable to compile?
Any error log? ;)

1technophile, 2019-04-12:
the issue is that ESP32 doesn't support wifimanager yet, even if i give you a bin file you will not be able to log to your wifi network :-(

closed 2019-04-13 by zala143

zala143, 2019-04-13:
Ok. Thanks for confirming, I'll dive in and try Arduino ide. Thanks