1technophile/OpenMQTTGateway #378: Offline after Home assistant Restart

SurfingSpider, 2019-04-07:
Is there a way of restarting OpenMqttGateway via an MQTT command? Each time I have to do a  restart of my Home assistant RPI, The ESP32 running OpenMqttGateway stops sending messages to MQTT (which is running as part of Hassio). I have to go and unplug and replug the ESP32 device to get it to start publishing BLE data from my Mi flora's. Looking at the incoming stream of data coming into mosquito it looks like the openmqttgateway just stops publishing. I see the LWT is offline

Is there a way to get it to retry and connect every 10 minutes or so? or a MQTT message that I could rend to do the restart instead, so I can automate it?

P.S. I tried the home assistant auto discovery new feature in the current dev. It added the sensors to HA great. But I had problems as each time one of the sensors of a mi flora plant was published the other sensors for that plant are zeroed.  Which gave me lots of errors for my watering automations. I had to add the sensors manually and add unhash `#define simplePublishing true` as layed out in the wiki. I am happy it is working just though I would give some feedback that I am not sure the current Auto discovery is quite dialed in yet. Thanks for all the hard work on OpenMQttGateway. It rocks!

1technophile, 2019-04-12:
> Each time I have to unplug my Home assistant RPI, The ESP32 running OpenMqttGateway stops sending messages to MQTT
Are you sure it is OMG related, as OMG is decoupled to HASS thanks to the mqtt broker who plays the role of a middle man, I don't see why the restart of HASS impact OMG.

If you restart HASS and monitor your broker could you check what is happening?

> P.S. I tried the home assistant auto discovery new feature in the current dev. It added the sensors to HA great. But I had problems as each time one of the sensors of a mi flora plant was published [...]

You mean if temperature is published, moisture will be zeroed?

> Thanks for all the hard work on OpenMQttGateway. It rocks!

Glad to hear that you like it

SurfingSpider, 2019-04-16:
My mqtt broker that I use is built into hassio. So when repowering the hardware it goes offline. When it comes back online it says it is online and version 9 but nothing else happens after that. Untill I repower the omg. 

You mean if temperature is published, moisture will be zeroed? That is correct. Technically is is set to null not zero

1technophile, 2019-05-04:
I need to see what happens if I restart my broker and by the way try to reproduce your issue

PDekker, 2019-05-09:
I have the same issue.

Openmqtt needs to be restarted after a Hass reboot.

Also, if I receive a temperature update, moisture etc. will be zeroed

1technophile, 2019-05-15:
@PDekker are you using also the embedded HA broker?

PDekker, 2019-05-15:
I’m using mosquito on a Hassbian installation on a raspberry pi 3.

I’ve done some testing myself. I think is has to do with the WiFi signal.

Still have the issue with other sensors being zeroed on a update of one of the sensors. Also not receiving battery life updates

kabturek, 2019-05-16:
@PDekker if you get undefined value (and using json templates) you have to use the is_defined filter in the value_template

```
    - platform: mqtt
      name: miflora_moisture
      expire_after: 21600 # 6 hours
      state_topic: "home/OpenMQTTGateway/BTtoMQTT/--SNIP--"
      value_template: "{{ value_json.moi | is_defined}}"
      availability_topic: "home/OpenMQTTGateway/LWT"
```

@1technophile created https://github.com/1technophile/OpenMQTTGateway/pull/403 with the change

PDekker, 2019-05-17:
Thats a great suggestion. Unfortunately I’m using MQTT discovery. Will definitely try a manual config when I’m back from holiday.

1technophile, 2019-06-21:
Should be OK now in dev branch

closed 2019-06-21 by 1technophile