1technophile/OpenMQTTGateway #380: Password encryption trouble

puyoo, 2019-04-12:
HI

Im using rasperry PI witch mosquitto   1.4.10  winth a sonoff bridge  openmqttserver  0.9 

precompiled firmware from (https://github.com/1technophile/OpenMQTTGateway/releases/download/v0.9/SONOFF_RFBRIDGE_OpenMQTTGateway_V09.ino.bin.zip

mosquitto.conf 
password_file /etc/mosquitto/mot
allow_anonymous false

user.txt

puyo: XXXXXXXX (encrupted hash pass)

 evry time i try to connect witch my sonoff bridge  i get this message 

1555099910: New connection from 192.168.1.100 on port 1883.
1555099910: Sending CONNACK to 192.168.1.100 (0, 5)

if i try witch  MQTT.FX  its working fine 

if i alow anonymous ilts ok 

there is not encryption activated  by default in the firmware ?
there is a way to disable hash encryption in mosquitto ?

thanks

Daniel

1technophile, 2019-04-12:
Hi,

> there is not encryption activated by default in the firmware ?
Encryption is not activated

> there is a way to disable hash encryption in mosquitto ?
yes you can deactivate SSL/TLS
https://mosquitto.org/man/mosquitto-conf-5.html
But it is deactivated per default

I suppose that you are entering mqtt credential into the web portal isn't it?

puyoo, 2019-04-12:
i entering  the credintial MQTT   in the  web page from the wifimanager  (192.168.4.1) openMQTTgateway). i dont activate  setting for encryption

if i Try plain-text password   i get this 

1555101216: Config loaded from test.conf.
1555101216: Error: Invalid password hash for user puyo, removing entry.

any way to force to accept  plain-text password ?

1technophile, 2019-06-21:
Hello,

Did you found a solution to your issue?

puyoo, 2019-08-21:
hi
yes i use now espurna

closed 2019-10-27 by stale[bot]