1technophile/OpenMQTTGateway #383: Crash when trying to transmit IR

bphermansson, 2019-04-16:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
A clear and concise description of what the bug is.

Esp8266 crashing when trying to transmit IR. 

**To Reproduce**
Steps to reproduce the behavior:
1. Publish to '...'
2. Subscribe on '....'
3. See error

I use this command to get my device to transmit an IR-command:
mosquitto_pub -h 192.168.1.79 -u '***' -P '***' -t home/OMG_Irext/commands/MQTTtoIR -m '{"value":1587632295,"protocol":3,"bits":32,"raw":}'

In the serial port the Esp responds with a crash:

```
Hey I got a callback 
MQTTtoIR json
MQTTtoIR data || raw ok
value
1587632295
raw

Exception (28):
epc1=0x4022ce1c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>
```

With a shorter Mqtt-message the Esp doesn't crash. 

**Expected behavior**
A clear and concise description of what you expected to happen.

No crash.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.4, 0.3, master...)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

OMG 0.9.1, fresh from Github today. 

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2019-04-16:
Hello,

Maybe try to send your command without the empty raw field : 
'{"value":1587632295,"protocol":3,"bits":32}'

1technophile, 2019-05-04:
Hello, I close the issue, if you have some feedback don't hesitate to add a comment

closed 2019-05-04 by 1technophile

cometfish, 2019-06-28:
Hi, I am also experiencing the same issue with sending raw IR messages. However mine doesn't have an empty raw field.
I have tried on two different types of esp8266 boards (a nodemcu and a sparkfun thing), and also checked I am using the included libraries from openmqtt (0.9.1).

Here's what I sent:
`mosquitto_pub -t home/OpenMQTTGateway2/commands/MQTTtoIR -m '{"value":4294967295,"protocol":3,"bits":0,"raw":"9088,2110,686"}'`

(I watched the IR LED through a camera lens, and it does start flashing... but the signal doesn't seem to work even when directly next to my TV, so I think it must be cutoff mid-send by the crash)

Here is the message displayed on the serial console.
```
Exception (28):
epc1=0x4000e1e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0290 end: 3fff0780 offset: 01a0

>>>stack>>>
3fff0430:  00000001 3fff1b0b 3fff04c0 40204fe0  
3fff0440:  3fff04c0 3ffe89e8 00000000 00000003  
3fff0450:  00000000 3fff0480 00000001 402042b4  
3fff0460:  4010441e 00000025 7fffffff 40203c9c  
3fff0470:  00000002 0000000d 00000000 40204360  
3fff0480:  00000002 00000000 3ffece40 3fff150c  
3fff0490:  00000000 4021d920 3ffeef04 3ffeef3b  
3fff04a0:  3fff04c0 3ffeef13 3fff1ae4 40205560  
3fff04b0:  3ffe8ac0 3fff04c0 00000200 00000048  
3fff04c0:  3fff04b0 3fff04c8 3fff04d8 3fff1ae4  
3fff04d0:  00000001 3fff1aea 3fff04e8 3fff1af5  
3fff04e0:  00000001 3fff1afe 3fff04f8 3fff1b00  
3fff04f0:  00000001 3fff1b05 00000000 3fff1b07  
3fff0500:  00000002 3fff1b0b 3fff114c 3ffece68  
3fff0510:  00000008 00000000 00000020 40102006  
3fff0520:  3ffe96d8 40101cb1 00000002 3fff153f  
3fff0530:  00000005 00000000 00000020 40102006  
3fff0540:  00000005 00000000 00000020 40102006  
3fff0550:  00000005 00000000 00000020 40102006  
3fff0560:  3ffe96d5 401052e3 3ffed160 40102ea8  
3fff0570:  40102d19 3ffed160 00000020 40102006  
3fff0580:  3ffed850 40101b9a 00000001 00000000  
3fff0590:  00000000 40101967 00000000 3ffed820  
3fff05a0:  00000001 00000000 00000002 40102006  
3fff05b0:  00000005 00000000 00000020 40102006  
3fff05c0:  3ffe96d5 401052e3 3ffed160 00000100  
3fff05d0:  40102d19 3ffed160 00000200 4000050c  
3fff05e0:  00007fff 02111bc5 3ffedb88 40102ea8  
3fff05f0:  3ffe9f68 00000000 00000000 00000000  
3fff0600:  00000000 02111bc5 401032e6 00000100  
3fff0610:  7fffffff 3ffe9f68 3ffe9f68 00000001  
3fff0620:  00000001 00007fff 00000000 4000050c  
3fff0630:  3fff0874 0000018b 0000018b 4010020c  
3fff0640:  00000000 3ffe93ec 00000001 3fff0874  
3fff0650:  00000020 3fff06e0 00000000 4010053d  
3fff0660:  402325aa 00000001 3ffe8ed5 402141c4  
3fff0670:  3fff1cbc 00000028 00000000 3ffeef3b  
3fff0680:  00000040 00000015 3ffef690 40212855  
3fff0690:  3ffe8ed4 00000015 3fff06e0 3ffeef3b  
3fff06a0:  3fff0874 00000150 00000150 4010020c  
3fff06b0:  00000040 3ffeef13 3fff06e0 3ffeef3b  
3fff06c0:  3fff04b0 3fff1b24 3fff1b19 4010040a  
3fff06d0:  3fff1b24 3ffeef13 3fff1ae4 4020560a  
3fff06e0:  00000000 00000000 00000000 40213090  
3fff06f0:  3fff0720 00000001 3ffef304 00000027  
3fff0700:  00008774 3ffef304 3ffeef04 40213a5c  
3fff0710:  00008774 3ffef304 3ffeef04 402060d0  
3fff0720:  00000001 3ffe8540 00000000 40203a31  
3fff0730:  00000001 00000004 3ffeef07 0000006b  
3fff0740:  3ffe8ec8 3800a8c0 feefeffe 3ffef74c  
3fff0750:  00008774 00000001 3ffeef04 402057b0  
3fff0760:  3fffdad0 00000000 3ffef744 40213884  
3fff0770:  feefeffe feefeffe 3ffef760 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

1technophile, 2019-06-30:
`mosquitto_pub -t home/OpenMQTTGateway2/commands/MQTTtoIR -m '{"value":4294967295,"protocol":3,"bits":0,"raw":"9088,2110,686"}'`

It is strange to have a bit value to 0, could you try:
'{"value":4294967295,"protocol":3}'`

cometfish, 2019-06-30:
Yes leaving out both the `bits` and `raw` values does not crash:
`'{"value":4294967295,"protocol":3}'`
Unfortunately I'm not near my TV at the moment, so I don't know if the IR code actually works without those fields - will update on this later today.

Leaving out just the `bits` part still crashes:
`'{"value":4294967295,"protocol":3,"raw":"9088,2110,686"}'`

cometfish, 2019-07-01:
Just realised I've previously recorded the values for several buttons from my remote - they all have the exact same `value` and `protocol` fields, so the TV won't be able to distinguish which button was pressed without the `raw` field:
```
key_on : {"value":4294967295,"protocol":3,"bits":0,"raw":"9012,2190,606"}
key_mute : {"value":4294967295,"protocol":3,"bits":0,"raw":"9062,2140,658"}
key_volumeDown : {"value":4294967295,"protocol":3,"bits":0,"raw":"9090,2112,686"}
key_channelDown : {"value":4294967295,"protocol":3,"bits":0,"raw":"9092,2110,686"}
key_off : {"value":4294967295,"protocol":3,"bits":0,"raw":"9012,2192,606"}
key_up : {"value":4294967295,"protocol":3,"bits":0,"raw":"9090,2114,684"}
key_down : {"value":4294967295,"protocol":3,"bits":0,"raw":"9092,2112,686"}
key_volumeUp : {"value":4294967295,"protocol":3,"bits":0,"raw":"9064,2140,684"}
key_info : {"value":4294967295,"protocol":3,"bits":0,"raw":"9036,2166,658"}
key_source : {"value":4294967295,"protocol":3,"bits":0,"raw":"9090,2114,684"}
key_channelUp : {"value":4294967295,"protocol":3,"bits":0,"raw":"9090,2112,686"} 
```

cometfish, 2019-07-11:
Saw that you updated IRRemote on the development branch, so I had another try with that latest (0.9.2beta at rev 24bcdbc).

Unfortunately it still crashes if I include the `raw` value, regardless of `bits` or `value` field.

I had a look through the code, and noticed it was expecting a `protocol_name` field.
This one doesn't crash:
`{"value":4294967295,"protocol":3,"bits":0,"raw":"9088,2110,686","protocol_name":"IR_Raw"}`

(it also doesn't work to control my TV, but that's probably an issue at my end :) )

1technophile, 2019-07-12:
You should not add a protocol number or value if using raw.
{,"raw":"9088,2110,686","protocol_name":"IR_Raw"}

See there:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-IR-09#send-data-by-mqtt-with-advanced-ir-parameters

cometfish, 2019-07-12:
Yes, I don't think that was an IR_Raw code, and that's why it didn't work with the tv.
I since recorded the button again with IR_Raw enabled, and got a much longer code that works. (The longer code still had protocol number and value set, and still worked, so that wasn't the issue.) 

So there appears to be two different issues:
One is openmqtt crashes if you don't send `protocol_name`.
Two is when receiving an IR code, `protocol_name` is not set, so you don't know what protocol to send when you try and send it back yourself.

It should be possible to record a code via IRtoMQTT and then send it straight back via MQTTtoIR. At the moment that doesn't work.

1technophile, 2019-07-12:
Indeed currently there is a protocol mapping:
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1910001295

I agree that it is not the best way to handle this.