1technophile/OpenMQTTGateway #384: IR json datastring format

Legion2, 2019-04-21:
Some of the MQTTtoIR protocols don't require a "value" property in the json request, they require more data. For example the "FujitsuAC" requires the data as a array of bytes. Currently the json property "datastring" is used to pass this data. The Problem is, that the raw byte representation of the "datastring" is used as input data to the `irsend` function. This makes it difficult to send the byte data correctly. For example we want to send the sequence of bytes: `0x14 0x63 0x00 0x10 0x10 0x02` to do this we have to convert the bytes to characters so we can send them in a json string. The resulting string is `c\0`, i could not even past the correct string here because it contains a null character. I had to escape it to `\0` but this is now the wrong string. The string represent the byte sequence `0x14 0x63 0x5C 0x30 0x10 0x10 0x02`, this is because the null character is not unescaped by any one. The same applies when try to send this command with OMG, the ArduinoJson library can't handle strings which contains null characters, means I can not send any command containing the byte `0x00`.
But that's not the only problem, like you can see, you can not see the string, because it contains not printable characters.

To solve this problems the "datastring" should contain only printable characters and encode the byte array in a readable form. e.g. `14 63 00 10 10 02` or `14,63,0,10,10,02`. Like the "raw" string, the "datastring" must then converted to a byte array and used as argument for `irsend`.

An alternative is to support `\x` hex escape in the "datastring", like `\x14c\x00\x10\x10\x02`

1technophile, 2019-04-22:
Thanks for pointing that out

Legion2, 2019-04-22:
I can do a pull request

1technophile, 2019-04-22:
It would be great !

PR #388 (2019-04-22): added IR json datastring hex decoding
This PR fixes #384
The format is now a comma separated list of hex values.
e.g. `14,63,0,10,10,02,FD` in the json it looks like this: `{"datastring":"14,63,0,10,10,02,FD","bits":7,"protocol_name":"IR_FUJITSU_AC","value":1}`
The "bits" must be the count of hex values in the "datastring".

This breaks old datastring usage and users must change the "datastring".
closed 2019-04-22 by 1technophile