1technophile/OpenMQTTGateway #387: Pilight json message has bad string encoding

jaakla, 2019-04-22:
**Describe the bug**
PilighttoMQTT message format is not recognized by json parsers, e.g. home-assistant. The `message` field JSON, but not for some reason escaped as json string. It is quite hard to even notice this issue, I spent literally hours troubleshooting why home-assistant does not read the data properly.

Notice that message contents is not really json object here, but just a string value of "message" top level element.
```
{
 "message":"{\"id\":186,\"temperature\":22.90,\"humidity\":10.00,\"battery\":1,\"channel\":1}",
 "protocol":"tfa",
 "length":"186",
 "repeats":2,
 "status":2
}
```

**To Reproduce**
Listen messages over Pilight connection. See also #355 .

**Expected behavior**
Proper JSON format, with sample above
```
{"message":
  {"id":186,
   "temperature":22.90,
   "humidity":10.00,
   "battery":1,
  "channel":1},
 "protocol":"tfa",
 "length":"186",
 "repeats":2,
 "status":2
}
```

**Environment (please complete the following information):**
* OpenMQTTGateway  v0.9.1
* Digoo dgr8s , but seems to happen with other devices too

jaakla, 2019-04-23:
As temporary workaround I changed in _User_config.h_: 
```
#define simplePublishing true //comment if you don't want to use simple publishing (one topic for one parameter)
```
This fixes the Pilight messages. Still it is not a final fix, as this way I'm making many other messages hard/impossible to process. For example now I have following set of BTLE messages (from 3 devides I believe) which are very difficult/impossible to make usable as messages from different devices are messed up:
```
home/home_presence/OpenMQTTGateway/id 5e:0b:62:21:6d:5f
home/home_presence/OpenMQTTGateway/manufacturerdata L
home/home_presence/OpenMQTTGateway/rssi -58
home/home_presence/OpenMQTTGateway/distance 0.8
home/home_presence/OpenMQTTGateway/id 70:19:e5:38:60:e9
home/home_presence/OpenMQTTGateway/manufacturerdata L
home/home_presence/OpenMQTTGateway/rssi -75
home/home_presence/OpenMQTTGateway/distance 5.8
home/home_presence/OpenMQTTGateway/id a0:99:9b:1d:05:ae
home/home_presence/OpenMQTTGateway/manufacturerdata L
home/home_presence/OpenMQTTGateway/rssi -58
home/home_presence/OpenMQTTGateway/distance 0.8
```

jesperuk, 2019-04-29:
I get the same with pilight, had to run it thru nodered to convert it to Json....

1technophile, 2019-04-30:
Should be better now, with v0.9.2beta

closed 2019-04-30 by 1technophile