1technophile/OpenMQTTGateway #389: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

opl-git, 2019-04-23:
Hello everyone,

I found this project and really like to try it.
I used the guide [https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP32](url)
Unfortunately there is a problem when compiling it for my esp32:
**error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6**

I am using Arduino 1.8.9 on a windows 10 machine.

Hope someone can help me,
kind regards
Daniek

opl-git, 2019-04-23:
Downgraded to ArduinoJson 5.X ... solved the problem.

1technophile, 2019-04-23:
Hmm, you forgot this step I think : 
8. Copy OpenMQTTGateway/lib to "sketchbook folder"/libraries

opl-git, 2019-04-24:
Thank you for your answer!
No, I copied all the libaries from OpenMQTT (A6lib...Wifi Manager) in my sktechbook folder under libaries.

1technophile, 2019-04-24:
And even arduinojson library that is located in this folder with the good version ?

opl-git, 2019-04-24:
Yes, I copied all the libaries from OpenMQTT to my sketchbook libary folder.
Maybe the arduinojson library got updated somehow, because when I downgraded it to version 5.13, everything works fine.

closed 2019-04-27 by 1technophile

thomasnordquist, 2019-04-27:
@1technophile Maybe take a look at [PlatformIO](https://platformio.org/), it provides library version management, isolates your project from the shared libraries of the Arduino IDE.
It's also great for automated builds/managing many different build targets.

1technophile, 2019-04-27:
Yep, need to put this on my todo, but I'm asking myself if going to platform.io will not make things more difficult for Arduino IDE traditional users. Maybe just a wrong view.

thomasnordquist, 2019-04-27:
PlatformIO is pretty easy to set up for Linux and MAC, the setup is a little more "complicated" for Windows but the Atom IDE Plugin guides you through the setup procedure.
Most people in home automation might already know it from [Tasmota](https://github.com/arendst/Sonoff-Tasmota)

[https://github.com/MarlinFirmware/Marlin](Marlin) supports platformio/arduino hybrid.

I never cared much for the Arduino IDE, I mostly use it for the Serial Monitor.
The [PlatformIO IDE](https://platformio.org/platformio-ide) is very powerful, with autocompletion and compile/lint-as-you-type.

You might get rid of your `lib/` directory, get a powerful IDE and have versioned packages.

I believe you could get started with platformio by simply adding a `platformio.ini` with no other changes to the project at all, retaining compatibility with the Arduino IDE.

1technophile, 2019-04-27:
Thanks for the infos, if I can keep both worlds working with the same repo it would be a nice progress. It seems to be time to put my hands on it!