1technophile/OpenMQTTGateway #390: Multiple RF gateways - duplicated MQTT messages

akasma74, 2019-04-24:
**Describe the bug**
I've recently added a second RF gateway to my setup and now there is one Arduino UNO with Ethernet shield (A, name: OpenMQTTGateway) and one Wemos D1 mini (W, name: OMG_2nd_floor).
The A gateway used to run 0.9.1beta, but I uploaded it with the same 0.9.1 release as W to have as similar code as possible to compare.
The issue is there is no de-duplication of MQTT messages, each gateway sends its message on receiving a RF signal.

**To Reproduce**
I used one of sensors, just opened and then closed it.
Here is the log of MQTT messages received (3 times of open/close sequences):

> 09:52:27: (0) home/OpenMQTTGateway/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":455}
> 09:52:27: (0) home/OMG_2nd_floor/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":455}
> 09:52:29: (0) home/OpenMQTTGateway/433toMQTT {"value":7295365,"protocol":1,"length":23,"delay":452}
> 
> 09:52:32: (0) home/OpenMQTTGateway/433toMQTT {"value":14590734,"protocol":1,"length":24,"delay":452}
> 09:52:32: (0) home/OMG_2nd_floor/433toMQTT {"value":14590734,"protocol":1,"length":24,"delay":452}
> 09:52:33: (0) home/OpenMQTTGateway/433toMQTT {"value":7295367,"protocol":1,"length":23,"delay":454}
> 
> 
> 
> 09:53:18: (0) home/OpenMQTTGateway/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":454}
> 09:53:18: (0) home/OMG_2nd_floor/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":453}
> 09:53:19: (0) home/OpenMQTTGateway/433toMQTT {"value":7295365,"protocol":1,"length":23,"delay":453}
> 09:53:20: (0) home/OMG_2nd_floor/433toMQTT {"value":511,"protocol":13,"length":8,"delay":61}
> 
> 09:53:26: (0) home/OpenMQTTGateway/433toMQTT {"value":14590734,"protocol":1,"length":24,"delay":452}
> 09:53:26: (0) home/OMG_2nd_floor/433toMQTT {"value":14590734,"protocol":1,"length":24,"delay":452}
> 09:53:27: (0) home/OpenMQTTGateway/433toMQTT {"value":7295367,"protocol":1,"length":23,"delay":454}
> 
> 
> 
> 09:53:54: (0) home/OpenMQTTGateway/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":455}
> 09:53:54: (0) home/OMG_2nd_floor/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":455}
> 09:53:55: (0) home/OpenMQTTGateway/433toMQTT {"value":7295365,"protocol":1,"length":23,"delay":452}
> 
> 09:53:59: (0) home/OpenMQTTGateway/433toMQTT {"value":14590734,"protocol":1,"length":24,"delay":452}
> 09:53:59: (0) home/OMG_2nd_floor/433toMQTT {"value":14590734,"protocol":1,"length":24,"delay":452}
> 09:54:00: (0) home/OpenMQTTGateway/433toMQTT {"value":7295367,"protocol":1,"length":23,"delay":454}

**Expected behavior**
There should be only one mqtt message on each RF signal from either A or W gateway.

**Environment (please complete the following information):**
* OpenMQTTGateway version 0.9.1

**Additional context**
It behaved exactly the same when I used 0.9.1beta on my A gateway.

On a slightly different topic, I noticed that the A gateway **always** adds a second message with some strange code and wonder if I it's a bug as I can see nothing like that with the W gateway.

1technophile, 2019-04-25:
Would it be possible to connect OMG_2nd_floor to the serial monitor and see if it receives the callback from OpenMQTTGateway before receiving the signal?

akasma74, 2019-04-26:
Here is the log:
> 13:24:28.935 -> WiFi ok with manual config credentials
> 13:24:28.935 -> OpenMQTTGateway mac: 
> 13:24:28.935 -> 68:C6:3A:EA:22:88
> 13:24:28.935 -> OpenMQTTGateway ip: 
> 13:24:28.935 -> 192.168.0.XX
> 13:24:28.935 -> 1883
> 13:24:28.935 -> Connecting to MQTT by IP adress
> 13:24:28.935 -> 192.168.0.XX
> 13:24:29.877 -> RF_EMITTER_PIN 
> 13:24:29.877 -> 3
> 13:24:29.877 -> RF_RECEIVER_PIN 
> 13:24:29.877 -> 0
> 13:24:29.877 -> RF setup ok
> 13:24:29.877 -> MQTT_MAX_PACKET_SIZE
> 13:24:29.877 -> 1024
> 13:24:29.877 -> Setup OpenMQTTGateway end
> 13:24:32.818 -> MQTT connection...
> 13:24:34.761 -> Connected to broker
> 13:24:34.761 -> Subscription OK to the subjects
> 13:24:49.163 -> Rcv. RF
> 13:24:49.163 -> isAduplicate?
> 13:24:49.163 -> Pub json into:
> 13:24:49.163 -> home/OMG_2nd_floor/433toMQTT
> 13:24:49.163 -> {"value":8895854,"protocol":1,"length":24,"delay":306}
> 13:24:49.163 -> Store to avoid duplicate
> 13:24:49.163 -> Min ind: 
> 13:24:49.163 -> 0
> 13:24:49.163 -> store code :
> 13:24:49.163 -> 8895854/21341
> 13:24:49.163 -> Col: val/timestamp
> 13:24:49.163 -> 0:8895854/21341
> 13:24:49.163 -> 1:0/0
> 13:24:49.163 -> 2:0/0
> 13:24:49.163 -> 3:0/0
> 13:24:49.163 -> 4:0/0
> 13:24:49.163 -> 5:0/0
> 13:24:49.163 -> 6:0/0
> 13:24:49.163 -> 7:0/0
> 13:24:49.163 -> 8:0/0
> 13:24:49.198 -> 9:0/0
> 13:24:49.198 -> 10:0/0
> 13:24:49.198 -> 11:0/0
> 13:24:49.198 -> Rcv. RF
> 13:24:49.198 -> isAduplicate?
> 13:24:49.198 -> --don't pub. duplicate--
> 13:24:49.265 -> Hey I got a callback 
> 13:24:49.265 -> Store signal
> 13:24:49.265 -> Min ind: 
> 13:24:49.265 -> 1
> 13:24:49.265 -> store code :
> 13:24:49.265 -> 8895854/21408
> 13:24:49.265 -> Col: val/timestamp
> 13:24:49.265 -> 0:8895854/21341
> 13:24:49.265 -> 1:8895854/21408
> 13:24:49.265 -> 2:0/0
> 13:24:49.265 -> 3:0/0
> 13:24:49.265 -> 4:0/0
> 13:24:49.265 -> 5:0/0
> 13:24:49.265 -> 6:0/0
> 13:24:49.265 -> 7:0/0
> 13:24:49.265 -> 8:0/0
> 13:24:49.265 -> 9:0/0
> 13:24:49.265 -> 10:0/0
> 13:24:49.265 -> 11:0/0
> 13:24:49.265 -> Data JSON stored
> 13:24:49.458 -> Hey I got a callback 
> 13:24:49.458 -> Store signal
> 13:24:49.458 -> Min ind: 
> 13:24:49.458 -> 2
> 13:24:49.458 -> store code :
> 13:24:49.458 -> 8895854/21645
> 13:24:49.458 -> Col: val/timestamp
> 13:24:49.458 -> 0:8895854/21341
> 13:24:49.458 -> 1:8895854/21408
> 13:24:49.458 -> 2:8895854/21645
> 13:24:49.458 -> 3:0/0
> 13:24:49.458 -> 4:0/0
> 13:24:49.458 -> 5:0/0
> 13:24:49.458 -> 6:0/0
> 13:24:49.458 -> 7:0/0
> 13:24:49.458 -> 8:0/0
> 13:24:49.458 -> 9:0/0
> 13:24:49.458 -> 10:0/0
> 13:24:49.458 -> 11:0/0
> 13:24:49.458 -> Data JSON stored
> 13:24:49.701 -> Hey I got a callback 
> 13:24:49.701 -> Store signal
> 13:24:49.701 -> Min ind: 
> 13:24:49.701 -> 3
> 13:24:49.701 -> store code :
> 13:24:49.701 -> 507246/21893
> 13:24:49.701 -> Col: val/timestamp
> 13:24:49.701 -> 0:8895854/21341
> 13:24:49.701 -> 1:8895854/21408
> 13:24:49.701 -> 2:8895854/21645
> 13:24:49.701 -> 3:507246/21893
> 13:24:49.701 -> 4:0/0
> 13:24:49.701 -> 5:0/0
> 13:24:49.701 -> 6:0/0
> 13:24:49.701 -> 7:0/0
> 13:24:49.734 -> 8:0/0
> 13:24:49.734 -> 9:0/0
> 13:24:49.734 -> 10:0/0
> 13:24:49.734 -> 11:0/0
> 13:24:49.734 -> Data JSON stored

1technophile, 2019-04-27:
Thanks for the extract. It seems that the wemos receives the RF signal before the callback. Could you gives me ideas on the distance between : 
* rf emitter - arduino
* wemos - arduino
* rf emitter - wemos

akasma74, 2019-04-27:
Distances:
1. 3-4 m
2. 1m (lay on the same table)
3. 3-4m

valdiks, 2019-08-13:
Each RF receiver will receive all RF messages because they don't have filters and they don't know who is sending RF message.
You should enable RF message filter on your automation system (Home Assistant, iobroker, node-red, OpenHab, Domoticz, etc.).
OMG is gateway and it does his job - it receives all messages (even not your home RF device messages) and translates all messages to MQTT Broker.

It should work something like this:
RF device -> OMG -> MQTT -> Automation system with filters -> MQTT -> OMG -> RF device / Relay / etc.

Then you could move your RF device around the house and any OMG with better signal will receive messages from RF device.

akasma74, 2019-08-13:
As far as I know OMG has code to avoid duplicate MQTT messages, and this issue shows that it doesn't always work.
I know there are other ways to deal with duplicate messages.
I'm not very keen on your proposed message cycle as it implies a lot of duplicate MQTT messages and requires republishing, not sure that's the best solution tbh.

valdiks, 2019-08-13:
Here you have both OMGs received two signals with different value:
09:53:18: (0) home/OpenMQTTGateway/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":454}
09:53:18: (0) home/OMG_2nd_floor/433toMQTT {"value":14590730,"protocol":1,"length":24,"delay":453}

09:53:19: (0) home/OpenMQTTGateway/433toMQTT {"value":7295365,"protocol":1,"length":23,"delay":453}
09:53:20: (0) home/OMG_2nd_floor/433toMQTT 
{"value":511,"protocol":13,"length":8,"delay":61}

What it is the device that sends signal? Maybe it is a some additional control message?

akasma74, 2019-08-13:
It's a door sensor as per my initial post.

valdiks, 2019-08-13:
It's possible that second message (values 7295365 and 7295367) is used for tamper switch?
Have a look [here](https://community.home-assistant.io/t/sonoff-dw1-433mhz-door-window-sensor-low-battery-warning-capable/43073/3).

As for value:511 I cannot say anything.

closed 2023-10-29 by github-actions[bot]