1technophile/OpenMQTTGateway #391: IRrecvDumpV2 from IRremoteESP8266 gives very different results

freman, 2019-04-27:
**Describe the bug**
Trying to learn the remote my projector uses, the codes that IRtoMQTT have been spitting out (and even the debugging) haven't been working.

Compiled the IRrcevDumpV2 example and put it on the ESP8266 and it spits out a code I can give to MQTTtoIR just fine

** Code as seen by OpenMQTTGateway **
```
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Creating IR buffer
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Rcv. IR
[2019-04-28 00:14:45] home/OpenMQTTGateway/log raw signal redirected
[2019-04-28 00:14:45] home/OpenMQTTGateway/log isAduplicate?
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Adv data IRtoMQTT
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Pub json into:
[2019-04-28 00:14:45] home/OpenMQTTGateway/log home/OpenMQTTGateway/IRtoMQTT
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Store to avoid duplicate
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Min ind:
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 4
[2019-04-28 00:14:45] home/OpenMQTTGateway/log store code :
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 803007/210369
[2019-04-28 00:14:45] home/OpenMQTTGateway/log Col: val/timestamp
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 0:803007/80890
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 1:4294967295/81032
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 2:803007/91243
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 3:4294967295/91385
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 4:803007/210369
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 5:0/0
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 6:0/0
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 7:0/0
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 8:0/0
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 9:0/0
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 10:0/0
[2019-04-28 00:14:45] home/OpenMQTTGateway/log 11:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Creating IR buffer
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Rcv. IR
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 9048,2242,618
[2019-04-28 00:14:46] home/OpenMQTTGateway/log raw signal redirected
[2019-04-28 00:14:46] home/OpenMQTTGateway/log isAduplicate?
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Adv data IRtoMQTT
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Pub json into:
[2019-04-28 00:14:46] home/OpenMQTTGateway/log home/OpenMQTTGateway/IRtoMQTT
[2019-04-28 00:14:46] home/OpenMQTTGateway/log {"value":4294967295,"protocol":3,"bits":0,"raw":"9048,2242,618"}
[2019-04-28 00:14:46] home/OpenMQTTGateway/IRtoMQTT {"value":4294967295,"protocol":3,"bits":0,"raw":"9048,2242,618"}
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Store to avoid duplicate
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Min ind:
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 5
[2019-04-28 00:14:46] home/OpenMQTTGateway/log store code :
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 4294967295/210511
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Col: val/timestamp
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 0:803007/80890
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 1:4294967295/81032
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 2:803007/91243
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 3:4294967295/91385
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 4:803007/210369
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 5:4294967295/210511
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 6:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 7:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 8:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 9:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 10:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log 11:0/0
[2019-04-28 00:14:46] home/OpenMQTTGateway/log Hey I got a callback
```

** Code as seen by IRrecvDumpV2 **
```
(D p:^11840ms) (loop) Timestamp : 000052.537
(D p:^0002ms) Encoding  : NEC
(D p:^0000ms) Code      : C40BF (32 bits)
(D p:^0001ms) Library   : v2.5.6
(D p:^0001ms)
(D p:^0008ms) Raw Timing[71]:
(D p:^0000ms)    +  9112, -  4444,    +   650, -   488,    +   646, -   490,    +   646, -   488,
(D p:^0002ms)    +   646, -   488,    +   646, -   490,    +   646, -   488,    +   650, -   488,
(D p:^0001ms)    +   646, -   488,    +   646, -   488,    +   646, -   488,    +   646, -   490,
(D p:^0001ms)    +   644, -   490,    +   646, -  1624,    +   646, -  1624,    +   646, -   490,
(D p:^0001ms)    +   646, -   488,    +   646, -   488,    +   650, -  1620,    +   646, -   488,
(D p:^0001ms)    +   646, -   494,    +   646, -   488,    +   646, -   488,    +   646, -   490,
(D p:^0001ms)    +   646, -   488,    +   646, -  1624,    +   646, -   488,    +   646, -  1624,
(D p:^0001ms)    +   646, -  1620,    +   650, -  1620,    +   646, -  1624,    +   646, -  1620,
(D p:^0001ms)    +   646, -  1620,    +   646, - 45912,    +  9116, -  2182,    +   646
(D p:^0001ms)
(D p:^0007ms) uint16_t rawData[71] = {9112, 4444,  650, 488,  646, 490,  646, 488,  646, 488,  646, 490,  646, 488,  650, 488,  646, 488,  646, 488,  646, 488,  646, 490,  644, 490,  646, 1624,  646, 1624,  646, 490,  646, 488,  646, 488,  650, 1620,  646, 488,  646, 494,  646, 488,  646, 488,  646, 490,  646, 488,  646, 1624,  646, 488,  646, 1624,  646, 1620,  650, 1620,  646, 1624,  646, 1620,  646, 1620,  646, 45912,  9116, 2182,  646};  // NEC C40BF
(D p:^0007ms) uint32_t address = 0x3000;
(D p:^0000ms) uint32_t command = 0x2;
(D p:^0001ms) uint64_t data = 0xC40BF;
(D p:^0000ms)
(D p:^0000ms)
```

1technophile, 2019-04-27:
You missed some additionnal informations from the bug template, in particular the version used  ?

freman, 2019-04-27:
> An update has just been made on the dev branch , is it the one you are using ?

Sorry, makes sense to provide that, it's just we hours of the morning and I was more concerned about sharing my tale of sadness.

OpenMQTTGateway is at the following commit (I cloned it)
```
commit 8b0e3eb8b4f52a9b2ecf3da03b77e28c74d74bf4 (HEAD -> development, origin/development, origin/HEAD)
Author: Legion2 <leon.k97@gmx.de>
Date:   Mon Apr 22 16:13:56 2019 +0200

    added IR json datastring HEX decoding
    this breaks current datastring usage
```

It has a couple of tiny modifications (I made disabled serial output, and switched the pins IR is on (esp-01, using tx and rx))

IRremoteESP8266 is on version 2.5.6

1technophile, 2019-04-27:
Could you try to deactivate the mqtt log. In some cases I have seen that it can cause unexpected behaviour.

freman, 2019-04-27:
I was originally running it without the mqtt log, I only turned it on as a last ditch effort to see if something was getting lost in translation.

 I couldn't get IR_GC working either, nothing was happening when I sent requests for it - which was another reason I turned the log on.

Without the log all I was getting back was variations of `{"value":4294967295,"protocol":3,"bits":0,"raw":"9048,2242,618"}` sometimes with different raws, sometimes with none. Interesting that the protocol detection was working, I fired my samsung aircon remote at it and it figured out it was samsung but the codes it spat back were no good for it too.

One thing I thought is that we've overflowed something?
4294,967295 is 2<sup>32</sup> − 1 (aka 11111111111111111111111111111111)

1technophile, 2019-04-27:
I think so, i should go to unsigned long long for the value handling

freman, 2019-04-27:
For the record, 0xC40BF is the correct value so I'm still not sure how we're overflowing

1technophile, 2019-05-04:
If 0xC40BF is what you are expecting this means a decimal value of 803007
this is produced by OMG on your logs:
home/OpenMQTTGateway/log 2:803007/91243

The question is indeed why there is 4294967295 just before or after

freman, 2019-05-04:
Oh see that, it never sent it over MQTT tho, just 4294967295.

Pretty sure the 4294967295 isn't interference from outside light sources as I did this test under a heavy quilt just to be sure I didn't pick up a stray signal

closed 2019-08-14 by stale[bot]