1technophile/OpenMQTTGateway #392: Restart via MQTT

Digitsl-dB, 2019-04-30:
I am using this code on the ESP32 and it is great! 
One function I would like to see is a reset via MQTT. 
eg from Tasmota:
`publish("broker", "cmnd/" + device_id + "/restart", "1")`
Is this possible or already implemented (I couldn't find it in the wiki)?
Thanks

1technophile, 2019-04-30:
Thanks, just for my curiosity could you explain why you need to restart the ESP?

Digitsl-dB, 2019-05-01:
Hi sometimes (once a month) the data to the MQTT broker stops but it is still online. 
I have to cycle the power to start the MQTT data again.
A reset could be issued by OpenHab if the data is not updating at least every 5 mins.

1technophile, 2019-05-01:
Hi, thanks for the info. Could you indicates what if the frequent of readings that you are using ?  Timebetweenreadings value in config_BT.h

Digitsl-dB, 2019-05-01:
Hi, I set TimeBtw_Read to 1000 as I use it to trigger events when a BT device is recognised.

1technophile, 2019-05-04:
Hi, so as to try to find the root cause, could you indicate me what is the version of ESP32 arduino env and what is the board used?

Digitsl-dB, 2019-05-05:
It is the ESP32 DEVKIT V1
The OMG is 0.8
The Arduino is 1.8.7
When I check the MQTT Broker (Mosquitto on OpenHabian running on a Pi 3) the device is online, it runs for a few weeks usually, but then it stops and no other data comes from it until I press the EN button on the board or cycle the power. 
Mosquitto shows it is online even if I reset Mosquitto is still shows as online but no further data.

RaZer0r, 2019-05-09:
FYI, Same thing happens here. interval for BT scan is unknown right now, but i can play around with the settings.
Once in a while it will stop updating BT addresses to the MQTT broker. I also have a few esp8266 devices and those run without a hickup.

1technophile, 2019-05-10:
Would it be possible with the latest version from the dev branch to issue this command once the gateway stop publishing BLE data:
`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/set -m '{"interval":0}'`

this should generate a scan and we will see if it wake up the BLE components

RaZer0r, 2019-05-26:
will do, upgrading from latest dev branch now

1technophile, 2019-05-26:
For info i'm going to commit dome modification to enable auto reconnection to mqtt broker when the reconnection last more than 1 minutes. This issue was maybe the root cause of your problem.

1technophile, 2019-05-30:
Modification done in dev branch.
I hope it will solve the issue of OMG being online and not publishing data.

Could you test it and if the test are not successfull I will add a restart function?

1technophile, 2019-06-21:
My esp32 is working since several weeks now
I'm closing this issue, if yours are not working properly and you need this restart function don't hesitate to reopen

closed 2019-06-21 by 1technophile