1technophile/OpenMQTTGateway #393: Json Publishing not being picked up by HA sensor

stredwizard, 2019-04-30:
First and foremost this is great so thank you, I've been using it on a nodemcu as just a IR, RF receiver and Sender for over a year, but decided I wanted some extras like PIR and a temperature sensor as well, I had a spare esp8266-12 laying around so I used that instead and its working great as an RF sender / Receiver, temp and PIR except.....

The payload message for HCSR501 IR detection is {"hcsr501":"true"} which does not match the auto discovered sensor in home assistant, if I publish a true or false message on the topic the sensor performs as expected in HA.

I found the line in ZsensorHCSR501 and I found the section about son publishing in User_config, but I'm afraid to make changes here as I'm not sure these will work or be a solution.

I'm using the latest OpenMQTTGateway version 0.9 master

cheers

1technophile, 2019-04-30:
Do you think we should better publish:
`{"hcsr501":true}`
instead of
`{"hcsr501":"true"}`

stredwizard, 2019-05-01:
unfortunately I don't know, 
`{"hcsr501":"true"}` 
As sent by the gateway doesn't work for the auto discovered sensor and if I manually publish your example
`{"hcsr501":true}`
that doesn't either, I of course may be missing a point somewhere that I don't understand.

cheers

1technophile, 2019-05-04:
I have made a correction to mqtt discovery with HA, could you try last dev branch please?

stredwizard, 2019-05-05:
I have loaded the dev branch on a nodemcu with HCSR501 sensor attached, everything seems to work correctly now thank you.

I also have an ESP8266-12 which has the HCSR501, DHT11 and the RF Transciever, takes a little more work to flash this one and hopefully will do this tomorrow, the RF transceiver on this one is paired with two recomended GS107 RF door sensors and I had to write a small node-red flow to get these to work with HA, hopefully the new changes in the dev branch will make these work natively in HA.

Cheers

closed 2019-05-12 by 1technophile