1technophile/OpenMQTTGateway #397: failed to connect to MQTT

opl-git, 2019-05-04:
Hello,

first: thank you again for this great openmqtt!

I am using openmqtt with two gateways. Both were working finde
Suddenly one the gateways can not connect to my Mosquitto MQTT Broker anymore.
The other is working fine.
Here is the output of the serial console:
`MQTT connection...
failure_number
1
failed, rc=
-2
try again in 5s
MQTT connection...
failure_number
2
failed, rc=
-2
try again in 5s
MQTT connection...
failure_number
3
failed, rc=
-2
try again in 5s
MQTT connection...
failure_number
4
failed, rc=
-2
try again in 5s
MQTT connection...
failure_number
5
failed, rc=
-2
try again in 5s
failed connecting to mqtt
`

I flashed the gateway serveral more time, changed the mqtt user and pass to "".
But still, I can not connect to the broker. 

Maybe someone had similiar problems?
Thank you!

akasma74, 2019-05-04:
Have you tried to restart your broker?

1technophile, 2019-05-04:
Do your gateways have different names ?

opl-git, 2019-05-04:
I already restarted the docker mosquitto mqtt. The mqtt broker does not have any user / password protections. 

Both gateways have different names. 
I unplugged one yesterday to put it in case 3d printed case. That`s the only change I made.
Since then it stopped working. 
It is still connecting to my wifi without any problems, so to me it is no connection problem.
Also, all the other devices using the broker running without any problems.

What I alreay tried:
- Flash another ESP32 with the same config the "broken one" uses
- Flash the other ESP32 with an other gateway name

1technophile, 2019-05-04:
could you check on mosquitto side what is going wrong by reading its log file?

2114L3, 2019-05-05:
i am getting the same thing
lots of other esp8266 arduino nodes on pubsubclient 2.6 (non-OMG) working with mosquitto version 1.4.10 (build date Wed, 13 Feb 2019 00:45:38 +0000)  

mosquitto logs on the server show;

1557035691: New connection from 192.168.1.55 on port 1883.
1557035691: Sending CONNACK to 192.168.1.55 (0, 5)
1557035691: Socket error on client <unknown>, disconnecting.
...over and over

1technophile, 2019-05-05:
And with the last version if OMG dev branch do you hve the same issue ?

2114L3, 2019-05-05:
sorry, i meant my old nodes are 2.6, but as per instructions 2.7 was used with OMG where the issue was found.

i have resolved my issue in OpenMQTTGateway.ino line 147
`if (client.connect(Gateway_Name, mqtt_user, mqtt_pass, will_Topic, will_QoS, will_Retain, will_Message)) {`
changed mqtt_user and mqtt_pass to NULL
`if (client.connect(Gateway_Name, NULL, NULL, will_Topic, will_QoS, will_Retain, will_Message)) {`

pubsubclient doc mentioned the following;

https://pubsubclient.knolleary.net/api.html
> boolean connect (clientID, username, password)
> Connects the client with a username and password specified.
> 
> Parameters
> clientID : the client ID to use when connecting to the server.
> username : the username to use. If NULL, no username or password is used (const char[])
> password : the password to use. If NULL, no password is used (const char[])

seems pubsubclient isnt getting triggered into anonymous mode correctly with the variables?

opl-git, 2019-05-05:
Thank you all for your help!
I let everything on my desk and went the night out.
Today I plugged everything in again (made no changes to the broker nor the gateway) and everything is working again... 
Sunday miracle??? 🤣

closed 2019-05-05 by opl-git

justoke, 2019-05-25:
I have had exactly the same issue, except this started happening after I installed a new mosquitto MQTT broker on a Raspberry Pi 3. This was to replaced a failed Pi. As it turns out the issue was that I had enabled anonymous access in the mosquitto configuration file, but I also had a passwords file. In this scenario, any client not supplying the username and password will be rejected! This is not a bug, but mosquitto by design and a gotcha if you just run default installation and don't want to have username and passwords for your home automation.

This took me a whole day to figure out, so here a few steps to take to save you the trouble:
Check the mosquitto log - you should easily see that the OpenMQTTGateway is trying to reconnect and getting rc= 5 - this is basically access denied code
tail -f /var/log/mosquitto/mosquitto.log 
**_1558811952: Socket error on client <unknown>, disconnecting.
1558811954: New connection from 192.168.1.69 on port 1883.
1558811954: Socket error on client <unknown>, disconnecting._**
Edit the websockets.conf file
sudo nano /etc/mosquitto/conf.d/websockets.conf - in my case I commented out the password file line
#password_file /etc/mosquitto/passwords
Restart the service
sudo systemctl restart mosquitto
Check the log, although you will probably be seeing data on your dashboard to confirm that it is working:
tail -f /var/log/mosquitto/mosquitto.log

henkiejan1, 2020-01-19:
I have exactly the same problems as the people here above. I want to connect with password but it gives continuesly a rc=5 error. When i change the password i got a connection reset by peer 104 error. I have changes anonymous allow on and off. the name of the password file changes, password changes. a few times flashed again but nothing works... Does someone suggestions?

mrespin, 2024-05-05:
Same problem rc=5
However, I have set Mosquitto to have both anonymous access and password file.
Solution: both username and password should be emptied in OMG setup for anonymous access to work.
Thanks for the suggestions.