1technophile/OpenMQTTGateway #401: Some mqtt messages don't get to the broker.

molda, 2019-05-07:
First of all THANK YOU for OpenMQTTGateway. It's awesome. 

I uploaded OpenMQTTGateway to my esp32 about a month ago.
I put bateries into two Xiaomi MiJia Temp sensors and as expected i started to recieve messages from both sensors. It worked for a month.
Unfortunatelly now i only recieve messages with .../home_presence/... a .../SYStoMQTT/... topics. 
I no longer recieve messages with `tem` and `hum` and `batt` values.

In the terminal i can see message:
```
Pub json into:
.../OpenMQTTGateway/BTtoMQTT/<device-id>
```
but this message never shows up in broker. And neither has the message above mentioned values for tem, hum and batt. See the screenshot bellow.
I can see every single message in the broker's terminal so i'm sure it's not getting there.

Terminal screenshot [here](https://drive.google.com/open?id=1CpU-ukCsew2Hl0N9ICCgvm9yJZhub2kI)

I will reupload OMG to my esp32 to see if it helps but thought it's worth sharing my experience.

1technophile, 2019-05-09:
Would it be possible with the latest version from the dev branch to issue this command once the gateway stop publishing BLE data:
`mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/set -m '{"interval":0}'`

this should generate a scan and we will see if it wake up the BLE components

molda, 2019-05-13:
So i tried to publish that and nothing happened, i mean no immediate mqtt message or anything. It took another minute or so to recieve home_presence message.  Unfortunatelly i didn't have my laptop with me  to connect the ESP32 to so i coudn't see the serial debug output.

I reflashed *v0.9.1* to one ESP32 and *v0.9.2beta* to the second ESP32.
After 2-3 resets both started to publish tem and hum values every 1 minute or so, but it's not very consistent. Sometimes only hum value is published sometimes only batt value. Sometimes it takes up to 4 minutes to get next message but mostly it's 1 minute.

Should the tem, hum and batt values be published at the same time(in the same mqtt message)?

1technophile, 2019-05-14:
>Should the tem, hum and batt values be published at the same time(in the same mqtt message)?
The method currently used scan the sensors advertisements. Mi flora publish one value at a time, and mi jia one or two values at a time.
Unfortunately when OMG publish only one value the other ones seems to be set to zero with HA (OpenHAB doesn't behave like that).
From this analysis we have several possibilities : 
* store previously published values and republish them each time one of them is scanned.
* wait for all the values to be scanned before publishing them.
* change the values retrieving method
* ask for an HA modification to don't zero values when they are not present in the json

Alternatively there is still the possibility for simplePublishing as a temporary way.

molda, 2019-05-14:
Thank you.
I was just going through the OMG User_config.h and noticed the simplePublishing  so i will give that a try.
Althought i mentioned the home_presence messages i'm not actually using HA nor OpenHAB so i'm not having an issues with other values being set to zero.

I was initially thinking that my MiJia sensor broadcasts all the values at once and OMG is having some problems parsing the data. Never thought the sensor would broadcast the values somewhat randomly.

Since yesterday the ESP32 with v0.9.2beta seems to run OK although sometimes it takes more than 2 minutes to get next tem value.

1technophile, 2019-05-21:
>Althought i mentioned the home_presence messages i'm not actually using HA nor OpenHAB so i'm not having an issues with other values being set to zero.

>I was initially thinking that my MiJia sensor broadcasts all the values at once and OMG is having some problems parsing the data. Never thought the sensor would broadcast the values somewhat randomly.

This point should be now OK with the last dev branch, if it is not the case don't hesitate to reopen the issue

closed 2019-05-21 by 1technophile