1technophile/OpenMQTTGateway #402: DHT22 sensor needs to begin()

cometfish, 2019-05-16:
I bought a DHT22 sensor, and attempted to hook it up to my OpenMQTTGateway.
Most of the time it returned 'Failed to read from DHT sensor!', but maybe 1 in 5 attempts it would return rubbish data, eg. humidity=3000 (obviously wrong) and temp=10 (it was 19C at the time).

I tested with the example library for the DHT sensor, and it was working fine. The only difference I could see between the example code and the ZsensorDHT.ino code was that the example called `dht.begin()` at the start.

I added that to ZsensorDHT.ino and now my sensor is working perfectly with OpenMQTTGateway:
```diff
diff --git a/ZsensorDHT.ino b/ZsensorDHT.ino
index c3d79f1..cc98b3b 100644
--- a/ZsensorDHT.ino
+++ b/ZsensorDHT.ino
@@ -35,8 +35,13 @@ DHT dht(DHT_RECEIVER_PIN,DHT_SENSOR_TYPE);
 
 //Time used to wait for an interval before resending temp and hum
 unsigned long timedht = 0;
+bool hasBegun = false;
 
 void MeasureTempAndHum(){
+  if (!hasBegun) {
+    dht.begin();
+    hasBegun = true;
+  }
   if (millis() > (timedht + TimeBetweenReadingDHT)) {//retriving value of temperature and humidity of the box from DHT every xUL
     timedht = millis();
     static float persistedh;

```

**Environment:**
* OpenMQTTGateway version 0.9.1 running on an ESP8266 with a DHT22 sensor
* DHT sensor library 1.3.4
* Sensor works with the example code from https://github.com/adafruit/DHT-sensor-library/blob/master/examples/DHTtester/DHTtester.ino

1technophile, 2019-05-16:
This should be due to a wrong version of DHT library used.
If you take the libraries from the [lib ](https://github.com/1technophile/OpenMQTTGateway/tree/development/lib) folder you should not have any issue.

1technophile, 2019-05-21:
I close the issue, if you have some questions don't hesitate to ask into the [forum](https://community.openmqttgateway.com/latest)

closed 2019-05-21 by 1technophile

cometfish, 2019-06-28:
Hi, sorry for the late response, took a while to get some time to test this - I reset back to default libraries (with the exception of WiFi Manager, because it will not compile with the included library due to https://github.com/tzapu/WiFiManager/issues/576 ) and it did indeed work without issue.
Thanks :)