1technophile/OpenMQTTGateway #404: New hardware revision of Sonoff 433MHz WiFi bridge.

tjhowse, 2019-05-21:
The PCBs in the documentation have different silkscreen markings to the current model as of 2019-04-21.

https://imgur.com/a/OMGYtrs

Flashing 0.9.1beta worked as per the instructions, and OTA updates work too, but I suspect the pin assignments may have changed? Though the traces from what I assume the RF hardware don't seem to have changed.

Here are logs. I was pressing the buttons on a 433MHz RF transmitter that is known to be working with a bare ESP8266 and these radios: https://www.banggood.com/433Mhz-RF-Transmitter-With-Receiver-Kit-For-Arduino-ARM-MCU-Wireless-p-74102.html?p=SZ25159101562014119R&custlinkid=154640&cur_warehouse=CN

MQTT Log:
top/OpenMQTTGatewayLivingRoom/LWT online                                                                              top/OpenMQTTGatewayLivingRoom/version 0.9.1beta                                                                       top/OpenMQTTGatewayLivingRoom/SYStoMQTT {"uptime":120,"freeMem":45376,"rssi":-70,"SSID":}

Serial log:
noleechforyou
.
.
.
.
.
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac:
60:01:94:CC:BD:01
OpenMQTTGateway ip:
192.168.1.242
1883
Connecting to MQTT by IP adress
192.168.1.50
ZgatewaySRFB setup done
Serial Baud19200
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects
Uptime (s)
120
Remaining memory
45376
RSSI
-59
SSID
noleechforyou
Activated modules
SRFB

1technophile, 2019-05-21:
To my knowledge the link between the Sonoff RF Bridge included decoder was not changed between the 2 revisions.
Do you have some other devices to test with your RF bridge?

The pin assignment in config_RF.h is only in case of doing the RF Bridge [hardware mod](https://1technophile.blogspot.com/2019/04/sonoff-rf-bridge-pilight-or-how-to.html)

1technophile, 2019-06-21:
I'm closing this issue, if you need other info don't hesitate to reply

closed 2019-06-21 by 1technophile