1technophile/OpenMQTTGateway #405: Default password to OpenMQTTGateway

ecajrt, 2019-05-23:
Hi

After flashing OpenMQTTGateway_1 using ESPEasy flasher the AP OpenMQTTGateway is showing as a wifi in my phone. When trying to connect it asks for password. I guess it's a default password to get in and do the initially settings. For a ESP8266 it is 'espeasy' but for this???

akasma74, 2019-05-23:
Have your tried your_password   ? ;)

ecajrt, 2019-05-24:
I tried it without progress. Byt this is in the installation process where
the rf bridge still act as a accesspoint. It should have its own temporary
password until I get in and configure it as a client to my own Wifi. At
least it work that way with other esp8266 stuff configured with the ESPEasy
Flasher. ...like this is.

Regards  Johan

ecajrt, 2019-05-24:
...or do you mean that your_password is the actual default password? :-)

Den fre 24 maj 2019 kl 12:48 skrev Johan Reit <johan.reit@gmail.com>:

akasma74, 2019-05-24:
that's exactly what I mean ;)

ecajrt, 2019-05-24:
Ok, thanks a lot! :-)

1technophile, 2019-05-26:
The default password is also indicated into the wiki for info:
https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP8266#software-setup

closed 2019-05-26 by 1technophile

agrynco, 2025-09-02:
What if I did not define any spasswords and not it does not aalow me to connect to the device?

1technophile, 2025-09-02:
If you uploaded the latest version the password is compulsory.
If you don't recall it, you can still do a new upload from scratch and Erase the flash