1technophile/OpenMQTTGateway #408: Lack of valueAsASubject in ZgatewayPilight messages

wired-dev, 2019-05-28:
I resolved the issue for myself by adding 
`
    RFPiLightdata.set("message", (char *)message.c_str());
      #ifdef valueAsASubject
        StaticJsonBuffer<JSON_MSG_BUFFER> jsonBuffer2;
        JsonObject& RFPiLightdatain=jsonBuffer2.parseObject((char *)message.c_str());
        RFPiLightdata.set("value", RFPiLightdatain["id"]);
      #endif
`
to ZgatewayPilight;
this also made it easier to read the Digoo-DG-R8S device.

wired-dev, 2019-06-10:
`    JsonObject& RFPiLightdatain=jsonBuffer2.parseObject((char *)message.c_str());
    if(isDigit(RFPiLightdatain["id"])){
    RFPiLightdata.set("value", RFPiLightdatain["id"]);
    } else {
    RFPiLightdata.set("value", RFPiLightdatain["id"].as<String>());  
    }`

wired-dev, 2019-06-10:
then in the main c file I changed it to do strings too.

`    #ifdef valueAsASubject
      if(isDigit(data["value"])){
        unsigned long value = data["value"];
//      if (value != 0){
        topic = topic + "/"+ String(value);
      } else if(data["value"].as<String>()!= 0) {
        topic = topic + "/"+ data["value"].as<String>();
      }
    #endif`
this makes it work for more devices on pilight.

akasma74, 2019-06-10:
Where did you add it and how does it help?
I own some DG-R8S sensors and use them with HA - could you share your relevant HA config?
I tried to use valueAsASubject, but didn't find it useful and currently use demultiplexers.

p.s maybe it's more reasonable to create a PR? ;)

wired-dev, 2019-06-11:
not sure how to do a PR; but yeah just starting with git

1technophile, 2019-06-21:
@wired-dev if you need help for the PR don't hesitate to ask questions, in some simple step:
- download github desktop
- fork OMG development branch
- clone it localy
- modify your files localy
- commit your modifications and publish them from github desktop
- go to your fork on github
- you should see a button to submit a PR

closed 2020-09-02 by 1technophile