1technophile/OpenMQTTGateway #409: HA discovery failing on identifiers

bvduijn, 2019-05-30:
I'm trying to set up the gateway and configure HA for autodiscovery. My setup is an Arduino Mega and a W5100.

I can see packages being published via MQTT but HA produces an error about the message containing no identifier

MQTT broker:
homeassistant/sensor/DEEDBAFE5495gatewayRF/config {"stat_t":"home/OpenMQTTGateway/433toMQTT","name":"gatewayRF","uniq_id":"DEEDBAFE5495gatewayRF","val_tpl":"{{ value_json.value }}","device":{"name":"OpenMQTTGateway","manufacturer":"OMG_community","sw_version":"0.9.1","identifiers":[]}}

home-assistant.log
ERROR (MainThread) [homeassistant.components.mqtt.sensor] Exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'value_template': '{{ value_json.value }}', 'state_topic': 'home/OpenMQTTGateway/433toMQTT', 'name': 'gatewayRF', 'unique_id': 'DEEDBAFE5495gatewayRF', 'platform': 'mqtt', 'device': {'name': 'OpenMQTTGateway', 'manufacturer': 'OMG_community', 'identifiers': [], 'sw_version': '0.9.1'}},)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/config_validation.py", line 663, in __call__
    return super().__call__(data)
  File "/srv/homeassistant/lib/python3.5/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/srv/homeassistant/lib/python3.5/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/srv/homeassistant/lib/python3.5/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: Device must have at least one identifying value in 'identifiers' and/or 'connections' for dictionary value @ data['device']

Note: This is a full install of home assistant, no hass.io

By changing JSON_ARRAY_SIZE(1) to JSON_ARRAY_SIZE(2) I now receive identifiers (mac address)

1technophile, 2019-05-30:
Could you try with v0.9.2beta (dev branch) it has been corrected on it

1technophile, 2019-05-30:
I read you message too fast, it was another that has been corrected.
I will do some checks to have appropriate JSON_ARRAY_SIZE.

1technophile, 2019-06-21:
done in dev

closed 2019-06-21 by 1technophile