1technophile/OpenMQTTGateway #410: Support for Meizi BT Hygrothermo

jamesyym, 2019-06-04:
This library doesn't seem to support Meizi BT Hygrothermo. It is an smart IR remote and temperature and humidity sensor.

Here is the project of Home-assistant sensor platform for Meizi BT Hygrothermo temperature and humidity sensor. 
https://github.com/373137461/Meizu_Hygrothermo

Will you please add the support for this device?
Thank you very much!

1technophile, 2019-06-04:
Hello Jamesyym,

Do you have some link to this device?

jamesyym, 2019-06-04:
I am sorry I could only find the links about the device in Chinese.

https://tech.sina.cn/mobile/pc/2016-08-04/detail-ifxutfpc4413069.d.html?from=wap

https://n.znds.com/article/12288.html

1technophile, 2019-06-04:
No problem, could you scan it with an app like beaconn simulator and see if it emits a "service data".
If it is the case I would be interested in having some service data extract and the corresponding sensor values.

jamesyym, 2019-06-04:
Which APP should I use? iOS or Android? Which platform is prefered? Thank you.

1technophile, 2019-06-05:
You can use this one : 
https://play.google.com/store/apps/details?id=net.alea.beaconsimulator

jamesyym, 2019-06-06:
FYI.
[image]
[image]
Thank you.

jamesyym, 2019-06-06:
the sevice udid is "000016f2-0000-1000-8000-00805f9b34fb"

1technophile, 2019-06-15:
may I understand that you don't need this ?

jamesyym, 2019-06-16:
> may I understand that you don't need this ?

Hi Florian,

Thanks for your concern and I‘ve got it working with [esp32-ble2mqtt](https://github.com/shmuelzon/esp32-ble2mqtt)

1technophile, 2019-06-16:
Hi,

Yes i'm still interested. We can let it open