1technophile/OpenMQTTGateway #411: Sonoff RFB hangs???

henkiejan1, 2019-06-12:
Hi all,

I have followed the instructions in the wiki. It's all clear and flashing is also no problem. Also booting is ok. But then it does nothing anymore. No topics nothing. I tested more RF controllers.

This is the log:

`*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.2.61
connected...yeey :)
saving config
{"mqtt_server":"192.168.2.4","mqtt_port":"1883","mqtt_user":"domoticz","mqtt_pass":"secret"}*WM: freeing allocated params!
OpenMQTTGateway mac:
60:01:94:BE:31:E7
OpenMQTTGateway ip:
192.168.2.61
1883
Connecting to MQTT by IP adress
192.168.2.4
ZgatewaySRFB setup done
Serial Baud19200
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects
Uptime (s)
120
Remaining memory
40928
RSSI
-62
SSID
FiberSpeed
Activated modules
SRFB`

Any suggestions? There is also not topic published and the switch is in the ON posititon.

1technophile, 2019-06-13:
Hello,

Did you changed the little switch position after uploading?
When you press your remote button, can you see the red light on in the front of the bridge?

henkiejan1, 2019-06-14:
I switched the little switch to off before flashing. After flashing i switched to on. But i didn't see any red flash on the board when i press a button...

1technophile, 2019-06-15:
Hello,

If you don't see the red light flashing this is due to the fact that your emitter protocol is not recognised by RF Bridge internal decoder (not the same thing as the esp8285 that you've flashed).
Your solution are:
* chech with an esp8266 and a rf receiver with OMG and ZgatewayRF or ZgatewayPilight  if your protocol is decoded, if it is the case stay with this solution or do the RF Bridge direct hack to bypass the RF bridge internal decoder
* change your RF device for a compatible one [here](https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=2126158079)

henkiejan1, 2019-06-16:
Thank you for your reply! I think i know the reason. I has uploaded alternative EFM8BB1 firmware from https://github.com/Portisch/RF-Bridge-EFM8BB1 ... Think the should not work with OpenMQTT?

1technophile, 2019-06-16:
Unfortunately I never tested this firmware. I can't say.

1technophile, 2019-06-21:
I'm closing this issue, if you need other info don't hesitate to reply

closed 2019-06-21 by 1technophile

Thoky81, 2019-07-08:
Sorry, I know that you have closes this issue but I have the same problem. I have have spend more than 8 hours to figure-out where and what I am doing wrong but nothing. I have done the direct hack with RF Bridge and so far ESPURNA direct is working. But I am not able to get the RF signal with Pilight - means that no Red LED is flashing and no received RF states in MQTT explorer.

I am flashing the OMG Pilight following this link https://1technophile.blogspot.com/2019/04/sonoff-rf-bridge-pilight-or-how-to.html?m=1 . I have even get to the point where I can access the AP Wifi and even I see the online topic in mqtt but like I said no RF codes. Any suggestion?

1technophile, 2019-07-12:
If you keep an ftdi connected do you see some traces that you could share?