1technophile/OpenMQTTGateway #412: How to get temperature/Humidity from Mijia model LYWSD02

xenio, 2019-06-15:
I flashed the release OpenMQTTGateway v0.9.1 on ESP32 and I am not getting the temperature and from the Clock Mijia model LYWSD02 as in this post:
https://1technophile.blogspot.com/2019/06/integration-of-mijia-clock-temperature.html

I just get the id, rssi, distance and service data.
Any Help?

1technophile, 2019-06-15:
It is available from v0.9.2beta. You can take the code from the dev branch.

xenio, 2019-06-16:
Perfect!!! Thanks you.

1technophile, 2019-06-16:
Don't forget to take the libraries from the lib foldet of the repository. The ble one supplied with arduino esp32 env is missing an important functionnality.

jamesyym, 2019-06-20:
> It is available from v0.9.2beta. You can take the code from the dev branch.

I tried the development branch with ESP8266 and HM-10(FW607) , but it still only get the id, rssi, distance and service data.

3F59C8828829 is the MAC address of LYWSD02 and I set the whitelist for it. Am I right?

Thank you.

[image]

1technophile, 2019-06-20:
Could you copy/paste on this issue different exempla of service data you are seeing for the LYWSD02  please

jamesyym, 2019-06-21:
From the serial monitor:
```
Uptime (s)
120
Remaining memory
40856
RSSI
-39
SSID
James
Activated modules
BT
4f4b2b444953413afbc96899bb9800a40e0d09536d6172745068696c616d700d0a4f4b2b4449534345
Creating BLE buffer
HomePresenceId
98:BB:99:68:C9:FB
BLErssi
-92
txPower
-59
BLE DISTANCE :
27.75
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"98:BB:99:68:C9:FB","rssi":-92,"distance":27.75274}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/98BB9968C9FB
{"id":"98:BB:99:68:C9:FB","rssi":-92,"distance":27.75274,"servicedata":"5068696c616d70"}
4f4b2b444953413afbc96899bb9800a51d0201060b02031802180f180a18f5fe0dff5048494c414dfbc96899bb980d0a
Creating BLE buffer
HomePresenceId
98:BB:99:68:C9:FB
BLErssi
-91
txPower
-59
BLE DISTANCE :
25.52
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"98:BB:99:68:C9:FB","rssi":-91,"distance":25.51913}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/98BB9968C9FB
{"id":"98:BB:99:68:C9:FB","rssi":-91,"distance":25.51913,"servicedata":"02180f180a18f5fe0dff5048494c414dfbc96899bb98"}
4f4b2b444953413a0ed7cc343e6800a80a09ff006052572d424c450d0a
4f4b2b444953413a0ed7cc343e6800a91f02010609ff903a138d728daa6a1109e9ad85e6978fe981a5e68ea7e599a8000d0a
Creating BLE buffer
HomePresenceId
68:3E:34:CC:D7:0E
BLErssi
-87
txPower
-59
BLE DISTANCE :
18.08
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"68:3E:34:CC:D7:0E","rssi":-87,"distance":18.07764}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/683E34CCD70E
{"id":"68:3E:34:CC:D7:0E","rssi":-87,"distance":18.07764,"servicedata":"138d728daa6a1109e9ad85e6978fe981a5e68ea7e599a800"}
4f4b2b444953413a298882c8593f00a01c02010603021a18141695fe70205b0461298882c8593f09061002d0020d0a
Creating BLE buffer
HomePresenceId
3F:59:C8:82:88:29
BLErssi
-96
txPower
-59
BLE DISTANCE :
38.49
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"3F:59:C8:82:88:29","rssi":-96,"distance":38.4873}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/3F59C8828829
{"id":"3F:59:C8:82:88:29","rssi":-96,"distance":38.4873,"servicedata":"141695fe70205b0461298882c8593f09061002d002"}
4f4b2b444953413afc0180fce40f02b41f1eff0600010920025e64c45a7dcba1d732d7298e35267f4a1756cc2cae66700d0a
Creating BLE buffer
HomePresenceId
0F:E4:FC:80:01:FC
BLErssi
-76
txPower
-59
BLE DISTANCE :
6.45
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"0F:E4:FC:80:01:FC","rssi":-76,"distance":6.44788}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/0FE4FC8001FC
{"id":"0F:E4:FC:80:01:FC","rssi":-76,"distance":6.44788,"servicedata":"025e64c45a7dcba1d732d7298e35267f4a1756cc2cae6670"}
4f4b2b444953413a6517f8d1746903ae0e02011a0aff4c00100501103b46900d0a
Creating BLE buffer
HomePresenceId
69:74:D1:F8:17:65
BLErssi
-82
txPower
-59
BLE DISTANCE :
11.49
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"69:74:D1:F8:17:65","rssi":-82,"distance":11.4949}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/6974D1F81765
{"id":"69:74:D1:F8:17:65","rssi":-82,"distance":11.4949,"servicedata":"100501103b4690"}
4f4b2b444953413aaa1bd353905303d70e02011a0aff4c0010050310db1cb40d0a
Creating BLE buffer
HomePresenceId
53:90:53:D3:1B:AA
BLErssi
-41
txPower
-59
BLE DISTANCE :
0.03
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"53:90:53:D3:1B:AA","rssi":-41,"distance":0.026261}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/539053D31BAA
{"id":"53:90:53:D3:1B:AA","rssi":-41,"distance":0.026261,"servicedata":"10050310db1cb4"}
4f4b2b444953413a1f1e8e923b2d02c60f02011b0bff4c000906033fc0a802150d0a
Creating BLE buffer
HomePresenceId
2D:3B:92:8E:1E:1F
BLErssi
-58
txPower
-59
BLE DISTANCE :
0.84
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"2D:3B:92:8E:1E:1F","rssi":-58,"distance":0.842869}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/2D3B928E1E1F
{"id":"2D:3B:92:8E:1E:1F","rssi":-58,"distance":0.842869,"servicedata":"0906033fc0a80215"}
4f4b2b444953413a40584f87007803c60e02011a0aff4c00100507141153740d0a
Creating BLE buffer
HomePresenceId
78:00:87:4F:58:40
BLErssi
-58
txPower
-59
BLE DISTANCE :
0.84
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"78:00:87:4F:58:40","rssi":-58,"distance":0.842869}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/7800874F5840
{"id":"78:00:87:4F:58:40","rssi":-58,"distance":0.842869,"servicedata":"10050714115374"}
4f4b2b444953413a7b4ad66f00d901b31b11072753706865726f2d5475a02b6f74bb22080942422d344137420d0a
Creating BLE buffer
HomePresenceId
D9:00:6F:D6:4A:7B
BLErssi
-77
txPower
-59
BLE DISTANCE :
7.12
Pub json into:
home/home_presence/OpenMQTTGateway
{"id":"D9:00:6F:D6:4A:7B","rssi":-77,"distance":7.119795}
Pub json into:
home/OpenMQTTGateway/BTtoMQTT/D9006FD64A7B
{"id":"D9:00:6F:D6:4A:7B","rssi":-77,"distance":7.119795,"servicedata":"726f2d5475a02b6f74bb22080942422d34413742"}
4f4b2b444953413a7b4ad66f00d901b40a02010603ff3330020af60d0a
```
And the whitelist seems still not working with the [last commit in the development branch.](https://github.com/1technophile/OpenMQTTGateway/issues/331#issuecomment-504172499)

jamesyym, 2019-06-21:
Sorry for the wrong whitelist pattern I've made. The whitelist works well in mqtt but still only get the id, rssi, distance and service data. Below the message shows in mqtt explorer.
```
{
  "id": "3F:59:C8:82:88:29",
  "rssi": -65,
  "distance": 2.009448,
  "servicedata": "595753443032"
}
```
And the sevicedata changes like `"595753443032"` `"141695fe70205b04cd298882c8593f09041002ff00"` `"141695fe70205b0469298882c8593f090410020301"`

1technophile, 2019-06-21:
@jamesyym thanks for pointing this error with HM10 use. It has been corrected into the dev branch

closed 2019-06-21 by 1technophile

jamesyym, 2019-06-27:
Hey,
Thanks for the update. I tried the latest development branch, and just get the humidity no temperature.

[image]

```
Using 'COM3' as serial port.
Showing logs:
[17:02:19]mounting FS...
[17:02:19]mounted file system
[17:02:19]reading config file
[17:02:19]opened config file
[17:02:19]{"mqtt_server":"192.168.2.8","mqtt_port":"1883","mqtt_user":"mqtt","mqtt_pass":"6566"}
[17:02:19]parsed json
[17:02:19]*WM: Adding parameter
[17:02:19]*WM: server
[17:02:19]*WM: Adding parameter
[17:02:19]*WM: port
[17:02:19]*WM: Adding parameter
[17:02:19]*WM: user
[17:02:19]*WM: Adding parameter
[17:02:19]*WM: pass
[17:02:19]*WM: 
[17:02:19]*WM: AutoConnect
[17:02:19]*WM: Connecting as wifi client...
[17:02:19]*WM: Using last saved values, should be faster
[17:02:28]*WM: Connection result: 
[17:02:28]*WM: 3
[17:02:28]*WM: IP Address:
[17:02:28]*WM: 192.168.2.170
[17:02:28]connected...yeey :)
[17:02:28]saving config
[17:02:28]{"mqtt_server":"192.168.2.8","mqtt_port":"1883","mqtt_user":"mqtt","mqtt_pass":"6566"}*WM: freeing allocated params!
[17:02:28]OpenMQTTGateway mac: 
[17:02:28]CC:50:E3:60:EB:A2
[17:02:28]OpenMQTTGateway ip: 
[17:02:28]192.168.2.170
[17:02:28]1883
[17:02:28]Connecting to MQTT by IP adress
[17:02:28]192.168.2.8
[17:02:29]BLEinterval btw scans
[17:02:29]55555
[17:02:30]ZgatewayBT HM1X setup done 
[17:02:30]MQTT_MAX_PACKET_SIZE
[17:02:30]1024
[17:02:30]Setup OpenMQTTGateway end
[17:02:30]MQTT connection...
[17:02:30]Connected to broker
[17:02:30]Subscription OK to the subjects
[17:02:30]Hey I got a callback 
[17:02:30]MQTTtoBT json set
[17:02:30]WL set
[17:02:30]3F59C8828829
[17:02:30]0
[17:02:30]1
[17:02:30]0
[17:04:19]Uptime (s)
[17:04:19]120
[17:04:19]Remaining memory
[17:04:19]40584
[17:04:19]RSSI
[17:04:19]-64
[17:04:19]SSID
[17:04:19]JamesESP
[17:04:19]Activated modules
[17:04:19]BT
[17:05:06]4f4b2b444953413afbc96899bb9800a50e0d09536d6172745068696c616d700d0a4f4b2b4449534345
[17:05:06]Creating BLE buffer
[17:06:01]4f4b2b444953413afbc96899bb9800a51d0201060b02031802180f180a18f5fe0dff5048494c414dfbc96899bb980d0a
[17:06:01]Creating BLE buffer
[... 313 lines cut ...]
[17:32:19]-64
[17:32:19]SSID
[17:32:19]JamesESP
[17:32:19]Activated modules
[17:32:19]BT
[17:32:52]4f4b2b444953413a7b4ad66f00d901b80a02010603ff3330020af60d0a
[17:32:52]4f4b2b444953413a61077118fa1c02c00f02011b0bff4c0009060341c0a802150d0a
[17:32:52]Creating BLE buffer
[17:33:48]4f4b2b444953413a2c6b6d40703f02af1f1eff0600010920028c7f5e09862b719aba0ebfd98d3fee7f6956ac0c9eca640d0a
[17:33:48]Creating BLE buffer
[17:34:19]Uptime (s)
[17:34:19]1920
[17:34:19]Remaining memory
[17:34:19]40016
[17:34:19]RSSI
[17:34:19]-62
[17:34:19]SSID
[17:34:19]JamesESP
[17:34:19]Activated modules
[17:34:19]BT
[17:34:43]4f4b2b444953413a220be654176803a40e02011a0aff4c0010050110f6ea7f0d0a
[17:34:43]Creating BLE buffer
[17:35:39]4f4b2b444953413a7e3cd55bdf7e03c00e02011a0aff4c001005071420c8340d0a
[17:35:39]Creating BLE buffer
```