1technophile/OpenMQTTGateway #413: Pinout documentation D3=1

h4nc, 2019-06-20:
This isn't really a bug. More like an improvement of the documentation.

In the Pinout section some pin names are decribed like

D3=1

or D2=0

I confused? What does that mean? Do I connect it to D3? What does =1 mean? Could you leave a note about that at the bottom of the list. 

Thanks!

akasma74, 2019-06-20:
yeah, it's a bit unclear initially.

have a look at this [picture](https://escapequotes.net/wp-content/uploads/2016/02/d1-mini-esp8266-board-sh_fixled.jpg)

D means digital pin, and you can see those D1-D8 pins on the board.
For different boards (Arduino, ESP8266, ESP32) they are different.
On the other hand, the code accesses not physical but logical pins regardless the board.
And at the library level they translate logical pins (marked as Arduino, probably for historical reasons) into physical pins of your particular board.
So if they say use pin 5, that means D1 for Wemos pictured, and D3=0 (i.e Arduino pin 0)

Does that make sense?

h4nc, 2019-06-20:
Makes sense. Thanks!

closed 2019-06-20 by h4nc

h4nc, 2019-06-20:
@akasma74 One example says D3=1 (Arduino Mega).

I did not find a pinout that says D3 is pin 1.

akasma74, 2019-06-20:
Mate, it's always great to refer to something real, like give a link or something..
Can't help without it.

h4nc, 2019-06-20:
Sure.

https://www.pinterest.at/pin/511932682643023502/

Looks like D2 = 6 ?