1technophile/OpenMQTTGateway #415: v 08 esp8266 rfm69

clubbit, 2019-06-23:
Hello! I use openMQTTGateway version 0.8. Downloaded entire folder "master" libraries used and folders "DSHI." flashed esp8266 with rfm69 module. as the manual was raskomentiruyte need the drain in the "user_config" . to test the transmitter used your code "rfm69sender". in the monitor of port writes Sending 1,-84,0 !!!!!!!!!! Fail .What could be the reason?. In the topic mqtt comes only the status of the gateway and version.

1technophile, 2019-06-23:
Hello,

There was some issues on v0.8 related to rfm69. Could you use the development branch with v0.9.2 beta.

clubbit, 2019-06-23:
well I'll try

clubbit, 2019-06-23:
unfortunately not working! and in version 0.9.2

1technophile, 2019-06-23:
if you want some help you need to give more details.
Please follow the issue template first.

clubbit, 2019-06-24:
does not work with RFM96 ver. 0.9.2.
I use NodeMcu v3 (ESP8266E) I connect according to a manual. In the configuration file "user_config" I uncommented the line "#define ESPWifiManualSetup true". I wrote my WI-FI and MQTT login-password settings. Comment out all modules except # ZgatewayRFM69.

In the config_RFM69 configuration file
in the line #define IS_RFM69HCW true changed to false. And changed the const char PROGMEM ENCRYPTKEY [] = "sampleEncryptKey"; on "1234567891234567". More than anything faulty.
The port issues:
MQTT connection ...
Connected to broker
Subscription OK to the subjects
Uptime (s)
120
Remaking memory
36832
RSSI
-60
SSID
(^. ^) (^. ^) (^. ^)
Activated modules
RFM69
When sending a message to the topic home / OpenMQTTGateway / commands / to the port shows "Hey I got a callback"

In Node-RED created topics:

home / OpenMQTTGateway / version
home / OpenMQTTGateway / LWT
home / OpenMQTTGateway / commands / #
home / OpenMQTTGateway / RFM69toMQTT
home / OpenMQTTGateway / commands / MQTTtoRFM69.
The gateway successfully connects to WI-FI and subscribes to topics. Node-RED receives a message about the device version and online status.

Transmitter:
Arduino nano with RFM 69 module. Code taken by your senderRFM69
connected by manual. Also corrected "sampleEncryptKey" to "1234567891234567" and # define IS_RFM69HCW false.

The port issues:
Arduino RFM69HCW Transmitter

Transmitting at 433 MHz
Network 200 Node 1

Sending 0, -128.0 !!!!!!!!!! Fail

clubbit, 2019-06-24:
Here are more events from the port:
mounting FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"172.20.20.104","mqtt_port":"1883","mqtt_user":"user","mqtt_pass":"Pass"}
parsed json
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to saved AP: (^.^)(^.^)(^.^)
*WM: [3] Using Password: pass
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 172.20.20.108
connected...yeey :)
saving config
{"mqtt_server":"172.20.20.104","mqtt_port":"1883","mqtt_user":"user","mqtt_pass":"pass"}*WM: [3] freeing allocated params! 
*WM: [3] unloading 
OpenMQTTGateway mac: 
DC:4F:22:1C:23:5D
OpenMQTTGateway ip: 
172.20.20.108
1883
Connecting to MQTT by IP adress
172.20.20.104
ZgatewayRFM69 initialization failed
ZgatewayRFM69 Listening and transmitting at
433
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
MQTT connection...
Connected to broker
Subscription OK to the subjects

1technophile, 2019-06-24:
> ZgatewayRFM69 initialization failed

Could you verify your wiring, it seems that the ESP can't initialize the RFM69

clubbit, 2019-06-25:
Checked Wires are fine. The manual correctly indicated pin NSS --- D1, and DIO0 --- D8? There are more options why the module does not start? Tried to use another module, the same result

1technophile, 2019-06-25:
could you post a picture of your wiring?

clubbit, 2019-06-26:
ок!
[image]
[image]

clubbit, 2019-06-30:
Is the wire okay? correct connection?

1technophile, 2019-08-05:
Did you checked with a multimeter that there is no short circuit between the cables?

rarenu, 2019-08-23:
Hi, I am 72 years old retired Telecom Engineer and my hobby is IoT. I am really fascinated by your versatile Gateway project. I am using RFM69HW in my trials and getting response in my MQTT server "home/OpenMQTTGateway/LWT" and  "home/OpenMQTTGateway/SYStoMQTT" messages. 
But, I am not getting anything on RFM69's transmission or reception, in my serial monitor. Do I have to use only RFM69HCW or will it work with HW version also. Please advise me what are the options to be selected in config files for RFM69HW and any simple way to confirm it is working. Please help me. Thank you for the wonderful project with Great Regards.

1technophile, 2019-08-24:
Hello,
Glad you like it ! 
I did my test with an HW version. It should work with it. Could you post the content of the serial monitor ?  Are you using v0.9.2 ?

rarenu, 2019-08-24:
Thank you for your reply. Yes I am using v0.9.2 and serial monitor out put
is:
ABCD
.
WiFi ok with manual config credentials
OpenMQTTGateway mac:
4C:11:AE:0D:1F:BC
OpenMQTTGateway ip:
192.168.1.254
1883
Connecting to MQTT by IP adress
192.168.1.120
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
client disconnected
MQTT connection...
Connected to broker
Subscription OK to the subjects

Please advice.
With Great Regards,
Ragothaman R.

rarenu, 2019-08-24:
Node_red Debug Node output for your views:
8/24/2019, 5:30:54 PMnode: d265f873.bd3dc8
<http://127.0.0.1:1880/#>home/OpenMQTTGateway/SYStoMQTT
: msg.payload : string[114]
"{"uptime":600,"freeMem":46320,"rssi":-57,"SSID":"FBI","ip":"192.168.1.254","mac":"4C:11:AE:0D:1F:BC","modules":""}"
8/24/2019, 5:32:54 PMnode: d265f873.bd3dc8
<http://127.0.0.1:1880/#>home/OpenMQTTGateway/SYStoMQTT
: msg.payload : string[114]
"{"uptime":720,"freeMem":46320,"rssi":-57,"SSID":"FBI","ip":"192.168.1.254","mac":"4C:11:AE:0D:1F:BC","modules":""}"

clubbit, 2019-08-25:
> Did you checked with a multimeter that there is no short circuit between the cables?

Hello!thank you for your project it is very cool.Thanks for your help. it was in esp8266 when I changed it to another all earned like clockwork. Apparently defective ESP

1technophile, 2019-08-25:
@rarenu coud we continue the discussion in https://community.openmqttgateway . It will be more appropriate than a github.

closed 2019-08-25 by 1technophile

rarenu, 2019-08-25:
Please reply, if I make this "#define IS_RFM69HCW "   false, am I selecting RFM69HW? 
or what else are to be set to use RFM69HW inplace of RFM69HCW?

### Important Request:
Please share the serial output of ESP8266 while using v0.92, if RFM69HW works fine so that I will try all options of RFM69 setup, till I get the result.

rarenu, 2019-08-28:
> I used the line #define IS_RFM69HCW true, the contacts are connected to
> manual.

Thank you, I will try again.
Best regards

clubbit, 2019-08-28:
I used the line #define IS_RFM69HCW true, the contacts are connected to manual.