1technophile/OpenMQTTGateway #418: ESP32 and I2C Modules

wbaars, 2019-07-03:
I am trying to use a BME280 sensor with my ESP32 from [Banggood](https://www.banggood.com/Geekcreit-30-Pin-ESP32-Development-Board-WiFibluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1461896.html) , but the Wiki does not list the pins to use on an ESP32.
I tried to use pins D21 for SDA, and D22 for SCL [Pinout here](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/) but its not working.
Still get "Failed to read from Weather Sensor BME280!"

Where in the code can I set the pins to use? Did anyone get this working?
Does OpenMQTTGateway support I2C modules on an ESP32?

Am using the latest version, and BT (for my LYWSD02, which works fine).

1technophile, 2019-07-06:
Hello, if you try to upload the examples given with the bme280 library, do you succeed in getting the values.

wbaars, 2019-07-10:
Hi,
That is what I tried to do.
I used this example: [Wiki page](https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-I2C-Modulesl)
But that does not list the ports on an ESP32.

closed 2019-09-15 by stale[bot]