1technophile/OpenMQTTGateway #419: Mi Jia (MJ_HT_V1) reading

eos1d3, 2019-07-06:
I am trying to use ESP32 with PlatformIO (using Platform Espressif 32, 1.9.0) to scan
Mi Jia. 

But from ZgatewayBT.ino, I find there is `advertisedDevice.getServiceDataCount()` and `advertisedDevice.getServiceData(j) ` This means there are more than just one data in the callback. But there current ESP32 library does not have these, with only advertisedDevice.getServiceData(). 

That is why I can get UUID 0000180f-0000-1000-8000-00805f9b34fb data only. But  no sensor data.

And would you please tell me which ESP library are you using? And how to read those data?

Thanks!

I am using example code to scan:

```
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 5; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}
```

And I get only this:
`Advertised Device: Name: MJ_HT_V1, Address: 58:2d:34:34:0f:a1, serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb `

1technophile, 2019-07-06:
Hello,

You can use ble arduino library from the lib folder and remove the BLE one from esp32 environment (this is detailled into the wiki)
. I'm using the fork of @fguiet 
https://github.com/nkolban/ESP32_BLE_Arduino/pull/24

eos1d3, 2019-07-06:
Hi,

Many thanks for your help. I am able to read the raw data from Mi Jia now.
Never expect ESP32 BLE library does not work.

eos1d3, 2019-07-06:
After using @fguiet fork for a while, everything works except one strange bug.

I have LYWSD02 and also MJ_HT_V1. When using ESP Arduino library and their scan example code, I can see both device names using:
`        Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
`

After switching to fguiet library and its scan example, I can only see MJ_HT_V1 device name. LYWSD02 name is always empty, `advertisedDevice.toString()` does not have its name, and `advertisedDevice.haveName()` is always false.

It is not a big issue, but is this normal?

eos1d3, 2019-07-07:
Hi @1technophile

@fguiet repository is based on old ESP32 library which causes these problem:

1. LYWSD02 name is missing
2. MJ_HT_V1 has two services but `getServiceDataCount` returns 1

But his PR is just perfect. I use ESP32 recent BLE library and change with his PR, now all these problems are fixed.

The update version is here:
https://github.com/eos1d3/ESP32_BLE

Tested with LYWSD02 and MJ_HT_V1. All work fine.

1technophile, 2019-07-07:
Nice ! Thanks for the lib. I will refer yours on next release.

1technophile, 2019-07-18:
with platformio integration I have added the[ link to the modified library](https://github.com/1technophile/OpenMQTTGateway/blob/8471585d3d3df76a2d931b201543fd89f2e8d5f4/platformio.ini#L63) in platformio.ini

closed 2019-07-18 by 1technophile

svwhisper, 2019-08-22:
I think you also need to add the link to the modified library in section [env:ttgo-lora32-v1].  I had the same compile error until I did that.

1technophile, 2019-08-22:
Are you using ble with LORA, maybe this is why you had the error?