1technophile/OpenMQTTGateway #420: New "All in One" PCB

ebrithiljonas, 2019-07-08:
I'm currently working on a PCB for OpenMQTTGateway, which will feature some of the modules which I think are the most important. Now my question really is, if there are other people who are interested in such a solution and when yes, if there is demand for more modules to add to the PCB.

**Currently there are these modules:**
-- RF (Send and Receive)
-- IR (Send and Receive)
-- DHT11

**And here is how the Board looks currently:**
[https://i.imgur.com/c3cOetN.png](url)

Of course once I'm a little bit further with the design and some people are interested, I will upload the design (Made with KiCAD)

AllenSlaghuis, 2019-07-08:
I would be interested. Please check the url. It does not take you to the link.

ebrithiljonas, 2019-07-08:
Ok, thanks, I fixed the link. Would also be interested in other modules, like PIR or Bluetooth?

QuagmireMan, 2019-07-10:
Maybe have spots for all the possible configurations? And people can add their own sensors.

https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP8266

I would be interested... No more breadboards! :smile:

QuagmireMan, 2019-07-11:
Also if people are going to be adding their own components maybe swap the surface mount resistors out for a through hole for ease of install, unless these are shipping with the parts on them.

This is just the PCB correct?

ebrithiljonas, 2019-07-11:
OK, when I finish the schematic, I will make 2 versions, TH and SMD, so you can choose. 
So you mean make it possible to add all Modules, even if they use the same pins, and then everyone can decide what modules they want to use?

AllenSlaghuis, 2019-07-11:
The bluetooth module add on sounds good.

QuagmireMan, 2019-07-13:
> OK, when I finish the schematic, I will make 2 versions, TH and SMD, so you can choose.
> So you mean make it possible to add all Modules, even if they use the same pins, and then everyone can decide what modules they want to use?
> 
> Oh and yeah, this is just a PCB.

Yep.  

You could also have a set of jumper locations allowing for users to move which pin goes to which sensor. It could go to something like this [https://www.amazon.com/dp/B00GYRDA82/](url) which might help future-proof any added sensors.  This item could be skipped and people could just solder as well.

broekema, 2019-07-18:
I did something similar recently. The only major difference I see is that I used a Wemos D1 mini as basis, not a nodeMCU. A couple of remarks:

- I believe the IR LEDs are driven purely from the IO pins of the micro-controller. This will significantly limit the output of the LEDs. Much better to drive them using a transistor, as shown on the wiki page.
- Powering via the USB interface of the micro-controller may cause issues, since there is only a limited supply available. Better to use an external supply and power the micro-controller via the +5V pin.
- I see no smoothing and/or decoupling capacitors. This may impact stability.

For reference, here's my attempt (Wemos D1 mini, with DHT22, BME280, DH1750, IR send and receive, andd RF send):

[image]

ebrithiljonas, 2019-07-18:
Thank you for your feedback, the IR LEDs are powered by a single transistor which is hidden under the nodemcu. Powering over USB should not be a big problem since almost all USB charger can handle 1A at least. And for the capacitors, I will look into it, thank you for the tip. But first all of the modules need to arrive at my house, so I can wire and test the circuit on a breadboard, before I make the final PCB design.

broekema, 2019-07-18:
A small clarification on my second point. I don't mean the capacity of the charger itself, but rather the capability of the nodemcu to provide enough power to supply all the components that you connect to it. The nodemcu contains a voltage regulator that supplies the micro-controller and the 5V and 3.3V pins with power. This will have limited capacity and will probably get hot when loaded significantly.

ebrithiljonas, 2019-07-20:
OK, I will change the way I power the PCB, but the transistor can handle all the LEDs. The resistors are dimensioned that each LED draws 25 mA max, and the transistor has a max forward current of 1A.
[image]

Here is the current schematic, if you want to have a look at it and maybe find any problems with it. Thank you so far for all the help and tips :)

broekema, 2019-07-20:
You're very welcome. Thank you for pointing out KiCAD, my designs so far have been in EasyEDA (which works fine, but has its limitations). The schematic above looks excellent, but I do have some small comments.

- with respect to BCP56: fair enough that should work. Only comment I have is that there is no through hole version available that I could find.
 - TSOP21xx should work fine with 5V
- HM-10 takes between 3.7-6V power, but the TX and RX pins are 3.3V. At least the TXD pin should be connected through a level-shifter (or use a voltage divider). It appears from experience on the net that the RXD pins works fine without (but level-shifting packages usually come with multiple channels, so why not use them?).

Finally a point for consideration: I think it is likely that a circular PCB will cost. Consider that the PCB will be produced in rectangular panels, with individual PCBs cut out. A circular design will be more difficult to cut automatically and will result in more waste.

ebrithiljonas, 2019-07-20:
I will make a through hole and a SMD Version of the board, once the schematic is testet.
The HM-10 has 3.3V logic, and the ESP8266 also, so there is no need of a level shifter if these logic levels are correct.
I planned on ordering the PCBs on JLCPCB and they don't charge extra for a round design.

Ans yes, definitely check out KiCAD, IMHO best ECAD, plus its free and open source.

broekema, 2019-07-20:
You are absolutely right, the ESP8266 is 3.3V so the level shifter is not needed.

Purclewan, 2019-07-23:
Great work. Very interested in a couple of boards when you have them ready.  

One thought, I found the little spring antenna a bit limiting in terms of range and sensitivity. on all my breadboard rigs I have soldered connections for a small 433mhz whip antenna and achieved much better results.   Not sure how practical it would be or if there is space but if you could have pads to solder SMA connectors for those who want to use bigger antenna  that would be useful.

ebrithiljonas, 2019-07-23:
Thats a pretty good idea, I will look into it if you can connect the antenna to the pcb and will add some sma connectors.

h4nc, 2019-07-29:
Just want to let you know that I made something similar:

https://community.openmqttgateway.com/t/buy-a-ready2use-omg-device/568

ebrithiljonas, 2019-07-31:
OK, thank you, but since your design isn't open, its not very useful for my development.

I'm almost finished with my first designed, I checked everything on the breadboard, and I'm going to order the PCB soon. The design is on github [here](https://github.com/ebrithiljonas/OMG_PCB), so we can also move the discussion and feature requests to there, so I'm going to close this "issue" here.

closed 2019-07-31 by ebrithiljonas

h4nc, 2019-07-31:
@ebrithiljonas I‘m curious what is the J2 connector for?

ebrithiljonas, 2019-07-31:
Oh thanks for the reminder, this is the Power input, totally forgot to add silkscreen there.

h4nc, 2019-07-31:
I like that you put all modules at one pcb, but I think the DHT11 module will not read correct values, unless you want to measure the temperature within the case.

Even if it’s without a case, it will probably pick up some heat of the esp8266 module. In my experience those heat up.

ebrithiljonas, 2019-07-31:
yeah that's true but there is always the possibility of using cables instead of directly soldering it to the board.