1technophile/OpenMQTTGateway #421: MQTT broker on ESP device

happytm, 2019-07-11:
Kindly requesting MQTT broker on esp device gateway itself to avoid cloud broker or local broker on linux device. Link to sample library is below :

https://github.com/martin-ger/uMQTTBroker

This can be very helpful for creating standalone IOT network and if possible programmed to setup broker in the code automatically without user configuration for MQTT broker.

Thanks

1technophile, 2019-08-05:
Interesting, will you participate in tests if I implement a beta version with this?

happytm, 2019-08-06:
I am always ready to test any new ideas. I do have 20 esp8266 devices on hand to test the limit.

Thanks

happytm, 2019-08-06:
Also one more interesting protocol missing in this firmware is ESP-Now receiver to receive data over ESP-Now protocol from bunch of low battery powered esp devices. These remote devices are mostly sleeping and every few minutes send sensor data. Example video is below:

https://www.youtube.com/watch?v=6NsBN42B80Q

For larger range this video explain it well:
https://www.youtube.com/watch?v=yCLb2eItDyE&feature=youtu.be

Thanks

happytm, 2019-08-08:
I am trying to build a low power network mix of some IR devices , Some RF devices and some esp8266 devices with ESP-Now devices explained in above video.Now my question is:

Is it easily possible to remove all Wifi and MQTT related code from your software and in place of MQTT protocol use serial protocol so gateway can connect to another device (only MQTT gateway with built in uMQTT broker ) hard wired to main gateway?

The other hard wired deviceshould have MQTT part of gateway only and basically it will be running SerialtoMQTT and MQTTtoSerial functions and publish/receive it to/from broker which could be run on same device using uMQTT broker.Now the gateway will transfer IRtoSerial,RFtoserial and ESP-Nowtoserial or vice versa just like you have it working it with MQTT right now.This way you are running only 1 device with high power consuming wifi device and all of the other devices on network could be low power devices without using wifi. ESP-Now protocol could be used very efficiently for low power but little longer range devices.Unfortunately we need to separate wifi & MQTT part of code from this main gateway and put it on second gateway as espnow and Wifi can not work together. 

Please let me know what you think about it.

Thanks

happytm, 2019-08-08:
To make it clear above comment following is my proposal:

[image]

Thanks

1technophile, 2019-08-13:
I never put my hands on esp now but i m quite interested. For my info how much current does an active esp draws in this case ?

happytm, 2019-08-13:
It is same current at 80 ma but as tested in following video it only is up for 220 ms vs 8000 ms with mqtt using wifi connection. Andreas in his video compares different battery power consumption for Wifi, esp-now & lora.

https://www.youtube.com/watch?v=6NsBN42B80Q

Thanks

1technophile, 2019-08-18:
If you want to have device that emits IR or RF you will need to have them continuously listening. And by the way not corresponding to your the low power characteristic.
Do you want to have only IR/ RF to something direction, the reverse or the both?

happytm, 2019-08-18:
The idea is to have network of low battery power devices using  Attiny85 or Atmel328P with IR, RF protocol or ESP8266 device using ESP-Now protocol to send data to main gateway and the main gateway transferring these data to OpenMQTT gateway by serial or I2C interface.IR devices are most suitable for indoor in the same room like door & window status sensors or motion sensors for security (using interrupt wake up) where mains power may not be available. For outdoor sensors like BME280 ,water tank level sensor, light sensor,rain sensor,soil moisture sensor in garden & outdoor motion sensors at gate ESP-Now protocol could be more suitable for longer range.In most cases these devices are sending unit and they don't have to receive anything. They send data every 15 or 20 minutes and up for only 200 to 250 ms and go back to sleep saving battery and possibly get charged with small 5V solar panel so in theory we have to never change battery on them. If any outdoor device has actuator on them to turn something on or off they have to be up all the time but they require more power anyway to supply power to the device being controlled. so they can not be truly low power device.

In above scenario there is only 1 wifi device (OpenMQTT gateway) communicating to outside world via internet and local network is still doing it's job without any wifi network. It is also more reliable and simpler then having many devices with each having IP addresses of their own trying to connect to gateway. Best example of above scenario  I can find is MySensors framework but I believe it is limited to NRF24 (RF) protocol only.

I do not know how difficult it will be to implement main gateway with IRtoSerial, RFtoSerial & ESP-Now to serial and MQTT gateway with SerialtoMQTT or I2CtoMQTT .

Thanks.

Thanks

happytm, 2019-08-19:
I found following project which uses ESP-Now protocol and serial transfer of data to MQTT gateway but there is no other protocol like IR,RF etc.:

https://github.com/gmag11/EnigmaIOT

Thanks

1technophile, 2019-08-19:
I understand the concept of :
BME280 or others sensor to ESPNow -> ESPNowtoMQTT
but not this one
RFtoESPNow or IRtoESPNow -> ESPNowtoMQTT
Indeed to receive RF or IR the device can't be a low power one. Do I miss something?

happytm, 2019-08-19:
bThe receiver part should be only used on main gateway described above in image which can receive either IR,RF,Lora or ESP-Now protocol and main gateway is always up so it has to be plugged into mains power next to OpenMqtt Gateway connected by Serial or I2C. The baqttery powered sendor unit could use any protocol out of IR,RF,Lora or ESP-Now.You have everything already implemented except ESP-Now protocol.The EnigmaIOT project I linked above does ESP-Now protocol so it is matter of including that protocol into your software.It already have serial connection between main gateway and MQTT gateway.

Sorry for confusion.

Thanks

happytm, 2019-09-11:
I have been testing ESP_Now ,Probereqeust and IR  protocols using your firmware with esp8266 for more than a week and it is working fine. It receive data from another esp8266 running all three protocols to send sensor data. All three are very fast compared to wifi/mqtt.

Thanks

1technophile, 2019-09-11:
Do you have some measures to gives so as to have some idea?

happytm, 2019-09-11:
Yes this was something I was discussing here for some time.If you want to test the feature I uploaded your Gateway I modified to work with espnow and there is sender sketch which send data in three different protocols to gateway.Below is link:

https://github.com/happytm/Triple-Sender/tree/master/Triple%20Sender

Thanks

hazee007, 2019-12-20:
hey @happytm  do you have any idea of how how many sender devices can be connected to the gateway using the espnow protocol and also are using using just esp32 or esp8266 or both of them? i asked that cause i had issues the last time i tried using esp32 and esp8266 to connect to each other when using the espnow protocol.

happytm, 2019-12-20:
Actually I have found more simpler and efficient way of creating wireless sensor network. Please look at my new efforts at   https://github.com/happytm/BatteryNode. There are 2 sketches - 1 for receiver device which also have uMQTT Broker and presence detector/ person locator. another sketch is for sender device. Right now they are used with ESP8266 devices only.

Thank you.

1technophile, 2023-06-12:
Not planning to integrate a uMQTT broker into OpenMQTTGateway for the moment

closed 2023-06-12 by 1technophile