1technophile/OpenMQTTGateway #425: MQTT disconnection - Socket error

patrul, 2019-07-19:
I am using this great gateway on an DOIT ESP32 DEVKITV1 for the routing of BT data for mi plants and room thermostats. I am using ZgatewayBT an it is the only one activated.

After some minutes of connection and sending data through MQTT, I can see with MQTT Explorer, the MQTT client stops working.

The log message on HassIO Official MQTT Broker is:

`Socket error on client OpenMQTTSalon, disconnecting.
`

I have tried with different values of MQTT_KEEPALIVE and MQTT_SOCKET_TIMEOUT on PubSubClient.h but the result is the same.

The broker I am using for other duties and work correctly.

From the MQTT client side, if I use the serial monitor what I can see is several connection tries but these are unsuccesful

I have been looking for similar behaviour but nothing turns me to the right direction.

1technophile, 2019-07-19:
Could you indicate the version you are using?

patrul, 2019-07-19:
Sure, the last one: 0.9.1

1technophile, 2019-07-19:
on the development branch I have corrected a reconnection issue when there a broker disconnection occurs,  could you test it?

patrul, 2019-07-19:
Yes, I will try
One test before, I have just changed the QoS to 2 and disables retain, let's see

patrul, 2019-07-19:
It didn't work que QoS adjustment

Now I am trying the dev branch but it gives me an error

After importing libraries from master branch, it says:

```
C:\Users\ricar\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

exit status 1
Error compilando para la tarjeta DOIT ESP32 DEVKIT V1.
```

1technophile, 2019-07-19:
Could you try with platformio?
The process is the following:
-download VScode
-install platformio extension from VScode
-download OpenMQTTGateway code
-with VScode open the folder OpenMQTTGateway 
-update user_config.h
-upload

Platformio take care about gathering the good libraries for you.

patrul, 2019-07-19:
I am not sure what I am doing wrong, but I imported Arduino Project and after fighting with the tool, it reports these errors:

`> Executing task: C:\Users\ricar\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing nodemcuv2 (platform: espressif8266; framework: arduino; board: nodemcuv2)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.2.2 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-mkspiffs 1.200.0 (2.0), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing PubSubClient
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\ricar\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\ricar\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ricar\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\ricar\.platformio\platforms\espressif8266\builder\main.py", line 212:
    target_elf = env.BuildProgram()
  File "C:\Users\ricar\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ricar\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 124:
    _build_project_deps(env)
  File "C:\Users\ricar\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 49:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\ricar\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\ricar\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\Users\ricar\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\ricar\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 353:
    force=force)
  File "c:\users\ricar\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 724:
    track=True)
  File "c:\users\ricar\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 509:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "c:\users\ricar\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 49:
    silent)
  File "c:\users\ricar\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 63:
    self.check_client()
  File "c:\users\ricar\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 130:
    "Please install Git client from https://git-scm.com/downloads")
===================================================================================================================================== [ERROR] Took 1.17 seconds =====================================================================================================================================  

Processing esp32dev (platform: espressif32; framework: arduino; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-mkspiffs 2.230.0 (2.30), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing PubSubClient
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\ricar\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\ricar\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\ricar\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
[... cut ...]
Environment esp32dev    [FAILED]
Environment atmega      [FAILED]
Environment uno         [FAILED]
=============================================================================================================================== 4 failed, 0 succeeded in 4.45 seconds ===============================================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.`

patrul, 2019-07-19:
After removing non used boards, the outcome is almost the same as with Arduino IDE

> Executing task in folder OpenMQTTGateway-development: C:\Users\ricar\.platformio\penv\Scripts\platformio.exe run <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.9.0 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 46 compatible libraries
Scanning dependencies...     
Dependency Graph
|-- <NewRemoteSwitch>    
|-- <WifiManager> 0.14   
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0   
|-- <Wire> 1.0.1
|-- <ESP32 BLE Arduino> 0.4.16
|-- <A6lib>
|-- <rc-switch> 2.6.2
|-- <DHT sensor library> 1.3.0
|   |-- <Adafruit Unified Sensor> 1.0.2
|-- <PubSubClient> 2.7
|-- <FS> 1.0
|-- <IRremote> 2.3.3
|-- <ArduinoJson> 5.13.3
|-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <esp8266_mdns>
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|   |-- <esp8266_mdns>
|   |   |-- <WiFi> 1.0
|-- <Adafruit TSL2561> 1.0.2
|   |-- <Wire> 1.0.1
|   |-- <Adafruit Unified Sensor> 1.0.2
|-- <EEPROM> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <LoRa> 0.5.0
|   |-- <SPI> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <SPI> 1.0
|-- <SparkFun BME280> 1.2.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <RFM69>
|   |-- <SPI> 1.0
|-- <Adafruit Unified Sensor> 1.0.2
|-- <IRremoteESP8266> 2.5.6
|-- <ESPiLight> 0.16.0
Compiling .pio\build\esp32doit-devkit-v1\src\main.ino.cpp.o
Archiving .pio\build\esp32doit-devkit-v1\lib41c\libWiFi.a
Archiving .pio\build\esp32doit-devkit-v1\lib380\libDNSServer.a
In file included from C:/Users/ricar/Documents/Arduino/OpenMQTTGateway-development/src/main.ino:58:0:
src\config_BT.h:37:11: warning: ISO C++11 requires whitespace after the macro name
 #define HM-10
           ^
Compiling .pio\build\esp32doit-devkit-v1\lib592\WiFiManager\WiFiManager.cpp.o
In file included from C:/Users/ricar/Documents/Arduino/OpenMQTTGateway-development/src/main.ino:119:0:
lib/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
In file included from lib\WiFiManager\WiFiManager.cpp:13:0:
lib\WiFiManager\WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\lib723\Wire\Wire.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libec0\ESP32_BLE_Arduino\BLE2902.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libec0\ESP32_BLE_Arduino\BLE2904.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\libec0\ESP32_BLE_Arduino\BLEAddress.cpp.o
*** [.pio\build\esp32doit-devkit-v1\src\main.ino.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib592\WiFiManager\WiFiManager.cpp.o] Error 1
===================================================================================================================================== [ERROR] Took 5.09 seconds =====================================================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

1technophile, 2019-07-19:
try maybe to delete the content of .pio and relaunch the build. Sometimes it keeps old versions that conflict with the current build

ColinShorts, 2019-07-19:
You shouldn't need to delete anything, just setup the user_config.h file after a fresh git clone (switch to the development branch as required) and run:

pio run -e esp32dev

(Or platformio instead of pio depending on your environment)

patrul, 2019-07-20:
After cloning `version = 0.9.2beta` I am geting the same from brocker, 

```
1563607548: New client connected from 192.168.1.182 as OpenMQTTSalon (p2, c1, k15, u'mqtt-user').
1563607633: Client OpenMQTTSalon has exceeded timeout, disconnecting.
```

1technophile, 2019-07-20:
Could you post also your serial monitor extract in the meantime you see this on the broker log.

patrul, 2019-07-20:
Sure, first boot last almost 10 minutes after changing KEEPALIVE (updated my last comment)

Second boot with serial monitor activated, reports this and only last 2 minutes working. After that, timeout from MQTT broker

```
> Executing task: C:\Users\ricar\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available ports:
---  1: COM3                 u'TUSB3410 Device (COM3)'
---  2: COM6                 u'Silicon Labs CP210x USB to UART Bridge (COM6)'
--- Enter port index or full name: 2
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
mounting FS...
E (36) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:52] begin(): Mounting SPIFFS failed! Error: -1
failed to mount FS
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [1] AutoConnect
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] Connecting to saved AP: MiFibra-4770
*WM: [3] Using Password: ********
*WM: [3] WiFi station enable
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.182
connected...yeey :)
saving config
[E][vfs_api.cpp:22] open(): File system is not mounted
{"mqtt_server":"192.168.1.11","mqtt_port":"1883","mqtt_user":"mqtt-user","mqtt_pass":"******"}*WM: [3] freeing allocated params! 
*WM: [3] unloading
OpenMQTTGateway mac:
A4:CF:12:75:B3:D8
OpenMQTTGateway ip:
192.168.1.182
1883
Connecting to MQTT by IP adress
192.168.1.11
BLEinterval btw scans
55555
ZgatewayBT multicore ESP32 setup done
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
BT Task running on core 0
MQTT connection...
Connected to broker
Subscription OK to the subjects
Creating BLE buffer
device detected
3CADBE17F67B
BLErssi
-83
txPower
-59
BLE DISTANCE :
12.61
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"3c:ad:be:17:f6:7b","manufacturerdata":"L","rssi":-83,"distance":12.61001}
[... 163 lines cut ...]
BLE DISTANCE :
ovf
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"f8:77:b8:62:5f:91","manufacturerdata":"u","rssi":-89,"txpower":9,"distance":8.94283e9}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/F877B8625F91
{"id":"f8:77:b8:62:5f:91","manufacturerdata":"u","rssi":-89,"txpower":9,"distance":8.94283e9}
Creating BLE buffer
device detected
F0B5D1838682
BLErssi
-91
txPower
-59
BLE DISTANCE :
25.52
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-91,"distance":25.51913}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/F0B5D1838682
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-91,"distance":25.51913}

```

patrul, 2019-07-20:
More logs ...
After some minutes of proper work:

```
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-92,"distance":27.75274}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/F0B5D1838682
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-92,"distance":27.75274}
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401b88b7  PS      : 0x00060330  A0      : 0x800e47ea  A1      : 0x3ffdc920
A2      : 0x3ffdde34  A3      : 0x00000011  A4      : 0x3ffddf04  A5      : 0x3ffdd118  
A6      : 0x3ffddc1c  A7      : 0x3ffdca0c  A8      : 0x3ffdde3c  A9      : 0x3ffdc900  
A10     : 0x3ffdde24  A11     : 0x0000002c  A12     : 0x3ffdd0dc  A13     : 0x3ffdc99c
A14     : 0x3ffdd0dc  A15     : 0x3ffdca0c  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000011  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x401b88b7:0x3ffdc920 0x400e47e7:0x3ffdc940 0x400e487b:0x3ffdc960 0x400e48fd:0x3ffdc990 0x400e4bd1:0x3ffdc9c0 0x400d2e7b:0x3ffdca00 0x400d2f1c:0x3ffdca60 0x4008e189:0x3ffdcab0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
mounting FS...
E (35) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:52] begin(): Mounting SPIFFS failed! Error: -1
failed to mount FS
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [1] AutoConnect
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] Connecting to saved AP: MiFibra-4770
*WM: [3] Using Password: zUYKCNdj
*WM: [3] WiFi station enable
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_DISCONNECTED
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [1] AutoConnect: FAILED
*WM: [2] AccessPoint set password is VALID
*WM: [1] your_password
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  OpenMQTTGateway
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] WiFi Scan ASYNC completed in 4019 ms
*WM: [2] WiFi Scan ASYNC found: 27
*WM: [2] Portal Timeout In 100 seconds
*WM: [2] Portal Timeout In 70 seconds
```

patrul, 2019-07-20:
I´m sorry. It doesn´t work.
Stops after some minutes with mqtt timeout error

1technophile, 2019-07-20:
The issue seems to come from this reboot that occurs when publishing, could you post all the build traces please and the complete serial monitor extract until the reboot?

patrul, 2019-07-20:
I enable the log option I know for mqtt broker, but no information about disconnection error

`log_dest stdout
log_type error
log_type warning
log_type notice
log_type information`

Serial monitor:

`--- Available ports:
---  1: COM3                 u'TUSB3410 Device (COM3)'
---  2: COM6                 u'Silicon Labs CP210x USB to UART Bridge (COM6)'
--- Enter port index or full name: 2
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
s Jun  8 2016mounting FS...
E (36) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:52] begin(): Mounting SPIFFS failed! Error: -1
failed to mount FS
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port  
*WM: [2] Added Parameter: user  
*WM: [2] Added Parameter: pass  
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled  
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect        
*WM: [1] Connecting to saved AP: MiFibra-4770
*WM: [3] Using Password: *******
*WM: [3] WiFi station enable
*WM: [1] connectTimeout not set, ESP waitForConnectResult...
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.182
connected...yeey :)
saving config
[E][vfs_api.cpp:22] open(): File system is not mounted
"mqtt_server":"192.168.1.11","mqtt_port":"1883","mqtt_user":"mqtt-user","mqtt_pass":"******"}*WM: [3] freeing allocated params! 
*WM: [3] unloading
OpenMQTTGateway mac: 
A4:CF:12:75:B3:D8
OpenMQTTGateway ip:
192.168.1.182
1883
Connecting to MQTT by IP adress
192.168.1.11
BLEinterval btw scans
55555
ZgatewayBT multicore ESP32 setup done
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
BT Task running on core 0
MQTT connection...
Connected to broker
Subscription OK to the subjects
Creating BLE buffer
device detected
434071A908BE
BLErssi
-75
txPower
-59
BLE DISTANCE :
5.83
[... 459 lines cut ...]
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/F877B8625F91
{"id":"f8:77:b8:62:5f:91","manufacturerdata":"u","rssi":-95,"distance":35.511}
Creating BLE buffer
device detected
F0B5D1838682
BLErssi
-88
txPower
-59
BLE DISTANCE :
19.73
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-88,"distance":19.7325}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/F0B5D1838682
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-88,"distance":19.7325}`

Serial monitor stopped writing logs, but broker said client was disconnected 

`1563649134: New client connected from 192.168.1.182 as OpenMQTTSalon (p2, c1, k60, u'mqtt-user').
1563649238: Client OpenMQTTSalon has exceeded timeout, disconnecting.`

Very strange behaviour, I think

1technophile, 2019-07-20:
I have made a modification to the development branch, the BLEscans were called too often.
Could you update your repository and try again?

patrul, 2019-07-20:
There was a typo on main.ino #892 changed to `       #ifndef ESP32`

More information on logs with new modifications:

```
Windows PowerShell
Copyright (C) Microsoft Corporation. Todos los derechos reservados.

Prueba la nueva tecnología PowerShell multiplataforma https://aka.ms/pscore6

PS C:\Users\ricar\Documents\Git\OpenMQTTGateway> pio device monitor -p COM6
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Creating BLE buffer
device detected    
84C0EFD67C10       
BLErssi
-94
txPower
-59
BLE DISTANCE :
32.74
Pub json into:
esp32/home_presence/OpenMQTTSalon       
{"id":"84:c0:ef:d6:7c:10","manufacturerdata":"u","rssi":-94,"distance":32.73764}
[E][WiFiClient.cpp:365] write(): fail on fd 55, errno: 113, "Software caused connection abort"
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/84C0EFD67C10
{"id":"84:c0:ef:d6:7c:10","manufacturerdata":"u","rssi":-94,"distance":32.73764}
Creating BLE buffer
device detected    
BLErssi
txPower
BLE DISTANCE :
35.51
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"6a:f1:32:ab:05:d8","manufacturerdata":"L","rssi":-95,"distance":35.511}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/6AF132AB05D8
{"id":"6a:f1:32:ab:05:d8","manufacturerdata":"L","rssi":-95,"distance":35.511}
Creating BLE buffer
device detected
3CADBE17F67B
BLErssi
-86
txPower
-59
BLE DISTANCE :
16.55
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"3c:ad:be:17:f6:7b","manufacturerdata":"L","rssi":-86,"distance":16.54559}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/3CADBE17F67B
{"id":"3c:ad:be:17:f6:7b","manufacturerdata":"L","rssi":-86,"distance":16.54559}

--- exit ---
PS C:\Users\ricar\Documents\Git\OpenMQTTGateway> pio device monitor -p COM6
could not open port u'COM6': could not open port u'COM6': WindowsError(5, 'Acceso denegado.')
PS C:\Users\ricar\Documents\Git\OpenMQTTGateway> pio device monitor -p COM6
could not open port u'COM6': could not open port u'COM6': WindowsError(5, 'Acceso denegado.')
PS C:\Users\ricar\Documents\Git\OpenMQTTGateway> pio device monitor -p COM6
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
emounting FS...
E (37) SPIFFS: mount failed, -10025
[E][SPIFFS.cpp:52] begin(): Mounting SPIFFS failed! Error: -1
failed to mount FS
*WM: [3] allocating params bytes: 20
[... 654 lines cut ...]
[E][WiFiClient.cpp:365] write(): fail on fd 59, errno: 113, "Software caused connection abort"
Creating BLE buffer
device detected
F0B5D1838682
BLErssi
-92
txPower
-59
BLE DISTANCE :
27.75
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-92,"distance":27.75274}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/F0B5D1838682
{"id":"f0:b5:d1:83:86:82","manufacturerdata":"","rssi":-92,"distance":27.75274}
BT Task running on core 0
failure_number
4
failed, rc=
-4
try again in 5s
MQTT connection...
[E][WiFiClient.cpp:365] write(): fail on fd 60, errno: 113, "Software caused connection abort"
```

1technophile, 2019-07-21:
does it reconnects automaticaly now?

patrul, 2019-07-21:
No, actually it is not connecting after flashing again. It is on a loop trying something:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
mounting FS...
mounted file system
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [1] AutoConnect
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client...
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED
*WM: [2] AccessPoint set password is VALID
*WM: [1] your_password
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  OpenMQTTGateway
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] WiFi Scan ASYNC completed in 3619 ms
*WM: [2] WiFi Scan ASYNC found: 23
*WM: [2] Portal Timeout In 90 seconds
*WM: [2] Portal Timeout In 60 seconds
*WM: [2] Portal Timeout In 30 seconds
*WM: [2] Portal Timeout In 0 seconds
*WM: [1] config portal has timed out 
*WM: [3] configportal abort
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_DISCONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
failed to connect and hit timeout
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
mounting FS...
mounted file system
```

1technophile, 2019-07-21:
try maybe without wifi manager by setting your wifi ssid and password into user_config.h:
* uncomment `//#define ESPWifiManualSetup true`
* set your wifi credentials and mqtt parameters

patrul, 2019-07-21:
It connects when stars but after a minute losses connection and it is not able to reconnect

```
Windows PowerShell
Copyright (C) Microsoft Corporation. Todos los derechos reservados.

Prueba la nueva tecnología PowerShell multiplataforma https://aka.ms/pscore6

PS C:\Users\ricar\Documents\Git\OpenMQTT> pio device monitor -p COM6
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Connecting to 
MiFibra-4770  
.
.
.
.
.
.
WiFi ok with manual config credentials
OpenMQTTGateway mac: 
A4:CF:12:75:B3:D8
OpenMQTTGateway ip:
192.168.1.182
1883
Connecting to MQTT by IP adress
192.168.1.11
BLEinterval btw scans
55555
ZgatewayBT multicore ESP32 setup done
MQTT_MAX_PACKET_SIZE
1024
Setup OpenMQTTGateway end
BT Task running on core 0
MQTT connection...
Connected to broker
Subscription OK to the subjects
Creating BLE buffer
device detected
28EBB613F586
BLErssi
-46
txPower
-59
BLE DISTANCE :
0.08
Pub json into:
esp32/home_presence/OpenMQTTSalon
{"id":"28:eb:b6:13:f5:86","manufacturerdata":"␆","rssi":-46,"distance":0.082996}
Pub json into:
esp32/OpenMQTTSalon/BTtoMQTT/28EBB613F586
{"id":"28:eb:b6:13:f5:86","manufacturerdata":"␆","rssi":-46,"distance":0.082996}

...

BT Task running on core 0
MQTT connection...
[E][WiFiClient.cpp:365] write(): fail on fd 56, errno: 113, "Software caused connection abort"
Creating BLE buffer
device detected
695F2AE1B06D

...

BT Task running on core 0
failure_number
1
failed, rc=
-4
try again in 5s
MQTT connection...
[E][WiFiClient.cpp:365] write(): fail on fd 57, errno: 113, "Software caused connection abort"
failure_number
2
failed, rc=
-4
try again in 5s
MQTT connection...

...

[E][WiFiClient.cpp:365] write(): fail on fd 58, errno: 113, "Software caused connection abort"
BT Task running on core 0
```

patrul, 2019-07-22:
The ESP32 losses conection and I am not able to ping or see DHCP lease on router

There is something producing the error on software side to fail on the connection (socket, timeout ...)

patrul, 2019-07-24:
Thanks Florian for your support. 

I´ve moved to ESPHome. By now it is enough to cover my needs. It is extremelly simple to have MiFlora and Xiaomi MiJia BLE Sensor data. 

I´ll come back for upcoming advanced setups

closed 2019-07-24 by patrul