1technophile/OpenMQTTGateway #430: Ability to send 433 mhz RF signal continuously (for pairing)

saikek, 2019-07-25:
**Problem:** Many of 433mhz dongles (you probably have one for car alarm, garage doors, light related) are programmable. 
In order to program them, you need to send continuous signal. (Press and hold button on the host remote). MqttTo433mhz sending routine is recognized by programmable dongle, but it is not enough for storing codes.

**Solution:** It would be great if MqttTo433 module would have a argument that would allow to send same value / signal for a period of time.

**Describe alternatives you've considered**
I was not able to find any at the moment.

**Additional context**
I'm using 433Mhz dongles with Home assistant to control lights in my flat. 
The latest dongle that I've received sends same value on all buttons.

mpember, 2019-07-29:
You should probably have a look at the **repeat** parameter.

saikek, 2019-07-29:
How do I chain multiple commands at one, is that `,` or html-style `&` ? 
I'm using mqtt-spy with ui, where I only specify topic and message.

mpember, 2019-07-29:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-RF-09

By sending a message that contains JSON-formatted content, you can define a number of parameters.

1technophile, 2019-08-06:
I close the issue, if you need more help don't hesitate to go to the community forum.

closed 2019-08-06 by 1technophile