1technophile/OpenMQTTGateway #432: Is it possible create some little rules?

itProfi, 2019-07-29:
Helllo, Florian! Is it possible to create any Rules engine inside gateway? For examle, it recieve data from Xiaomi Mijia Temperature and inside firmware (for example through mqtt, like ON_OFF Actuator) and if temp>x turn pin ON, if temp<x turn pin OFF)..
On the basis of this could be to create a stand-alone thermostat without any IoT systems...!!

1technophile, 2019-07-29:
Hello, not right now, but I'm interested in integrating this kind of function to OMG.
It could cover use cases also like triggering a siren when PIR sensor detect a presence.

itProfi, 2019-07-30:
Ok, we are wait it! :)

Legion2, 2019-08-04:
@itProfi this is simple, use your favorite programming language and connect ot the MQTT Broker. Than you can subscribe to a topic and process messages and send new messages on topics.
The MQTT Broker is the integration point of the system.

itProfi, 2019-08-05:
@Legion2 Thanks! I know that. I want have 5-7 stand-alone thermostat without any IoT systems...!!

Legion2, 2019-08-05:
Then you don't need OMG with a MQTT broker. You only need libraries to read the temperature and then make the simple if-then-else in Arduino code.

akasma74, 2019-08-05:
there is also esphome that can be used with or without HA, comes with a lot of flexible configuration options and extensibility.

itProfi, 2019-08-05:
Ok, i try...

mpember, 2019-08-07:
ESPEasy may be better suited to your needs.
https://www.letscontrolit.com/wiki/index.php?title=Tutorial_Rules

itProfi, 2019-08-08:
@mpember Thanks, but ESPEasy can't Xiaomi Mijia Temperature (no Bluetooth sensor)..(

closed 2020-07-18 by itProfi