1technophile/OpenMQTTGateway #434: Sonoff RF bridge (v2) and Kaku

patvdleer, 2019-07-30:
Is it possible to receive KaKu on the Sonoff RF Bridge (v2)? I tried to enable `ZgatewayRF2` but this doesn't seem to do much

I'm currently  testing with the APC3-1500R and APC3-2300R but both don't seem to do much...

User config:
```c
#define OMG_VERSION "0.9.1"

/*-------------DEFINE YOUR  NETWORK PARAMETERS BELOW----------------*/
#if defined(ESP8266)  // for nodemcu, weemos and esp8266
  #define ESPWifiManualSetup true //uncomment you don't want to use wifimanager for your credential settings on ESP
#endif

#if defined(ESP32) || defined(ESPWifiManualSetup) // for nodemcu, weemos and esp8266
  #define wifi_ssid "0x43"
  #define wifi_password "xxx"
#else // for arduino + W5100
  const byte mac[] = {  0xDE, 0xED, 0xBA, 0xFE, 0x54, 0x95 }; //W5100 ethernet shield mac adress
#endif

#define WifiManager_password "xxx" //this is going to be the WPA2-PSK password for the initial setup access point 
#define WifiManager_ssid "OpenMQTTGateway" //this is the network name of the initial setup access point
#define WifiManager_ConfigPortalTimeOut 120

/*-------------DEFINE YOUR MQTT PARAMETERS BELOW----------------*/
//MQTT Parameters definition
//#define mqtt_server_name "xxx" // instead of defining the server by its IP you can define it by its name, uncomment this line and set the correct MQTT server host name
char mqtt_user[20] = "OpenMQTTGateway"; // not compulsory only if your broker needs authentication
char mqtt_pass[30] = "xxx"; // not compulsory only if your broker needs authentication
char mqtt_server[40] = "192.168.178.38";
char mqtt_port[6] = "1883";

#define Gateway_Name "OpenMQTTGateway"

//uncomment the line below to integrate msg value into the subject when receiving
//#define valueAsASubject true

/*-------------DEFINE THE MODULES YOU WANT BELOW----------------*/
//Addons and module management, comment the Z line

//#define ZgatewayRF     "RF"       //ESP8266, Arduino, ESP32
//#define ZgatewayRF315  "RF315"    //ESP8266, Arduino, ESP32
//#define ZgatewayIR     "IR"       //ESP8266, Arduino,         Sonoff RF Bridge
//#define ZgatewayLORA   "LORA"       //ESP8266, Arduino, ESP32
//#define ZgatewayPilight "Pilight" //ESP8266, Arduino, ESP32
//#define ZgatewayBT     "BT"       //ESP8266, Arduino, ESP32
//#define ZgatewayRF2    "RF2"      //ESP8266, Arduino, ESP32
#define ZgatewaySRFB   "SRFB"     //                          Sonoff RF Bridge
//#define Zgateway2G     "2G"       //ESP8266, Arduino, ESP32
//#define ZgatewayRFM69  "RFM69"    //ESP8266, Arduino, ESP32
//#define ZactuatorONOFF "ONOFF"    //ESP8266, Arduino, ESP32,  Sonoff RF Bridge
//#define ZsensorINA226  "INA226"   //ESP8266, Arduino, ESP32
//#define ZsensorHCSR501 "HCSR501"  //ESP8266, Arduino, ESP32,  Sonoff RF Bridge
//#define ZsensorADC     "ADC"      //ESP8266, Arduino, ESP32
//#define ZsensorBH1750  "BH1750"   //ESP8266, Arduino, ESP32
//#define ZsensorTSL2561 "TSL2561"  //ESP8266, Arduino, ESP32
//#define ZsensorBME280  "BME280"   //ESP8266, Arduino, ESP32
//#define ZsensorDHT     "DHT"      //ESP8266, Arduino, ESP32,  Sonoff RF Bridge
//#define ZsensorGPIOKeyCode "GPIOKeyCode" //ESP8266, Arduino, ESP32
//#define ZsensorGPIOInput "GPIOInput" //ESP8266, Arduino, ESP32
//#define ZmqttDiscovery "HADiscovery"//ESP8266, Arduino, ESP32, Sonoff RF Bridge

/*-------------DEFINE YOUR ADVANCED NETWORK PARAMETERS BELOW----------------*/
//#define MDNS_SD //uncomment if you  want to use mdns for discovering automatically your ip server, please note that MDNS with ESP32 can cause the BLE to not work
#define maxMQTTretry 4 //maximum MQTT connection attempts before going to wifi setup

//set minimum quality of signal so it ignores AP's under that quality
#define MinimumWifiSignalQuality 8

// these values are only used if no dhcp configuration is available
const byte ip[] = { 192, 168, 178, 240 }; //ip adress
const byte gateway[] = { 0, 0, 0, 0 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte Dns[] = { 0, 0, 0, 0 }; //ip adress, if first value is different from 0 advanced config network will be used and you should fill gateway & dns
const byte subnet[] = { 255, 255, 255, 0 }; //ip adress

/*-------------DEFINE YOUR MQTT ADVANCED PARAMETERS BELOW----------------*/
#define Base_Topic "/home/"
#define version_Topic  Base_Topic Gateway_Name "/version"
#define will_Topic  Base_Topic Gateway_Name "/LWT"
#define will_QoS 0
#define will_Retain true
#define will_Message "offline"
#define Gateway_AnnouncementMsg "online"

#define jsonPublishing true //comment if you don't want to use Json  publishing  (one topic for all the parameters)
//example home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4 {"rssi":-63,"servicedata":"fe0000000000000000000000000000000000000000"}
#define jsonReceiving true //comment if you don't want to use Json  reception analysis

//#define simplePublishing true //comment if you don't want to use simple publishing (one topic for one parameter)
//example 
// home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4/rssi -63.0
// home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4/servicedata fe0000000000000000000000000000000000000000
#define simpleReceiving true //comment if you don't want to use old way reception analysis

/*-------------DEFINE YOUR OTA PARAMETERS BELOW----------------*/
#define ota_hostname Gateway_Name "-1"
#define ota_password "xxx"
#define ota_port 8266

/*-------------DEFINE PINs FOR STATUS LEDs----------------*/
#ifdef ESP8266
  #define led_receive 40
  #define led_send 42
  #define led_info 44
#elif ESP32
  #define led_receive 40
  #define led_send 42
  #define led_info 44
#elif __AVR_ATmega2560__ //arduino mega
  #define led_receive 40
  #define led_send 42
  #define led_info 44
#else //arduino uno/nano
  #define led_receive 40
  #define led_send 42
  #define led_info 44
#endif

//      VCC   ------------D|-----------/\/\/\/\ -----------------  Arduino PIN
//                        LED       Resistor 270-510R

/*----------------------------OTHER PARAMETERS-----------------------------*/
/*-------------------CHANGING THEM IS NOT COMPULSORY-----------------------*/
/*----------------------------USER PARAMETERS-----------------------------*/
#ifdef ZgatewaySRFB
  #define SERIAL_BAUD 19200
#else
  #define SERIAL_BAUD 115200
#endif
/*--------------MQTT general topics-----------------*/
// global MQTT subject listened by the gateway to execute commands (send RF, IR or others)
#define subjectMQTTtoX  Base_Topic Gateway_Name "/commands/#"
#define subjectMultiGTWKey "toMQTT"
#define subjectGTWSendKey "MQTTto"

//variables to avoid duplicates
#define time_avoid_duplicate 3000 // if you want to avoid duplicate mqtt message received set this to > 0, the value is the time in milliseconds during which we don't publish duplicates

//uncomment to use multicore function of ESP32 for BLE
#ifdef ESP32
  #define multiCore //comment to don't use multicore function of ESP32 for BLE
#endif

#if defined(ESP8266) || defined(ESP32) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)
  #define JSON_MSG_BUFFER 512 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
#else // boards with smaller memory
  #define JSON_MSG_BUFFER 64 // Json message max buffer size, don't put 1024 or higher it is causing unexpected behaviour on ESP8266
#endif

#define TimeBetweenReadingSYS 120000 // time between system readings (like memory)
#define subjectSYStoMQTT  Base_Topic Gateway_Name "/SYStoMQTT"

#define subjectTRACEtoMQTT Base_Topic Gateway_Name "/log" //uncomment if you want to see traces on /log topic

/*-------------------ACTIVATE TRACES----------------------*/
#define TRACE 1  // commented =  trace off, uncommented = trace on
```

1technophile, 2019-07-30:
Hello,

Yes it is possible if you apply the rfbridge direct mod and load the firmware with ZgatewayRF2.
Nevertheless as a first approach I will verify with a nodemcu/arduino, an emitter/receiver and OMG loaded if it works with your device. And if OK you could [modify your RF Bridge](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack).

1technophile, 2019-07-30:
I close the issue, if you want to discuss about that the community is more appropriate

closed 2019-07-30 by 1technophile