1technophile/OpenMQTTGateway #435: Upgrade from 0.5 to 0.9.1 - Door sensor codes not reliably reporting to Home Assistant

Kanga-Who, 2019-07-31:
**Describe the bug**
After updating from 0.5 to 0.9.1, I am having issues with many of my 2 code door sensors not reporting the open/closed state.

It does not seem to be time-related, as I can close a door, leave it for 1 min, open, and no state change occurs in Home Assistant.

I can see the codes are being sent using MQTT Lens.

Single code motion sensors are working perfectly. 2 different brands of door sensors have been tried, the same result. I have also tried using the repeat function, this has not helped

**Expected behavior**
OMG to reliably report the state to HA

**Environment (please complete the following information):**
* OpenMQTTGateway version used 0.9.1
* Library version related to the problem you have - All libraries used are the ones provided in the repo

Below is the code I am using for Binary Sensor in HA. 

```
- platform: mqtt
  name: "Front Door"
  qos: 0
  state_topic: "home/OpenMQTTGateway2home/433toMQTT"
  value_template: '{{ value_json.value }}'
  payload_on: "133651"
  payload_off: "133657"
  device_class: opening
```

akasma74, 2019-08-01:
> After updating from 0.5 to 0.9.1, I am having issues with many of my 2 code door sensors not reporting the open/closed state.

it's a HUGE jump, isn't it?

> It does not seem to be time-related, as I can close a door, leave it for 1 min, open, and no state change occurs in Home Assistant.

I notice something similar - have a Pilight @ Sonoff RF Bridge on the ground floor and RF @ Wemos D1 mini on the 1st, and open signals from my Kerui 026 door contact on the first floor are almost always received by Pilight, but closed signals occasionally aren't. Don't know if that's because of battery level or distance (but then why it's reliable with open?), atm I just added code to receive RF codes as well and publish state to appropriate MQTT topic so it works much more reliable now (but I have plans to look into it when I have time as I don't like such a workaround).

> I can see the codes are being sent using MQTT Lens.

what codes?

Kanga-Who, 2019-08-01:
>  what codes?

The RF codes from the door sensor.

It seems that OMG is not passing the codes through to HA for some reason.

akasma74, 2019-08-01:
Then I'd suggest to connect your OMG device to Arduino editor, open Serial monitor and post here what's there when you notice anything like that as it might give the information needed for debugging.

closed 2019-10-07 by stale[bot]