1technophile/OpenMQTTGateway #441: Outdated wiki pages

Legion2, 2019-08-07:
Some wiki pages contain old information from previous versions of OMG, that are now no more valid. This includes references to moved, deleted or modified code. Some pages include broken links that resolve to 404 Not Found and some step-by-step explanations don't work with the current OMG version.
Here a list of some pages where the information is not consistent with the current status of OMG. If you find more such outdated information please post them in the comments:
- [ ] [Arduino I2C-Bus Modules #Software setup](https://github.com/1technophile/OpenMQTTGateway/wiki/Arduino-I2C-Modules#software-setup)
- [ ] [Module Overview ESP8266 #Software setup](https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP8266#software-setup)
- [ ] [Module Overview ESP32 #Software setup](https://github.com/1technophile/OpenMQTTGateway/wiki/Module-Overview-ESP32#software-setup)
- [ ] [Decreasing Gateway SRAM usage for arduino AVR boards](https://github.com/1technophile/OpenMQTTGateway/wiki/Decreasing-Gateway-SRAM-usage-for-arduino-AVR-boards) OpenMQTTGateway.ino was renamed
- [ ] [PIR Modules](https://github.com/1technophile/OpenMQTTGateway/wiki/PIR-Modules) broken links
- [ ] [User guide json](https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-json) code is outdated
- [ ] [User guide ADC 09](https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-ADC-09) broken links

valdiks, 2019-08-13:
Broken link to user_config.h:
https://github.com/1technophile/OpenMQTTGateway/wiki/GSM-GPRS---2G-gateway
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-RFM69-Module
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-IR-Send-and-Receive
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-I2C-Modules
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-BT-Module
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-DHT-Modules
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP32-RF-Send-and-Receive
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP32-IR-Send-and-Receive
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP32-BT-Module
https://github.com/1technophile/OpenMQTTGateway/wiki/ESP32-LORA-Module
https://github.com/1technophile/OpenMQTTGateway/wiki/Sonoff-RFB

1technophile, 2019-08-18:
Done, if someone wants to do a review don't hesitate.

Legion2, 2019-08-18:
You only removed the setup parts?

1technophile, 2019-08-18:
For wifi and mqtt it is on a [single ](https://github.com/1technophile/OpenMQTTGateway/wiki/Wifi-and-mqtt-connection) location.

For platformio it is [here ](https://github.com/1technophile/OpenMQTTGateway/wiki/Platformio-ini-configuration)

These 2 links are called from the [How to start](https://github.com/1technophile/OpenMQTTGateway/wiki/HowToStart)

closed 2019-08-18 by Legion2

1technophile, 2019-09-23:
I have made a complete rework of the doc below:
https://1technophile.github.io/OpenMQTTGateway

There can be some improvments or comments to add. If you find some don't hesitate to submit a PR to the gh-pages branch.