1technophile/OpenMQTTGateway #445: Lora crc option in User_config.h

svwhisper, 2019-08-27:
The new Lora support is great to have.  I really appreciate your implementing that feature.  Now that I am deploying sensor nodes, I find there are a large number of rubbish messages received by the gateway.  Experimenting with various options led me to try enabling crc on the gateway and sensor nodes.  Problem solved.

Would it be possible to add a feature to enabling crc in User_config.h?  I realise it can now be enabled by sending an command via mqtt, but I think better to code it in the config file as default.  I would do this and offer it to you, but I'm not yet competent with GitHub.

1technophile, 2019-08-27:
Hello,

Great to have some feedback about this new gateway ! 
For my curiosity could you explain why setting it up by mqtt is not sufficient ?

svwhisper, 2019-08-30:
If I set to up by MQTT, that means I need to have logic elsewhere, e.g. node red, to configure OMG before it is able to receive data from my sensor nodes.  It seems simpler to have the OMG default config in the config file rather than elsewhere.  It wouldn't be at all complex to do it in node red, of course, and I appreciate this wouldn't be a high priority change.  

My first use of OMG Lora is a bit silly.  I have a remote home-built solar-powered street light at the entrance to our farm, about 900 metres from the gateway.  I want to monitor voltage and light brightness (controlled by arduino based on battery state of charge) to see whether I've got sufficient solar and battery capacity.

svwhisper, 2019-08-30:
The right config file would be config_LORA.h, of course.  Not User_config.h.

1technophile, 2019-08-31:
Is CRC the only interesting option from the Lora API that we need to set by default on config_LORA.h or there is others?

svwhisper, 2019-09-03:
That's the only option I've found that's useful and missing from the config file.

My prior use of Lora on Arduino was using the Radiohead library to provide reliable packet delivery.  I miss the Radiohead concept of an "ack" and automatic re-transmission to ensure delivery.  However, the approach you've chosen with Mistry's Lora library is simpler and easier for people to quickly get working.  The downside is that packets don't always arrive and that needs to be handled by the user.

1technophile, 2019-09-20:
done

closed 2019-09-20 by 1technophile