1technophile/OpenMQTTGateway #446: openmqtt mijia temp/humid sensor

praepoke, 2019-08-27:
Hello,

I've been using OMG 0.8.1 for a couple of month now to get the data from a mijia temp/humid sensor. It's been working great until for some reason i had to move the gateway thus rebooting it.

The gateway seems to be working fine, i get some data from the sensors but not the temperature/humidity/battery. i have 5 of them and none report data but they do work if i poll using an app on my phone.

here is what i get when connected to serial 
BLE DISTANCE :
12.61
Pub json into:
home/bt/home_presence/OpenMQTTGateway
{"id":"58:2d:34:34:4d:f9","name":"MJ_HT_V1","rssi":-83,"distance":12.61001}
Get service data
Pub json into:
home/bt/OpenMQTTGateway/BTtoMQTT/582D34344DF9
{"id":"58:2d:34:34:4d:f9","name":"MJ_HT_V1","rssi":-83,"distance":12.61001,"servicedata":"caab91251972","servicedatauuid":"0000ffff-0000-1000-8000-00805f9b34fb"}

and in mqtt : 
home/bt/home_presence/OpenMQTTGateway {"id":"58:2d:34:34:4d:f9","name":"MJ_HT_V1","rssi":-83,"distance":12.61001}
home/bt/OpenMQTTGateway/BTtoMQTT/582D34344DF9 {"id":"58:2d:34:34:4d:f9","name":"MJ_HT_V1","rssi":-83,"distance":12.61001,"servicedata":"caab91251972","servicedatauuid":"0000ffff-0000-1000-8000-00805f9b34fb"}

anything i'm missing ?

thanks

1technophile, 2019-08-27:
Hello,

Strange, by the time did you updated the sensor firmware ? 
Did you try to disconnect and reconnect the gateway ?

praepoke, 2019-08-28:
hi,
no i did not update anything (gateway or sensors) none of the 5 send their data other than what i pasted.
and yes i've tried to restart the gateway multiple times, tried to restart the sensors as well (remove the battery)

nothing helped :/

1technophile, 2019-08-28:
Hmm curious. Did you try with another esp32 and the last version ?

praepoke, 2019-08-30:
not yet. i've ordered another esp32 and will try to flash the one i have.
will get back to you

praepoke, 2019-09-01:
hi, well i tried everything i could understand, not a big dev and platformio took a toll on me.
anyways, tried flashing 0.9.1 to another esp and didn't have muche more luck
tried 0.9.2 and no luck either. the sensors are not even reported to mqtt

1technophile, 2019-09-01:
Is your phone connected to them?

praepoke, 2019-09-01:
nop

molda, 2019-09-03:
I seem to have the same problem here. I have 2 sensor which have been working for a couple of month without an issue and 2 days ago they both stopped reporting temperature/humidity/battery, all i get is the service data.
I will check the version of OMG and try to restart both sensors as well as gateway and report back.

praepoke, 2019-09-03:
finally received my new esp32s. and flashed 0.9.2 on it. Seems to work from time to time. it doesn't catch as much points as before.
one more question, i can see it sees my mi band 3. anyway it can catch heartbeat/steps ?
thanks

1technophile, 2019-09-03:
> Seems to work from time to time. it doesn't catch as much points as before.
Did you had the same ble interval in config_BT.h

>anyway it can catch heartbeat/steps ?
Do you see a service data field for your mi band ?

praepoke, 2019-09-03:
servicedata a21e0000
servicedatauuid 0000fee0-0000-1000-8000-00805f9b34fb

1technophile, 2019-09-19:
1e = 30 does this correspond to a real value on your mi band?

praepoke, 2019-09-28:
So far as i can tell, the first 4 digits gives you step counter 
a21e ->1ea2 = 7842 steps.

1technophile, 2019-10-24:
Hello,

I have integrated miband into this branch,
would it be possible to test it?
https://github.com/1technophile/OpenMQTTGateway/tree/miband-steps

1technophile, 2019-11-02:
#478

closed 2019-11-02 by 1technophile