1technophile/OpenMQTTGateway #447: Ultrasonic distance sensor

mpember, 2019-08-28:
**Is your feature request related to a problem? Please describe.**
I am looking to add support for an ultrasonic distance sensor.

**Describe the solution you'd like**
I am having a go at writing a module for it.

**Describe alternatives you've considered**
Would rather not use one of the dedicated 'garage' projects.

**Additional context**
While I wait for the hardware to arrive, I was wondering if anyone sees any obvious flaws in the code I hacked together (from existing OMG modules and HC-SR04 example code).

https://github.com/mpember/OpenMQTTGateway/commit/5e62c2986f6953d5ef2df57eaacdc34ec6878f7b

1technophile, 2019-08-27:
Maybe you could submit the PR so as to discuss around your code proposal ?

mpember, 2019-08-28:
Since I only use wemos/esp8266 boards, I am hoping someone can offer guidance on which pins are best used on other boards.

mpember, 2019-08-28:
@1technophile 
I was going to hold back creating a PR until I had a chance to test out the code.  I just figured I would see if anyone has any comments about the code in the meantime.  The main area I am wondering about is the pin selection.  I want to avoid stepping on toes where others have connected a different sensor.

mpember, 2019-08-28:
I used that list to pick D5 and D6 for the wemos.  I figured it would have the least impact if I 'stole' the pins from the RF module I don't use.  That left room for the RF module I do have, a DHT sensor, IR and the new distance sensor.

This means I don't need to make major changes between the code for my two gateway setups.  I just turn on/off the modules I need with platformio tasks and compile both from the same codebase.

1technophile, 2019-08-28:
It makes senses did you take a look to the different proposed pinouts here : 
https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit?usp=drivesdk

1technophile, 2019-09-11:
Pr done

closed 2019-09-11 by 1technophile