1technophile/OpenMQTTGateway #449: MiFlora

eljuanchete, 2019-08-30:
Hello,
One of my MiFlora is not being recognized as such. BLE data is scanned and reported to MQTT but device data (temperature, humidity, etc.) is not reported to MQTT daemon. Reported data is id, name, rssi, distance, servicedata and servicedatauuid
I have been checking the code ZgatewayBT.ino and I think that the problem is because this device servicedata value is different. In all of my others devices servicedata begins with 71209800. However the device not recognized servicedata begins with 31029800
In this domoticz forum thread there are references to other device with a similar servicedata 
https://www.domoticz.com/forum/viewtopic.php?t=13306&start=200
This is the first MiFlora I purchased almost two years ago. Maybe this is an older model and it is not recognized?
Thanks in advanced

1technophile, 2019-08-31:
Hello,

Did you try to update the firmware of this mi flora with the xiaomi app?
Maybe it could come from an old version of the mi flora firmware
Regards

eljuanchete, 2019-08-31:
Yes, i did update the firmware (didn't do it sin I bough it) and the app said FW update was fine. I double checked it again and turns out it didn't update firmware. 
In the end the problem is the battery level. The battery level was low, enough to retrieve data from the sensor, but not enough to upgrade firmware. Once I replaced the battery FW update was successful, sensor servicedata begins with 71209800 and all values can be read.
Sorry to bother you with this problem.
You can close the issue.
Thank you very much

1technophile, 2019-08-31:
No problem, it will enable others to know why their mi flora values are not read if they fall on this issue.

closed 2019-08-31 by 1technophile