1technophile/OpenMQTTGateway #450: RF AND Pilight together

wimpie007, 2019-08-31:
as it seems impossible to use the RF (gatewayRF) and Pilight (gatewayPilight) togther on 1 device, any chance that it would work in the future?
(combining those 2 libraries would be great... i now have devices that work with RF, but not with pilight, and the other way around)
Thanks for openmqqtGW!

odis68, 2019-11-08:
Have an idea, maybe not a new one. RF and on different Pilight receivers. Use two independent receivers on different GPIO channels. please consider this opportunity.

odis68, 2019-12-02:
Please comment on my previous post !!!

1technophile, 2019-12-02:
> Have an idea, maybe not a new one. RF and on different Pilight receivers. Use two independent receivers on different GPIO channels. please consider this opportunity.

Why not, it should be tested to confirm

1technophile, 2019-12-02:
> As I pointed out above, it's a slow process because requires plenty of time and knowledge (and because there is no way to simply "transfer" support for devices from RF to Pilight) and currently it mainly relies on the author of OMG.
Therefore it is crucial to the project's success to have a) more people willing to donate their time and b) clear instructions how to add support for new devices to OMG Pilight.
I'm happy to do my bit as still have some 433MHz devices that are supported by OMG RF only.

Indeed adding a protocol to Pilight was a long process, I'm going to see also as an alternative if I can add easily protocols to rcswitch, if yes I will document it.

odis68, 2020-02-03:
The problem is not outdated !!! We are patiently waiting !!!

1technophile, 2020-02-03:
> The problem is not outdated !!! We are patiently waiting !!!

Thanks for your patience, I keep that in mind

cyraxxx, 2020-12-09:
Hi, did someone actually test @odis68 's idea? If not, I can try it (adding 2 receivers and test on an ESP32), but I am not a programmer. Now my setup is using Pilight. If I modify config_RF.h (adding RF_RECEIVER_GPIO2) and ZgatewayRF.ino to use RF_RECEIVER_GPIO2 then enable ZgatewayRF and ZgatewayPilight, is it worth a shot or do I need to do much more then this?

iWildStyle, 2021-03-06:
Faced a problem - RF and Pilight do not work at the same time. They do not work, since the libraries use attachInterrupt to the same pin, which cannot be done.
I did the following - instead of installing a second receiver, I divided the date line from the receiver into two input pins D8 and D7
Then I split the config_RF.h configuration file into two (config_Pilight.h) and in the second I brought out all the settings for Pilight.
Also changed the settings a bit in ZgatewayPilight.ino and main.ino and voila:
```
N: Subject: / PilighttoMQTT
N: Received json: {"message": {"id": 86, "channel": 1, "battery": 1, "temperature": 18.0, "humidity": 41}, "protocol": "nexus", "length": "86", "value": "86", "repeats": 2, "status": 2}
N: Subject: / 433toMQTT
N: Received json: {"value": 15145729, "protocol": 1, "length": 24, "delay": 391}
```
the first object is processed by Pilight, the second is processed by RF
PS: I have Wemos D1 mini module and cheap RX470B 433 receiver

melyux, 2021-11-29:
@iWildStyle Would you share all the code/settings you changed to get this to work? Specifically ZgatewayPilight.ino and main.ino?

melyux, 2021-11-30:
I too made modifications that let me run both Pilight and RF together on my Wemos D1 mini as well, but when I connect the 433MHz receiver's DATA pin to two separate pins on the Wemos (D7 and D8, like @iWildStyle suggested), I stop receiving data on both. As soon as I make it so the receiver's DATA pin only has contact with one pin on the Wemos, I start receiving signals again.

Software-wise, it should work, but I'm at a loss.

> I divided the data line from the receiver into two input pins D8 and D7

How exactly did you do this?

melyux, 2021-12-01:
I got it to work! This is incredible! Both Pilight and RF (RCSwitch) working at the same time. For some reason pins D3, D4, and (maybe) D8 didn't work and would simply smother the data. I tried pins D6 and D5 and they both worked. So seems you can use D5, D6, and D7 for the receiver inputs at the same time, but not D3 or D4...

So in summation, I modified the code to allow the RF and Pilight modules to not be mutually exclusive (pretty trivial, including splitting the RF_RECEIVER_GPIO into multiple variables so you can assign a separate GPIO input for RF and Pilight, since like @iWildStyle said, you can't do `attachInterrupt()` to the same pin twice), and then simply connected the DATA pin of my 433MHz receiver to both D5 (or D6) and D7 of the Wemos D1 mini. And it works!

I didn't try D8 (I use it for my transmitter) or D0 or D1 (I use it for another sensor) or D2 (I use it for another sensor). But D3 and D4 definitely don't work for some reason.

I didn't try with anything other than Wemos D1 mini since I don't own anything else, so I can't test it for other microcontrollers. I can provide the code and maybe we can get a pull request going. Essentially I replaced a switch statement with multiple if statements so you can have both RF and Pilight going at the same time, split the RF GPIO variable like I mentioned above, and :TODO:

I'm going to try getting RF2 to work too, but I don't have any KAKU devices to test that with. But seems like there's some more complex thing that won't let RF and RF2 work together according to the docs (?), so I don't know if that'd work. But you can probably also run RF2 and Pilight at the same time too just like this.

Given how difficult it is to add new protocols to both RCswitch and Pilight though, I really think the future is with RTL_433 for 433MHz stuff. I need to get my hands on one of those builds.

melyux, 2021-12-02:
This is my code patch for now:

```
diff --git a/main/ZgatewayPilight.ino b/main/ZgatewayPilight.ino
index a998376..1ec491b 100644
--- a/main/ZgatewayPilight.ino
+++ b/main/ZgatewayPilight.ino
@@ -85,10 +85,10 @@ void setupPilight() {
   ELECHOUSE_cc1101.SetRx(CC1101_FREQUENCY);
 #  endif
   rf.setCallback(pilightCallback);
-  rf.initReceiver(RF_RECEIVER_GPIO);
+  rf.initReceiver(RF_PILIGHT_RECEIVER_GPIO);
   pinMode(RF_EMITTER_GPIO, OUTPUT); // Set this here, because if this is the RX pin it was reset to INPUT by Serial.end();
   Log.notice(F("RF_EMITTER_GPIO: %d " CR), RF_EMITTER_GPIO);
-  Log.notice(F("RF_RECEIVER_GPIO: %d " CR), RF_RECEIVER_GPIO);
+  Log.notice(F("RF_PILIGHT_RECEIVER_GPIO: %d " CR), RF_PILIGHT_RECEIVER_GPIO);
   Log.trace(F("ZgatewayPilight command topic: %s%s%s" CR), mqtt_topic, gateway_name, subjectMQTTtoPilight);
   Log.trace(F("ZgatewayPilight setup done " CR));
 }
@@ -224,7 +224,7 @@ extern void enablePilightReceive() {
   ELECHOUSE_cc1101.SetRx(receiveMhz); // set Receive on
 #  endif
   rf.setCallback(pilightCallback);
-  rf.initReceiver(RF_RECEIVER_GPIO);
+  rf.initReceiver(RF_PILIGHT_RECEIVER_GPIO);
   pinMode(RF_EMITTER_GPIO, OUTPUT); // Set this here, because if this is the RX pin it was reset to INPUT by Serial.end();
 };
 #endif
diff --git a/main/ZgatewayRF.ino b/main/ZgatewayRF.ino
index a4e8fae..93e638c 100644
--- a/main/ZgatewayRF.ino
+++ b/main/ZgatewayRF.ino
@@ -109,7 +109,7 @@ void RFtoMQTTdiscovery(SIGNAL_SIZE_UL_ULL MQTTvalue) {
 void setupRF() {
   //RF init parameters
   Log.notice(F("RF_EMITTER_GPIO: %d " CR), RF_EMITTER_GPIO);
-  Log.notice(F("RF_RECEIVER_GPIO: %d " CR), RF_RECEIVER_GPIO);
+  Log.notice(F("RF_RF_RECEIVER_GPIO: %d " CR), RF_RF_RECEIVER_GPIO);
 #  ifdef ZradioCC1101 //receiving with CC1101
   ELECHOUSE_cc1101.Init();
   ELECHOUSE_cc1101.SetRx(receiveMhz);
@@ -120,7 +120,7 @@ void setupRF() {
   mySwitch.enableTransmit(RF_EMITTER_GPIO);
 #  endif
   mySwitch.setRepeatTransmit(RF_EMITTER_REPEAT);
-  mySwitch.enableReceive(RF_RECEIVER_GPIO);
+  mySwitch.enableReceive(RF_RF_RECEIVER_GPIO);
   Log.trace(F("ZgatewayRF command topic: %s%s%s" CR), mqtt_topic, gateway_name, subjectMQTTtoRF);
   Log.trace(F("ZgatewayRF setup done" CR));
 }
@@ -235,7 +235,7 @@ void MQTTtoRF(char* topicOri, char* datacallback) {
 #    ifdef ZradioCC1101 // set Receive on and Transmitt off
   ELECHOUSE_cc1101.SetRx(receiveMhz);
   mySwitch.disableTransmit();
-  mySwitch.enableReceive(RF_RECEIVER_GPIO);
+  mySwitch.enableReceive(RF_RF_RECEIVER_GPIO);
 #    endif
 }
 #  endif
@@ -329,7 +329,7 @@ void enableRFReceive() {
   ELECHOUSE_cc1101.SetRx(receiveMhz);
 #  endif
   mySwitch.disableTransmit();
-  receiveInterupt = RF_RECEIVER_GPIO;
-  mySwitch.enableReceive(RF_RECEIVER_GPIO);
+  receiveInterupt = RF_RF_RECEIVER_GPIO;
+  mySwitch.enableReceive(RF_RF_RECEIVER_GPIO);
 }
 #endif
[... 151 lines cut ...]
+    enableRFReceive();
 #  endif
 #  ifdef ZgatewayRTL_433
-    case ACTIVE_RTL:
-      enableRTLreceive();
-      break;
+  if ((activeReceiver & ACTIVE_RTL) == ACTIVE_RTL)
+    enableRTLreceive();
 #  endif
 #  ifdef ZgatewayRF2
-    case ACTIVE_RF2:
-      enableRF2Receive();
-      break;
+  if ((activeReceiver & ACTIVE_RF2) == ACTIVE_RF2)
+    enableRF2Receive();
 #  endif
-#  ifndef ARDUINO_AVR_UNO // Space issues with the UNO
-    default:
-      Log.error(F("ERROR: unsupported receiver %d" CR), activeReceiver);
-#  endif
-  }
   currentReceiver = activeReceiver;
 }
 #endif
```

NorthernMan54, 2021-12-04:
Would this potentially cause resource contention from having too many interrupts contending for cpu resources?

Pro Tip - For testing of RF2/KAKU, I used a ESP32 connected to a CC1101 as a transmitter with the RF2 module enabled.  And had it transmit KAKU signals for testing the receive function.  Seemed to work pretty well.

melyux, 2022-02-23:
I noticed my code isn't complete for the "active receiver" thing despite me converting it into a bitmask; there is still code in `main.ino` that  MQTT still shows `actRec` as showing only Pilight even though both RF and Pilight are working together perfectly without missing anything. Now that I look at it, `enablePilightReceive()` calls `disableRFReceive` so I don't get how I got it working in the first place...

But it _is_ working flawlessly somehow on my ESP8266. I'll try to figure out what's going on with the software and clean up the code.  @1technophile Are you interested in getting this merged somehow?

1technophile, 2022-02-23:
@melyux yes of course !

piano-thomas, 2023-01-19:
> 

@melyux would you be able to share your ZgatewayPilight.ino file for making both Pilight and RF (RC-switch) work? I'm trying to get the same result, but since I have little experience I'm having a hard time in easily replacing my code with yours.

melyux, 2023-01-19:
@piano-thomas I'm creating a pull request to enable multi-gateway support in the next few days (hopefully). I'm updating the code to work with latest master, but the ZGatewayPilight.ino changes in the patch I posted above should still be working

melyux, 2023-01-19:
Code doesn't seem to work with RTL433 (I think the CC1101 just doesn't support multiple gateways because it's a single device over SPI and can't do multiple interrupts and you can't attach the receiver pin to multiple pins on the ESP like you can with the regular 433MHz receivers; I tried it and it didn't work), so I'm working on excluding it and just having RF and Pilight working together (and probably RF2, but don't have any RF2 devices to test with)

piano-thomas, 2023-01-20:
> updating

@melyux that would be great! The only thing I am struggling with is implement the ZGatewayPilight.ino changes in the patch. I'm lacking a lot of experience and experimenting with everything, couldn't figure out how to use the git diff feature to create a custom .ino file. Changing everything by hand will for sure cause errors ;)..

rinie, 2024-02-17:
I have a rough Lilygo working with the other RF libraries…

[“LilyGo_SSD1306”,“WebUI”,“RF”,“RF2”,“Pilight”,“rtl_433”]

[github.com](https://github.com/rinie/OpenMQTTGateway/tree/ZradioSX127x)
[GitHub - rinie/OpenMQTTGateway at ZradioSX127x](https://github.com/rinie/OpenMQTTGateway/tree/ZradioSX127x)
[ZradioSX127x](https://github.com/rinie/OpenMQTTGateway/tree/ZradioSX127x)

MQTT gateway for ESP8266 or ESP32 with bidirectional 433mhz/315mhz/868mhz, Infrared communications, BLE, Bluetooth, beacons detection, mi flora, mi jia, LYWSD02, LYWSD03MMC, Mi Scale, TPMS, BBQ the...

And patching the other libs…

I isolated the decoding of the libraries from the receiving by RTL_433_ESP…
Now struggling to get transmitting to work?..

NorthernMan54, 2024-02-17:
Fyi - rtl_433_esp and pilight libraries have conflicting names and can not be used together

rinie, 2024-02-17:
> Fyi - rtl_433_esp and pilight libraries have conflicting names and can not be used together

https://github.com/rinie/ESPiLight/tree/ZradioSX127x

I modified this to fix that.
Difficult to get EspPiLight compling...

NorthernMan54, 2024-02-17:
I never looked at the transmit capability of the sx127x chipset, but you need to find a mode where a pin can control the transmitter output.  Similar to direct mode I use for receiving signals

also the common rf functions should be placed into

rinie, 2024-02-18:
I thought DIO2 could be used for Transmit as for Receiving...
For now I added this to your rtl_433_ESP as that does the actual receiving. Just using the decoding of the other modules...
RFLINK32 seems to support transmit, but I get no output.
https://github.com/SevenW/embapps/blob/master/ArduinoRFM69/rfm69-RF12demo-ook-TX/rfm69-RF12demo-ook-TX.ino uses packetmode/FIFO timing ...

gerashk, 2024-05-10:
> I have a rough Lilygo working with the other RF libraries…
> 
> [“LilyGo_SSD1306”,“WebUI”,“RF”,“RF2”,“Pilight”,“rtl_433”]
> 
> [github.com](https://github.com/rinie/OpenMQTTGateway/tree/ZradioSX127x) [GitHub - rinie/OpenMQTTGateway at ZradioSX127x](https://github.com/rinie/OpenMQTTGateway/tree/ZradioSX127x) [ZradioSX127x](https://github.com/rinie/OpenMQTTGateway/tree/ZradioSX127x)
> 
> I isolated the decoding of the libraries from the receiving by RTL_433_ESP… As the other libs assume direct OOK signal on input pins, and SX127x must be initialized... Now struggling to get transmitting to work?..

I tried to run your code. but for some reason I got an error

"Compilation error: declaration of template parameter 'T' shadows template parameter"

closed 2024-08-24 by github-actions[bot]

melyux, 2024-12-11:
I made a draft PR where I share my code for how I implemented RF and Pilight together, simply by enabling both of them at once: #2130

melyux, 2024-12-11:
I use a single receiver, but its data pin is connected to multiple pins on the ESP, as @iWildStyle talked about above.

rinie, 2024-12-12:
Ok, but would it not make sense to do that in software instead of hardware.
1 Physical receiver using 2 or more RF decoders?

melyux, 2024-12-12:
It is a single receiver. But you have to attach its DATA pin to multiple pins on the ESP because the ESP uses hardware interrupts per-pin.