1technophile/OpenMQTTGateway #452: Missing library NewRemoteTransmitter.h

xstrex, 2019-09-07:
**Describe the bug**
Compilation fails in version 0.9.2 if #define ZgatewayRF2    "RF2" is enabled
Fails to find NewRemoteTransmitter.h referenced in ZgatewayRF2.ino

**To Reproduce**
Steps to reproduce the behavior:
1. Enable 
```#define ZgatewayRF2    "RF2"```
2. Run
```platformio run```
3. See error

**Expected behavior**
Clean compile

**Screenshots**
```
platformio run
Processing esp32dev-rf (platform: espressif32@1.9.0; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.7
|-- <ArduinoJson> 5.13.4
|-- <WifiManager> 1.0.0 #c3ff582
|   |-- <WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|-- <rc-switch> 2.6.2 #90adb34
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <FS> 1.0
|-- <WebServer> 1.0
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <Update> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pio/build/esp32dev-rf/src/main.ino.cpp.o
Generating partitions .pio/build/esp32dev-rf/partitions.bin
Compiling .pio/build/esp32dev-rf/libd65/PubSubClient_ID89/PubSubClient.cpp.o
Archiving .pio/build/esp32dev-rf/lib06a/libArduinoJson_ID64.a
Indexing .pio/build/esp32dev-rf/lib06a/libArduinoJson_ID64.a
Compiling .pio/build/esp32dev-rf/libf79/WiFi/ETH.cpp.o
Compiling .pio/build/esp32dev-rf/libf79/WiFi/WiFi.cpp.o
/Users/Strex/Documents/Programming/Arduino/OpenMQTTGateway-code-v0.9.2/main/ZgatewayRF2.ino:40:34: fatal error: NewRemoteTransmitter.h: No such file or directory

******************************************************************************
* Looking for NewRemoteTransmitter.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NewRemoteTransmitter.h"
* Web  > https://platformio.org/lib/search?query=header:NewRemoteTransmitter.h
*
******************************************************************************

compilation terminated.
*** [.pio/build/esp32dev-rf/src/main.ino.cpp.o] Error 1
========================================================== [ERROR] Took 9.82 seconds ==========================================================

================================================================== [SUMMARY] ==================================================================
Environment rfbridge      	[IGNORED]
Environment esp32dev-all  	[IGNORED]
Environment esp32dev-rf   	[FAILED]
Environment esp32dev-ir   	[IGNORED]
Environment esp32dev-ble  	[IGNORED]
Environment ttgo-lora32-v1	[IGNORED]
Environment nodemcuv2-all 	[IGNORED]
Environment nodemcuv2-ble 	[IGNORED]
Environment nodemcuv2-ir  	[IGNORED]
Environment nodemcuv2-rf  	[IGNORED]
Environment atmega-all    	[IGNORED]
Environment uno-rf        	[IGNORED]
Environment uno-fastled   	[IGNORED]
==================================================== 1 failed, 0 succeeded in 9.83 seconds ====================================================
```

**Environment (please complete the following information):**
* 0.9.2
* Unknown, missing

1technophile, 2019-09-07:
Hello,

Per default 
`[env:nodemcuv2-rf]`

is set with rcswitch library

To make it compile with ZgatewayRF2 you need to add on`lib_deps` section:
`  ${libraries.newremoteswitch}`

Note that you can also add:
`  '-DZgatewayRF2="RF2"'`

Directly to your `build_flags` so as to have all defined in the platformio.ini file or a prod_env.ini file.

1technophile, 2019-09-11:
I'm closing the issue , if the solution proposed is not ok feel free to reopen it

closed 2019-09-11 by 1technophile

xstrex, 2019-09-16:
Thanks @1technophile 
I believe this is related, how would use the PiLight library?

```
Compiling .pio/build/esp32dev-rf/src/main.ino.cpp.o
Generating partitions .pio/build/esp32dev-rf/partitions.bin
Compiling .pio/build/esp32dev-rf/libd65/PubSubClient_ID89/PubSubClient.cpp.o
Archiving .pio/build/esp32dev-rf/lib06a/libArduinoJson_ID64.a
Indexing .pio/build/esp32dev-rf/lib06a/libArduinoJson_ID64.a
Compiling .pio/build/esp32dev-rf/libf01/FS/FS.cpp.o
/Users/Strex/Documents/Programming/Arduino/OpenMQTTGateway-code-v0.9.2/main/ZgatewayPilight.ino:31:24: fatal error: ESPiLight.h: No such file or directory

*******************************************************************
* Looking for ESPiLight.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPiLight.h"
* Web  > https://platformio.org/lib/search?query=header:ESPiLight.h
*
*******************************************************************

compilation terminated.
*** [.pio/build/esp32dev-rf/src/main.ino.cpp.o] Error 1
Compiling .pio/build/esp32dev-rf/libf01/FS/vfs_api.cpp.o
=================================== [ERROR] Took 2.80 seconds ===================================

=========================================== [SUMMARY] ===========================================
Environment rfbridge      	[IGNORED]
Environment esp32dev-all  	[IGNORED]
Environment esp32dev-rf   	[FAILED]
Environment esp32dev-ir   	[IGNORED]
Environment esp32dev-ble  	[IGNORED]
Environment ttgo-lora32-v1	[IGNORED]
Environment nodemcuv2-all 	[IGNORED]
Environment nodemcuv2-ble 	[IGNORED]
Environment nodemcuv2-ir  	[IGNORED]
Environment nodemcuv2-rf  	[IGNORED]
Environment atmega-all    	[IGNORED]
Environment uno-rf        	[IGNORED]
Environment uno-fastled   	[IGNORED]
============================= 1 failed, 0 succeeded in 2.80 seconds =============================
```

1technophile, 2019-09-18:
Hello,

To make it compile with ZgatewayPilight you need to add on`lib_deps` section:
`  ${libraries.pilight}`

And
`   '-DZgatewayPilight="Pilight"''`

Directly to your `build_flags` so as to have all defined in the platformio.ini file or a prod_env.ini file.

1technophile, 2019-09-18:
Maybe I need to define one environment per gateway type as examples?

xstrex, 2019-09-18:
Thank you for the feedback, I think I understand. 

I've tried this, in platformio.ini:
```
[env:esp32dev-rf]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.pilight}
build_flags =
  ${com-esp.build_flags}
  # '-DZgatewayRF="RF"'
  '-DZgatewayPilight="Pilight"'
  '-DGateway_Name="OpenMQTTGateway_ESP32_RF"'
```
With this near the top `default_envs = esp32dev-rf`

But it still fails to compile, with the following:
```
pio run
Processing esp32dev-rf (framework: arduino; platform: espressif32@1.9.0; board: esp32dev)
--------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 1.9.0 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <WiFi> 1.0
|-- <ESP32 BLE Arduino> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ArduinoJson> 5.13.4
|-- <WifiManager> 1.0.0 #c3ff582
|   |-- <WebServer> 1.0
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|-- <rc-switch> 2.6.2 #90adb34
|-- <PubSubClient> 2.7
|-- <EEPROM> 1.0
|-- <SPI> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
Compiling .pio/build/esp32dev-rf/src/main.ino.cpp.o
Generating partitions .pio/build/esp32dev-rf/partitions.bin
Compiling .pio/build/esp32dev-rf/libf79/WiFi/ETH.cpp.o
Compiling .pio/build/esp32dev-rf/libf79/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32dev-rf/libf79/WiFi/WiFiAP.cpp.o
/Users/Strex/Documents/Programming/Arduino/OpenMQTTGateway-code-v0.9.2/main/ZgatewayPilight.ino:31:24: fatal error: ESPiLight.h: No such file or directory

*******************************************************************
* Looking for ESPiLight.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPiLight.h"
* Web  > https://platformio.org/lib/search?query=header:ESPiLight.h
*
*******************************************************************

compilation terminated.
*** [.pio/build/esp32dev-rf/src/main.ino.cpp.o] Error 1
=================================== [ERROR] Took 3.79 seconds ===================================

=========================================== [SUMMARY] ===========================================
Environment rfbridge      	[IGNORED]
Environment esp32dev-all  	[IGNORED]
Environment esp32dev-rf   	[FAILED]
Environment esp32dev-ir   	[IGNORED]
Environment esp32dev-ble  	[IGNORED]
Environment ttgo-lora32-v1	[IGNORED]
Environment nodemcuv2-all 	[IGNORED]
Environment nodemcuv2-ble 	[IGNORED]
Environment nodemcuv2-ir  	[IGNORED]
Environment nodemcuv2-rf  	[IGNORED]
Environment atmega-all    	[IGNORED]
Environment uno-rf        	[IGNORED]
Environment uno-fastled   	[IGNORED]
============================= 1 failed, 0 succeeded in 3.79 seconds =============================
```

1technophile, 2019-09-19:
My mistake, it should be:

`  ${libraries.esppilight}`

xstrex, 2019-09-19:
That fixed it, thank you.

1technophile, 2019-09-19:
examples added